openrazer/openrazer #2163: Support for [Razer BlackWidow v4]

JoeWhizer, 2023-11-14:
### Device Info

* Name: <BlackWidow v4>
* Product Number: <See under device>
* Model Number: <See under device>

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```

```
Please post the output of
    ls -lah /dev/input/by-id/
```

<details>

```
Please post the output of
    usbhid-dump -m 1532 -ed
```

</details>

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

KarkanAlzwayed, 2023-11-21:
I have this same keyboard and I'm getting a "failed" in front of the "Check device compatibilty" with:

- the following PIDs are not supported in the latest stable release of openrazer
- 1532:0287
- Check OpenRazer's issues and pull requests to see the status for your device

Everything else passes. Device is not recognized. I can provide more info if needed. Please let me know. 
Thanks

JoeWhizer, 2023-11-25:
Wasn't successful yet, I've found the [commit](https://github.com/openrazer/openrazer/commit/1278b22b67e0288c58cda9a1eb1145ef3a3f6c85) for the v4 pro and did all the stuff related to v4 Pro for the v4, added the device ID also executed scripts for fakedriver generation and appstream file. All seems good, it compiled without problems.

I also could insmod the kernel driver without kernel panic, but the deamon  doesn't recognize the keyboard.

JoeWhizer, 2023-11-25:
there is device_serial after insmod the driver:
-r--r----- 1 root root 4,1k 25. Nov 14:08 /sys/bus/hid/drivers/razerkbd/0003:1532:0287.000E/device_serial

Also with the right product ID. 

but restarting the daemon with the loaded kernel module only find the mouse:
`
Nov 25 14:09:18 ArchDesk openrazer-daemon[99103]: 2023-11-25 14:09:18 | razer                          | INFO     | Initialising Daemon (v3.7.0). Pid: 99103
Nov 25 14:09:18 ArchDesk openrazer-daemon[99103]: 2023-11-25 14:09:18 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
Nov 25 14:09:18 ArchDesk openrazer-daemon[99103]: 2023-11-25 14:09:18 | razer                          | INFO     | Found device.0: 0003:1532:0099.0002
Nov 25 14:09:18 ArchDesk openrazer-daemon[99103]: 2023-11-25 14:09:18 | razer.device0                  | INFO     | Initialising device.0 RazerBasiliskV3
Nov 25 14:09:18 ArchDesk openrazer-daemon[99103]: 2023-11-25 14:09:18 | razer                          | INFO     | Serving DBus
`

z3ntu, 2023-11-25:
Okay the driver side sounds good then. Are you sure then that the daemon you're running has support for your BlackWidow V4 with the correct PID there? When launching the daemon (with `--verbose` flag) then it should print all the devices it supports.

If you upload your branch somewhere I can also take a look.

JoeWhizer, 2023-11-25:
Ahh, yeah you're right. The daemon doesn't output the v4, only v4 Pro

<details><summary>debug output</summary>
<p>
```
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEliteDVaEdition ------------- (1532:006A)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEssential ------------------- (1532:006B)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerAtherisReceiver -------------------- (1532:0062)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationChroma ------------------ (1532:0F08)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationV2Chroma ---------------- (1532:0F20)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBasilisk --------------------------- (1532:0064)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskEssential ------------------ (1532:0065)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateReceiver ----------- (1532:0088)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateWired -------------- (1532:0086)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV2 ------------------------- (1532:0085)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV3 ------------------------- (1532:0099)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV3ProWired ----------------- (1532:00AA)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV3ProWireless -------------- (1532:00AB)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskXHyperSpeed ---------------- (1532:0083)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2019 --------------------- (1532:0241)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowElite -------------------- (1532:0228)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowEssential ---------------- (1532:0237)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowLite --------------------- (1532:0235)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealth ------------------ (1532:011B)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealthEdition ----------- (1532:010E)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowTournamentEdition2014 ---- (1532:011C)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowV3 ----------------------- (1532:024E)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowV3MiniHyperspeed --------- (1532:0258)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowV3MiniHyperspeedWireless - (1532:0271)
[... cut ...]
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerViperV2ProWireless ----------------- (1532:00A6)
2023-11-25 16:15:12 | razer                          | DEBUG    | Loaded device specification: RazerViperV3HyperSpeed ------------------ (1532:00B8)
2023-11-25 16:15:12 | razer                          | INFO     | Found device.0: 0003:1532:0099.0002
2023-11-25 16:15:12 | razer.device0                  | INFO     | Initialising device.0 RazerBasiliskV3
```
</p>
</details>

here are all the changes I did:
https://github.com/JoeWhizer/openrazer/commit/9168e7b1685f4c5a0e5c579ca36abd7d217558d1

z3ntu, 2023-11-25:
Not sure how you've installed the modified parts, but you could use e.g. this command to run the daemon from the source checkout: https://gist.github.com/z3ntu/c5ca8f6040d8934fafc5e686b234890e#file-gistfile1-txt-L12

JoeWhizer, 2023-11-25:
Thanks! now at least I got something, but somehow it gets permission denied on device_serial and poll_rate and it throws an exception:

<details><summary>Debug output</summary>
```
2023-11-25 17:22:55 | razer                          | DEBUG    | Emitted Device Added Signal
2023-11-25 17:22:55 | razer                          | INFO     | Found valid device.1: 0003:1532:0287.001D
2023-11-25 17:22:55 | razer.device1                  | INFO     | Initialising device.1 RazerBlackWidowV4
2023-11-25 17:22:55 | razer.device1                  | WARNING  | getting serial: [Errno 13] Permission denied: '/sys/devices/pci0000:00/0000:00:08.1/0000:0d:00.3/usb5/5-3/5-3.2/5-3.2.2/5-3.2.2:1.2/0003:1532:0287.001D/device_serial'
2023-11-25 17:22:55 | razer.device1                  | DEBUG    | getting serial:  count:1
2023-11-25 17:22:55 | razer.device1                  | WARNING  | getting serial: [Errno 13] Permission denied: '/sys/devices/pci0000:00/0000:00:08.1/0000:0d:00.3/usb5/5-3/5-3.2/5-3.2.2/5-3.2.2:1.2/0003:1532:0287.001D/device_serial'
2023-11-25 17:22:55 | razer.device1                  | DEBUG    | getting serial:  count:2
2023-11-25 17:22:55 | razer.device1                  | WARNING  | getting serial: [Errno 13] Permission denied: '/sys/devices/pci0000:00/0000:00:08.1/0000:0d:00.3/usb5/5-3/5-3.2/5-3.2.2/5-3.2.2:1.2/0003:1532:0287.001D/device_serial'
2023-11-25 17:22:55 | razer.device1                  | DEBUG    | getting serial:  count:3
2023-11-25 17:22:55 | razer.device1                  | WARNING  | getting serial: [Errno 13] Permission denied: '/sys/devices/pci0000:00/0000:00:08.1/0000:0d:00.3/usb5/5-3/5-3.2/5-3.2.2/5-3.2.2:1.2/0003:1532:0287.001D/device_serial'
2023-11-25 17:22:55 | razer.device1                  | DEBUG    | getting serial:  count:4
2023-11-25 17:22:55 | razer.device1                  | WARNING  | getting serial: [Errno 13] Permission denied: '/sys/devices/pci0000:00/0000:00:08.1/0000:0d:00.3/usb5/5-3/5-3.2/5-3.2.2/5-3.2.2:1.2/0003:1532:0287.001D/device_serial'
2023-11-25 17:22:55 | razer.device1                  | DEBUG    | getting serial:  count:5

2023-11-25 17:22:55 | razer.device1                  | DEBUG    | DBus call set_poll_rate
Exception in thread Thread-2 (_collecting_udev_method):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1045, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 982, in run
    self._target(*self._args, **self._kwargs)
  File "/home/zalkortis/razerfork/daemon/openrazer_daemon/daemon.py", line 605, in _collecting_udev_method
    self._add_device(d)
  File "/home/zalkortis/razerfork/daemon/openrazer_daemon/daemon.py", line 530, in _add_device
    razer_device = device_class(device_path=sys_path, device_number=device_number, config=self._config,
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zalkortis/razerfork/daemon/openrazer_daemon/hardware/keyboards.py", line 73, in __init__
    super().__init__(*args, **kwargs)
  File "/home/zalkortis/razerfork/daemon/openrazer_daemon/hardware/keyboards.py", line 25, in __init__
    super().__init__(*args, **kwargs)
  File "/home/zalkortis/razerfork/daemon/openrazer_daemon/hardware/device_base.py", line 1296, in __init__
    super().__init__(*args, **kwargs)
  File "/home/zalkortis/razerfork/daemon/openrazer_daemon/hardware/device_base.py", line 323, in __init__
    self.restore_dpi_poll_rate()
  File "/home/zalkortis/razerfork/daemon/openrazer_daemon/hardware/device_base.py", line 376, in restore_dpi_poll_rate
    poll_rate_func(self.poll_rate)
  File "/home/zalkortis/razerfork/daemon/openrazer_daemon/dbus_services/dbus_methods/mamba.py", line 319, in set_poll_rate
    with open(driver_path, 'w') as driver_file:
         ^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/sys/devices/pci0000:00/0000:00:08.1/0000:0d:00.3/usb5/5-3/5-3.2/5-3.2.2/5-3.2.2:1.2/0003:1532:0287.001D/poll_rate'
```
</details> 

I ran this from source folder:
PYTHONPATH="pylib:daemon" python3 ./daemon/run_openrazer_daemon.py -Fv --config=$PWD/daemon/resources/razer.conf

With testdata its recognized and also listed in RazerGenie, but its not really making a connection to the device I suppose, only to the generated fake devices. So something probably is different from v4 Pro to v4 that causes this issues.

z3ntu, 2023-11-25:
Then basically means the udev rules haven't run correctly since that should chown the files to plugdev

Legion495, 2023-12-23:
Hello Everyone.

So I was wondering why my clicky V4 Pro would not work and it led me to this.
I am on Fedora. Now since I am very much clueless I will provide as much data as the OP here.

If someone would direct me with some steps I might be able to help.

<details>
  <summary>lsusb</summary>
  legion495@fedora ~> lsusb -d 1532: -v

Bus 001 Device 004: ID 1532:028d Razer USA, Ltd [unknown]
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x028d [unknown]
  bcdDevice            1.01
  iManufacturer           1 Razer
  iProduct                2 Razer BlackWidow V4 Pro
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0086
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
[... 188 lines cut ...]
 25 7F 75 08 95 01 81 06 05 01 09 38 15 81 25 7F
 75 08 95 01 81 06 09 30 09 31 16 00 80 26 FF 7F
 75 10 95 02 81 06 C0 C0

001:004:001:DESCRIPTOR         1703350925.442235
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 29 A0 15 00 25 01
 75 01 95 A0 81 02 05 08 19 01 29 03 15 00 25 01
 75 01 95 03 91 02 95 05 91 01 C0 05 0C 09 01 A1
 01 85 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75
 10 81 00 75 08 95 13 81 01 C0 05 01 09 80 A1 01
 85 03 19 81 29 83 15 00 25 01 75 01 95 03 81 02
 95 05 81 01 75 08 95 14 81 01 C0 05 01 09 00 A1
 01 85 04 09 03 15 00 26 FF 00 35 00 46 FF 00 75
 08 95 15 81 00 C0 05 01 09 00 A1 01 85 05 09 03
 15 00 26 FF 00 35 00 46 FF 00 75 08 95 15 81 00
 C0

001:004:000:DESCRIPTOR         1703350925.445238
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

 </details>

z3ntu, 2023-12-28:
@Legion495 You have a different device, see #2044 for yours, and since that's part of OpenRazer v3.7.0 you should be able to update to that version and it should work. If not see https://github.com/openrazer/openrazer/wiki/Troubleshooting and if you need more help, join our community chat (see wiki) or open a new issue of type support.

apj570, 2024-02-09:
Hello, just got this same keyboard and was wondering if you guys have come up with a solution.

vividnightmare, 2024-02-18:
What is needed to progress this device to being supported?

z3ntu, 2024-03-03:
@vividnightmare Somebody needs to add support for this device, https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md has a rough outline of what needs to be done.

xavierog, 2024-03-04:
@JoeWhizer Did you get any progress with this keyboard? Your last message hints you were one permission/udev issue away from a breakthrough.

xavierog, 2024-03-04:
FWIW, I have spawned a Debian Sid VM, cloned this repository, applied JoeWhizer's patch on top of it and my first tests (i.e. just playing with the various scripts in `examples/`) reflect it simply works. I am going to set up a desktop environment and polychromatic to ensure it works from end to end (including udev rules). Stay tuned.

xavierog, 2024-03-04:
I think I clicked everything that could be clicked in Polychromatic and, overall, it works:

- brightness works
- built-in effects (and their options) work
- game mode works
- macros work, except on M6 (presumably just a small extra change)
- custom effects work; caveat: the "device graphic" that is shown is the Pro variant and thus:
  - shows too many LEDs
  - shows a round unnamed button instead of the M6 key

Still TODO:
- test udev rules
- update the device graphic to reflect the non-pro variant
- add M6 to the list of macro-friendly keys

xavierog, 2024-03-25:
FWIW, I have published DEB packages for Debian Sid with support for the Razer BlackWidow V4 keyboard:
```console
# apt install apt-transport-https
# wget -q https://xavier.kindwolf.org/deb/conf/xavier.pkgrepos%40kindwolf.org.gpg.key -O /etc/apt/keyrings/xavier-kindwolf.asc
# cat > /etc/apt/sources.list.d/kindwolf.sources <<EOF
Types: deb
URIs: https://xavier.kindwolf.org/deb/
Suites: sid
Components: main
EOF
# apt update
# apt install openrazer-meta=3.7.0+dfsg-1+kindwolf1
```

vividnightmare, 2024-03-25:
Right.

I think the idea here is to confirm that the patches are functional, add proper support for the device, and get them added to the upstream repository.

I have been using the patches linked by @JoeWhizer for several weeks now and can confirm that they add the necessary backend support for the device.

Can we get a developer/maintainer to review?  Or can @JoeWhizer perhaps create a pull request so we can get support into the next release?

xavierog, 2024-03-25:
@vividnightmare: there is already a pull request (which includes @JoeWhizer's work): #2211 
So we indeed need developers/maintainers to take a look at it. But a quick glance at recent pull requests reflects little to no activity for several weeks now.

closed 2024-04-06 by z3ntu

xavierog, 2024-04-21:
@z3ntu thanks for merging and releasing; 3.8.0 is now available in Debian Sid and I confirm it works flawlessly.

PR #2211 (2024-04-06): Add support for Razer BlackWidow V4
This pull request combines the initial work made in #2163 (the relevant commit retains its author) with a few extra lines related to the `M6` key.

Fixes #2163