openrazer/openrazer #2167: Support for Basilisk V3 Pro

joebertj, 2023-11-20:
### Device Info

* Name: Razer Basilisk V3 Pro
* Product Number: RZ01-04620200
* Model Number: RZ01-0462

### Device Information
```
Bus 003 Device 011: ID 1532:00aa Razer USA, Ltd Razer Basilisk V3 Pro
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x00aa 
  bcdDevice            1.00
  iManufacturer           0 
  iProduct                2 Razer Basilisk V3 Pro
  iSerial                 3 000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
[... 179 lines cut ...]
003:011:002:DESCRIPTOR         1700484953.016594
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

003:011:000:DESCRIPTOR         1700484953.019589
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 0B 81 03
 05 0C 0A 38 02 15 81 25 7F 75 08 95 01 81 06 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0

```

</details>

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

joebertj, 2023-11-20:
It works if connected via cable and dongle but not via BT.

z3ntu, 2023-11-22:
Why did you close this? Since `1532:00aa` I'd say we keep it open until the Bluetooth version is supported, no?

z3ntu, 2023-12-12:
If the mouse itself doesn't work then yes, OpenRazer wouldn't fix it. OpenRazer is enabling you to control the RGB, DPI, etc.

Mattron2000, 2024-07-15:
Hello guys, I propose my solution to this problem, no... it is not related to this beautiful OpenRazer ;-), but simply the Bluetooth configuration. (It seems strange to me that no one has come here to take this torment from the shoulders of many, but let’s try to close this issue)

MY SETUP:

**Device**: Thinkpad T14 Gen 2i
**OS**: Arch Linux BTW
**Linux** version: 6.6.31-1-lts
**DE**: KDE 6.0.4 Wayland
**bluetoothctl/bluez**: 5.76

**MY SOLUTION**

Press at the same time:

- the two side buttons (not the third called "Multi-function trigger")
- the mouse wheel until the click is heard (even if it rotates it does not matter...)

Stay like this for a few seconds until the Razer LED logo starts to make two fast beets of blue lights separated by a longer interval (example in Morse code: ..  .. .. ..)

Attempt the connection from the Bluetooth settings of KDE (or GNOME or what you have...) and you will appear "BSK V3 PRO".

Et Voilá connected this mouse to Linux, enjoy it!

If in case it still doesn’t solve the problem that Bluetooth settings doesn’t find it yet, try this troubleshoot below.

**TROUBLESHOOTING**

Open your terminal and open the following file `/etc/bluetooth/main.conf` with your text editor with root permssion as you need to uncomment a line.

Look for the following line `# ControllerMode = dual` and remove only `#`.

Save the file, close the editor, and type `sudo systemctl restart bluetooth` on the terminal.

If it doesn’t work yet, try again after a PC restart.

I hope this guide will help you with a lot of people.

Bye!

_I'm not native English speaker and for this translation I used the extension "Firefox Translation"_

LaptopDev, 2026-01-28:
If this is a standard spot for support for this mouse, I find my system freezes in arch when VT switching (switching ttys) from some tty to a tty with X running on it.

titandrive, 2026-10-06:
Im working on a PR to add full BT support for the basilisk. So far, it works perfectly in CachyOS and supports all OpenRazer features. 

https://github.com/titandrive/openrazer/tree/basilisk-v3-pro-bluetooth