openrazer/openrazer #2168: Razer Naga Epic Chroma does not support effects anymore

daxpedda, 2023-11-22:
### Distribution

Manjaro Linux
Kernel 6.6.1-1

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

When I try to change the effect or color of my scroll wheel or backlight on my Razer Naga Epic Chroma, I get an error that this is seemingly unsupported.
Checking the kernel logs I can see the following message:
`razermouse: matrix_effect_static not supported for this model`

This worked correctly before the current v3.7.0 and git (8b42841) version.

z3ntu, 2023-11-22:
Thanks for opening, I made some changes lately that probably broke your device, I'll look into fixing it asap!

daxpedda, 2023-11-24:
As far as I can tell nothing changed.
You can see the change I did here: https://github.com/daxpedda/openrazer/tree/naga-epic-chroma-effects.

I used that to install the DKMS modules and the daemon just to make sure. I think only the DKMS modules were required, am I missing anything?

z3ntu, 2023-11-24:
Did you use this for installing? https://github.com/openrazer/openrazer/wiki/Building-a-package#build-packages-from-a-different-github-repository

And then after installing the package, reboot so the new kernel modules get loaded.

z3ntu, 2023-11-24:
Yes just installing the updated `openrazer-driver-dkms-git` should be enough. Can you send `pacman -Q openrazer-driver-dkms-git` output?

daxpedda, 2023-11-24:
```
> pacman -Q openrazer-driver-dkms-git
openrazer-driver-dkms-git 3.7.0.r1.g8b428410-1
```
What is this supposed to show? I also tried with `--verbose` and `--info`, but it doesn't seem to show any relevant information, like which git repository was used.
I did confirm though that my fork and branch was used through the build logs.

z3ntu, 2023-11-24:
Yeah you didn't build from the branch you patched but from master branch (commit 8b428410), it should show fe1e6a5d in there. Please double check your PKGBUILD that you point it to the correct branch.

daxpedda, 2023-11-24:
So before I was using an edited build file inside of Pamac, which at least seemed to work.
Building it myself seemed to cause some problems in the `pkgver()` function but I managed to solve that by just using `echo 1`.

Will report back when I restart.

daxpedda, 2023-11-25:
Everything seems to work now!

z3ntu, 2023-11-25:
Already pushed, thanks! I'm currently not planning a bugfix release just for that device, I hope it's okay for you to run openrazer-git for now until the next release. Unless plenty more people with the same device are hitting this issue, but I'd guess the numbers of people with this device is rather low.

Thanks for reporting and testing it!

closed 2023-11-25 by z3ntu

daxpedda, 2023-11-25:
No worries, thank you for the quick fix!

commit 6efd65a0 (2023-11-25): driver: Fix effects on Naga Epic Chroma

During refactoring I missed adding the device into the functions that
are used.

Closes #2168

Fixes: 7b8d48e2 ("Move Naga Epic Chroma to effect API in kernel")