openrazer/openrazer #2169: Support for [Razer Strider Chroma]

SalColon, 2023-11-22:
###Device Info
Razer Strider Chroma

RZ02-04490
###Device Information

❯ lsusb -d 1532: -v

Bus 005 Device 004: ID 1532:00aa Razer USA, Ltd Razer Basilisk V3 Pro
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x00aa 
  bcdDevice            1.00
  iManufacturer           0 
  iProduct                2 Razer Basilisk V3 Pro
  iSerial                 3 000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
[... 461 lines cut ...]
 06 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 C0

001:003:001:DESCRIPTOR         1700674178.768140
 05 01 09 06 A1 01 85 01 05 07 05 07 19 E0 29 E7
 15 00 25 01 75 01 95 08 81 02 19 00 29 A0 15 00
 25 01 75 01 95 A0 81 02 75 08 95 02 81 01 05 08
 19 01 29 03 15 00 25 01 75 01 95 03 91 02 95 05
 91 01 C0 05 0C 09 01 A1 01 85 02 19 00 2A 3C 02
 15 00 26 3C 02 95 01 75 10 81 00 75 08 95 15 81
 01 C0 05 01 09 80 A1 01 85 03 19 81 29 83 15 00
 25 01 75 01 95 03 81 02 95 05 81 01 75 08 95 16
 81 01 C0 05 01 09 00 A1 01 85 04 09 03 15 00 26
 FF 00 35 00 46 FF 00 75 08 95 17 81 00 C0 05 01
 09 00 A1 01 85 05 09 03 15 00 26 FF 00 35 00 46
 FF 00 75 08 95 17 81 00 C0 05 01 09 00 A1 01 85
 07 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 17 81 00 C0

001:003:000:DESCRIPTOR         1700674178.772126
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

SalColon, 2023-11-26:
There seems to be support for it in OpenRGB. I could have sworn that OpenRGB uses the OpenRazer back-end for
support of Razer products. I'm looking into how to get "pcap".

z3ntu, 2023-11-27:
See https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol and https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md

And yes also to my knowledge openrgb uses OpenRazer as backend

SalColon, 2023-12-08:
OpenRazer already lists the Chroma Strider as being supported. I opened OpenRgb and was able to control it from there.

z3ntu, 2023-12-08:
> OpenRazer already lists the Chroma Strider as being supported.

Where do you see this?

SalColon, 2023-12-17:
Inside of OpenRGB, under supported devices. I'm using OpenRGB to control the Strider Chroma right now.

z3ntu, 2023-12-28:
Okay, no clue how that works with OpenRGB. In any case OpenRazer directly doesn't support this device yet.

salamipurple, 2024-12-31:
When will we have support for the Strider in the whole OpenRazer database?

z3ntu, 2025-01-02:
@salamipurple Once someone opens a PR to add support for the device. Until then not.

ckefalianou, 2025-05-04:
Guys, any news for **Strider** support by openRazer? OpenRGB has a disclaimer with possible hardware issues https://gitlab.com/CalcProgrammer1/OpenRGB#warning , so I won't be using that on a 135 euro mousemat...

ckefalianou, 2025-05-04:
> [@salamipurple](https://github.com/salamipurple) Once someone opens a PR to add support for the device. Until then not.

https://github.com/openrazer/openrazer/pull/2461

ckefalianou, 2025-05-04:
@z3ntu 

Tried to resolve this on my own. Following the Goliathus Chroma Support commit, I've created a PR for supporting  Strider Chroma: https://github.com/openrazer/openrazer/pull/2461  Please take a look and let me know if this works, as I urgently need this.

Thank you

PR #2461 (2025-05-07): Added Strider Chroma Extended support (1532:0C05) 
Tested and working for Razer Strider Chroma Extended Mat .

Tested on Ubuntu 24.10.

Effects tested successfuly:

1. Off
2. Spectrum
3. Wave
5. Reactive
6. Static
7. Breath

resolve #2458, resolve  #2169 resolve #2347
closed 2025-05-07 by z3ntu