openrazer/openrazer #2171: Random Keyboard Disconnection Issue on Ubuntu

viarnes, 2023-11-24:
I have been using a Blackwidow Lite (2018) keyboard on Ubuntu 20.04 for the past 6 months without any issues. However, two days ago, the keyboard started randomly disconnecting after a certain period of use. The disconnections occur unpredictably, with no specific pattern.

I have followed the troubleshooting guide and all outputs seem normal when the keyboard is connected. However, when it disconnects, it does not even appear in the output of the `lsusb` command.

I have tried connecting the keyboard to different USB ports, but the same issue persists. Additionally, I use the keyboard on a dual-boot PC (shared with Windows 10), and it works perfectly fine without any disconnections on Windows. I have also tested the keyboard on a MacBook, and it functions properly there as well. Furthermore, I am using a Razer mouse without any issues or disconnections.

Possible Causes (If I have to guess...)

- I recently updated to OpenRazer 3.7.0, so it might be related to that 
- I used the keyboard on Windows for the first time a few days ago. I have encountered a similar issue with a Bluetooth headphone, where I need to remove the device from Ubuntu and rediscover it in order to make it work again. Could a similar situation be happening here? Note that the keyboard is not a bluetooth device but who knows

These are my log outputs:

```
~ lsmod | grep razer
razerkbd               98304  0
razermouse            131072  0
hid                   147456  4 usbhid,razerkbd,hid_generic,razermouse
```
```
~ dmesg | grep razer
...
[10143.602499] razerkbd 0003:1532:0235.0033: input,hidraw6: USB HID v1.11 Mouse [Razer Razer BlackWidow Lite] on usb-0000:01:00.0-4/input2
```
```
~ journalctl -b | grep -i razer_mount
nov 24 14:24:26 desktop razer_mount[16114]: Driver razerkbd
nov 24 14:24:26 desktop razer_mount[16115]: Device_ID 1-4
nov 24 14:24:26 desktop razer_mount[16122]: Driver razerkbd
nov 24 14:24:26 desktop razer_mount[16127]: Device_ID 1-4:1.0
nov 24 14:24:26 desktop razer_mount[16129]: Driver razerkbd
nov 24 14:24:26 desktop razer_mount[16131]: Driver razerkbd
nov 24 14:24:26 desktop razer_mount[16132]: Device_ID 1-4:1.2
nov 24 14:24:26 desktop razer_mount[16133]: Device_ID 1-4:1.1
nov 24 14:24:26 desktop razer_mount[16135]: Driver razerkbd
nov 24 14:24:26 desktop razer_mount[16137]: Device_ID 0003:1532:0235.0028
nov 24 14:24:26 desktop razer_mount[16141]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0235.0028/
nov 24 14:24:26 desktop razer_mount[16140]: Driver razerkbd
nov 24 14:24:26 desktop razer_mount[16144]: Device_ID 0003:1532:0235.002A
nov 24 14:24:26 desktop razer_mount[16145]: Driver razerkbd
nov 24 14:24:26 desktop razer_mount[16146]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0235.0028/
nov 24 14:24:26 desktop razer_mount[16147]: Device_ID 0003:1532:0235.0029
nov 24 14:24:26 desktop razer_mount[16148]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0235.002A/
nov 24 14:24:26 desktop razer_mount[16150]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0235.0029/
nov 24 14:24:26 desktop razer_mount[16154]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0235.002A/
nov 24 14:24:26 desktop razer_mount[16155]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0235.0029/
nov 24 14:24:33 desktop razer_mount[16178]: Driver razerkbd
nov 24 14:24:33 desktop razer_mount[16179]: Device_ID 1-9
nov 24 14:24:33 desktop razer_mount[16184]: Driver razerkbd
nov 24 14:24:33 desktop razer_mount[16185]: Driver razerkbd
nov 24 14:24:33 desktop razer_mount[16189]: Device_ID 1-9:1.0
nov 24 14:24:33 desktop razer_mount[16191]: Device_ID 1-9:1.2
nov 24 14:24:33 desktop razer_mount[16192]: Driver razerkbd
nov 24 14:24:33 desktop razer_mount[16194]: Device_ID 1-9:1.1
nov 24 14:24:33 desktop razer_mount[16197]: Driver razerkbd
nov 24 14:24:33 desktop razer_mount[16200]: Driver razerkbd
nov 24 14:24:33 desktop razer_mount[16201]: Driver razerkbd
nov 24 14:24:33 desktop razer_mount[16202]: Device_ID 0003:1532:0235.002B
nov 24 14:24:33 desktop razer_mount[16203]: Device_ID 0003:1532:0235.002D
nov 24 14:24:33 desktop razer_mount[16204]: Device_ID 0003:1532:0235.002C
nov 24 14:24:33 desktop razer_mount[16205]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0235.002B/
nov 24 14:24:33 desktop razer_mount[16207]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0235.002D/
nov 24 14:24:33 desktop razer_mount[16209]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0235.002C/
nov 24 14:24:33 desktop razer_mount[16212]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0235.002B/
nov 24 14:24:33 desktop razer_mount[16213]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0235.002D/
nov 24 14:24:33 desktop razer_mount[16214]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0235.002C/
nov 24 14:24:41 desktop razer_mount[16244]: Driver razerkbd
nov 24 14:24:41 desktop razer_mount[16245]: Device_ID 1-4
nov 24 14:24:41 desktop razer_mount[16252]: Driver razerkbd
nov 24 14:24:41 desktop razer_mount[16253]: Device_ID 1-4:1.0
[... 91 lines cut ...]
-r--r----- 1 root plugdev 4096 nov 24 14:25 device_serial
-r--r----- 1 root plugdev 4096 nov 24 14:25 device_type
lrwxrwxrwx 1 root plugdev    0 nov 24 14:25 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
-r--r----- 1 root plugdev 4096 nov 24 14:25 firmware_version
-rw-rw---- 1 root plugdev 4096 nov 24 14:25 game_led_state
drwxr-xr-x 3 root plugdev    0 nov 24 14:25 hidraw
drwxr-xr-x 3 root plugdev    0 nov 24 14:25 input
-r--r----- 1 root plugdev 4096 nov 24 14:25 kbd_layout
-rw-rw---- 1 root plugdev 4096 nov 24 14:25 macro_led_effect
-rw-rw---- 1 root plugdev 4096 nov 24 14:25 macro_led_state
-rw-rw---- 1 root plugdev 4096 nov 24 14:25 matrix_brightness
--w--w---- 1 root plugdev 4096 nov 24 14:25 matrix_effect_breath
--w--w---- 1 root plugdev 4096 nov 24 14:25 matrix_effect_none
--w--w---- 1 root plugdev 4096 nov 24 14:25 matrix_effect_static
-r--r--r-- 1 root plugdev 4096 nov 24 14:25 modalias
drwxr-xr-x 2 root plugdev    0 nov 24 14:25 power
-r--r--r-- 1 root plugdev 4096 nov 24 14:25 report_descriptor
lrwxrwxrwx 1 root plugdev    0 nov 24 14:25 subsystem -> ../../../../../../../../bus/hid
-rw-rw---- 1 root plugdev 4096 nov 24 14:25 test
-rw-r--r-- 1 root plugdev 4096 nov 24 14:25 uevent
-r--r----- 1 root plugdev 4096 nov 24 14:25 version

```

Thanks in advance

lah7, 2023-11-24:
Here's a few other troubleshooting areas to check:

1. Check `dmesg` again but look for `usb` lines and around that. Maybe the kernel will state the reason for removal.

       sudo dmesg | grep usb

    Around the time I bought a [new chair,](https://www.reddit.com/r/IKEA/comments/10fj30l/markus_office_chair_static/) maybe once or twice the keyboard/mouse randomly went on/off. The kernel [suggested electromagnetic interference (EMI)](https://superuser.com/questions/1254898/usb-ports-get-disabled-by-hub-due-to-emi) could be the reason the USB controller got "reset". Haven't had it happen since.

1. Is Razer Synapse installed on Windows? If so, could it have updated the keyboard's firmware?

1. Has the Ubuntu kernel been recently updated? (`/var/log/apt/*`) 

    You could try booting the last one by holding SHIFT when the computer boots; this should bring up the GRUB menu to choose "Advanced options" → "Ubuntu ...... (fallback)" or similar and see if the previous kernel version has the problem too.

viarnes, 2023-11-26:
Hey @lah7 thanks for your answer.

Went back to Windows and updated Razer Synapse, along with performing other system updates. Surprisingly, the keyboard is no longer recognized on Windows, and the same issue persists on Linux and Mac. This situation suggests that the problem might be related to the keyboard's firmware.

However, I'm facing a conundrum: How can I possibly update the firmware on a keyboard that isn't being detected by any system? 

Given these circumstances, it appears that my next course of action is to pursue a warranty claim for the keyboard. If you have any further advice or suggestions before I proceed with that, I would greatly appreciate it.

lah7, 2023-11-26:
Just to check, did you perform the firmware update on _bare metal_? (as in, on a physical computer, not a virtual machine with USB passthrough?)

In Linux, you'll be able to see if the device is still recognised at all - the hardware could be stuck in DFU (firmware update) mode:

    lsusb

_(I'm curious what it comes up as)_

Or... the keyboard might have ended up in "demo mode" - depends if the keys are lit up, or if it's dead as a brick. While there isn't something specifically for BlackWidow Lite, some keyboards can be reset by holding specific keys before plugging it in:

https://mysupport.razer.com/app/answers/detail/a_id/1697/kw/reset%20keyboard

If `lsusb` could see the device, you might be in luck. Over on Windows, Razer's official "Device Detection Troubleshoot" tool might help recover it - haven't tried or heard of this tool to know for sure:

https://mysupport.razer.com/app/answers/detail/a_id/5776

Best of luck! If no system can see it as a USB device (even as a non-keyboard), it likely has bricked/corrupted firmware. Luckily it's still under warranty in that case.

viarnes, 2023-11-26:
Well, I reseted the keyboard and things got awkward.

Before resetting the keyboard, it was not recognized by the system at all, as confirmed by the absence of the device in the `lsusb` output.

However, after performing a reset on the keyboard (achieved by holding down the Esc, Caps Lock, and Spacebar keys), the computer began to detect the keyboard and disconnect it almost immeadiatly. During one of these brief connection windows, I successfully executed `lsusb` and confirmed the keyboard's presence before it disconnected again. The experience was somewhat different on a Windows machine: the keyboard was recognized, and the backlight would flash erratically between one to ten times before losing connection (all in less that a second)

I have attempted to resolve the issue using the Razer Detection Tool and the Driver Removal Tool provided by them, but neither yielded any positive results. Additionally, the Windows Device Manager does not list the keyboard or show any unidentified devices after it disconnects.

I have reached out to Razer support for assistance and am awaiting their response.

Thanks for your time! Will come back with any updates.

PS: Yes, I performed the update on bare metal.

closed 2025-02-21 by github-actions[bot]