openrazer/openrazer #2172: Support for [Razer Nommo V2]

robosito, 2023-11-24:
### Device Info

* Name: Nommo V2
* Product Number: RZ05-04750100-R3G1
* Model Number: RZ05-0475

### Device Information
```
Bus 005 Device 004: ID 1532:055c Razer USA, Ltd Razer Nommo V2
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x055c 
  bcdDevice            0.00
  iManufacturer           1 Actions
  iProduct                2 Razer Nommo V2
  iSerial                 3 ACTIONS-DEV-SN0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0098
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      74
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0028
        bInCollection           1
        baInterfaceNr(0)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0101 USB Streaming
        bAssocTerminal          0
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      3 (OUTPUT_TERMINAL)
        bTerminalID             3
        wTerminalType      0x0301 Speaker
        bAssocTerminal          0
        bSourceID               9
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      6 (FEATURE_UNIT)
        bUnitID                 9
        bSourceID               1
        bControlSize            1
        bmaControls(0)       0x03
          Mute Control
          Volume Control
        bmaControls(1)       0x00
        bmaControls(2)       0x00
        iFeature                0 
      Endpoint Descriptor:
        bLength                 9
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval              10
        bRefresh                0
        bSynchAddress           0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      2 Streaming
      bInterfaceProtocol      0 
      iInterface              0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       1
      bNumEndpoints           1
      bInterfaceClass         1 Audio
      bInterfaceSubClass      2 Streaming
      bInterfaceProtocol      0 
      iInterface              0 
      AudioStreaming Interface Descriptor:
        bLength                 7
        bDescriptorType        36
        bDescriptorSubtype      1 (AS_GENERAL)
        bTerminalLink           1
        bDelay                  1 frames
        wFormatTag         0x0001 PCM
      AudioStreaming Interface Descriptor:
        bLength                14
        bDescriptorType        36
        bDescriptorSubtype      2 (FORMAT_TYPE)
        bFormatType             1 (FORMAT_TYPE_I)
        bNrChannels             2
        bSubframeSize           2
        bBitResolution         16
        bSamFreqType            1 Discrete
        tSamFreq[ 0]        48000
        junk at descriptor end: 00 00 00
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            9
          Transfer Type            Isochronous
          Synch Type               Adaptive
          Usage Type               Data
        wMaxPacketSize     0x00c0  1x 192 bytes
        bInterval               1
        AudioStreaming Endpoint Descriptor:
          bLength                 7
          bDescriptorType        37
          bDescriptorSubtype      1 (EP_GENERAL)
          bmAttributes         0x01
            Sampling Frequency
          bLockDelayUnits         1 Milliseconds
          wLockDelay         0x0001

```

```
total 0
drwxr-xr-x 2 root root 260 nov.  24 14:05 .
drwxr-xr-x 4 root root 600 nov.  24 14:05 ..
lrwxrwxrwx 1 root root  10 nov.  24 08:05 usb-046d_Logitech_BRIO_6E8F4912-event-if00 -> ../event11
lrwxrwxrwx 1 root root   9 nov.  24 08:05 usb-046d_Logitech_BRIO_6E8F4912-event-if05 -> ../event6
lrwxrwxrwx 1 root root  10 nov.  24 14:05 usb-Actions_Razer_Nommo_V2_ACTIONS-DEV-SN0-event-if00 -> ../event22
lrwxrwxrwx 1 root root   9 nov.  24 08:05 usb-Razer_Razer_DeathAdder_V2-event-mouse -> ../event3
lrwxrwxrwx 1 root root  10 nov.  24 08:05 usb-Razer_Razer_DeathAdder_V2-if01-event-kbd -> ../event16
lrwxrwxrwx 1 root root  10 nov.  24 08:05 usb-Razer_Razer_DeathAdder_V2-if02-event-kbd -> ../event19
lrwxrwxrwx 1 root root   9 nov.  24 08:05 usb-Razer_Razer_DeathAdder_V2-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 nov.  24 08:05 usb-Razer_Razer_Huntsman_Elite-event-kbd -> ../event9
lrwxrwxrwx 1 root root  10 nov.  24 08:05 usb-Razer_Razer_Huntsman_Elite-if01-event-kbd -> ../event21
lrwxrwxrwx 1 root root  10 nov.  24 08:05 usb-Razer_Razer_Huntsman_Elite-if02-event-mouse -> ../event10
lrwxrwxrwx 1 root root   9 nov.  24 08:05 usb-Razer_Razer_Huntsman_Elite-if02-mouse -> ../mouse1

```

<details>

```
Please post the output of
    usbhid-dump -m 1532 -ed
```

</details>

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
[image]
[image]
[image]
[image]

LinuxSquare, 2024-03-01:
@z3ntu I also have the 055C w. Arch Linux.
However with [this guide](https://github.com/openrazer/openrazer/wiki/Request-support-for-new-device), I can't find out how to create the "pcap".

Would you mind pasting the necessary commands to provide the pcap, so I could somehow help out?
Thanks

z3ntu, 2024-03-03:
Thanks for opening the issue! You can also take a look at https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md how you could add support for this device yourself.

yokharian, 2024-12-10:
I also have this device, I support its development

zSnails, 2025-06-19:
Hi @robosito, did you end up doing any work on this?

robosito, 2025-06-20:
Hello !

No, sorry, there is no progress on my side.

spicalous, 2025-12-01:
_Disclaimer: not a C dev_

I made a very naive commit and added the Nommo V2 where I saw the Nommo (V1)
https://github.com/openrazer/openrazer/compare/master...spicalous:openrazer:add-nommo-v2-support

If anyone can help test - this would be much appreciated, otherwise I will try to find some time to learn how to test the changes.

spicalous, 2025-12-02:
After some debug logging in the python daemon I found that the following condition is not true when trying to match the device with the newly added class

```
        # device_id == 0003:1532:055C.0010 
        if re.match(pattern, device_id) is not None:

            # dev_path == /sys/devices/pci0000:00/0000:00:02.1/0000:02:00.0/0000:03:0c.0/0000:0a:00.0/usb3/3-3/3-3:1.0/0003:1532:055C.0010
            # os.listdir(dev_path)
            # uevent
            # physical_location
            # firmware_node
            # report_descriptor
            # power
            # driver
            # subsystem
            # input
            # hidraw
            # country
            # modalias
            if 'device_type' in os.listdir(dev_path):
                return True

        return False
```

spicalous, 2025-12-02:
Doing some debug logging in the kernel

```
dev_info(&intf->dev, "my log %d", dev->usb_interface_protocol);
dev_info(&intf->dev, "my log %d", USB_INTERFACE_PROTOCOL_MOUSE);
dev_info(&intf->dev, "my log %d", USB_INTERFACE_PROTOCOL_KEYBOARD);
```
results in 
```
[ 9738.179928] usbhid 3-3:1.0: my log 0
[ 9738.179932] usbhid 3-3:1.0: my log 2
[ 9738.179935] usbhid 3-3:1.0: my log 1
```

Which leads me to trying to set the expectedProtocol of the Nommo to be 0

This then resulted in the Python daemon complaining that 
```
/sys/devices/pci0000:00/0000:00:02.1/0000:02:00.0/0000:03:0c.0/0000:0a:00.0/usb3/3-3/3-3:1.0/0003:1532:055C.0015/device_type, file is not owned by plugdev
```

hlriffel, 2026-08-15:
hey, guys,

I've raised a PR to add support for this device. I tested it on my own and more testing notes are in the PR itself. It would be great if someone else could test it to make sure it covers all use cases.

@z3ntu how does the CR process work? Should I assign it to specific people? Thanks in advance