openrazer/openrazer #2177: Kernel module error, openrazer not working

RayZ3R0, 2023-12-05:
### Fedora 39, 6.5.6-300.fc39.x86_64

### Support

- [ ] Have you tried rebooting? -> Yes
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page? -> Yes

I installed polychromatic and openrazer, they had no issues when installing, but after troubleshooting and using the commands listed, I'm getting these errors.
[image]
[image]
[image]

z3ntu, 2023-12-05:
What happens when you run `sudo dnf install kernel-devel` ?

Sid127, 2024-06-13:
just ran into this on Fedora 40 as well... the kernel modules do not get built automatically on installing `openrazer-meta`
I had to manually build and install them via dkms. Hopefully this can be fixed soon

z3ntu, 2024-06-13:
We do have the dkms commands in the Fedora package so they should get called https://github.com/openrazer/OBS-packaging/blob/master/openrazer.spec#L154-L171

Did you maybe see an error or something during first Installation?

Sid127, 2024-06-13:
Nope, did not see any errors. Looking at the dkms command, it seems to be hitting an edge case where it builds only for the running kernel, and not every kernel on the disk.

My installation looked like
- Fresh install of F40 KDE
- dnf update -y
- installation of nvidia modules and openrazer
- reboot (to a new kernel version)

-------- Original Message --------

z3ntu, 2024-06-13:
Right, the man page also says that...

> If the kernel option is not specified it assumes the currently running kernel.

Unfortunately the `dkms build` / `dkms install` command doesn't support `--all` which would be what we want then.. Maybe we need to manually iterate over the installed kernels and provide them with `-k`?

z3ntu, 2024-07-02:
@Sid127 Are you interested in working on this? We probably just need a few lines of bash added around the lines I linked in the .spec file above that gets a list of installed kernels to build against, and then we iterate in `dkms install` over them. I don't think we need to adjust `dkms remove` part at all.

Sid127, 2024-07-02:
sure, I'll take a look at it tomorrow and make a Pull Request accordingly

z3ntu, 2024-07-02:
Thanks, appreciate it!

z3ntu, 2025-02-21:
That's another perk of having a stale bot 😉  (though it should've triggered earlier but due to a configuration error the bot never made it to this older issue in its run)

I'll mark it as a bug & packaging-related now, seems you know/knew what was wrong.