openrazer/openrazer #2178: Not syncing with razer viper ultimate wireless(1532:007B) on Linux Mint(Cinnammon)

GustavoSatig, 2023-12-06:
### Distribution
LInux MInt with Cinnammon

RazerGenie Version: 1.1.0
OopenRazer Daemon Version: 3.2.0
### Support

- [ yes ] Have you tried rebooting?
- [ yes ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

 My razer viper ultimate wireless is not syncing with the app razerGenie
[image]

z3ntu, 2023-12-06:
What's the output of these commands?

* `sudo dkms install openrazer-driver/3.2.0`
* `sudo modprobe razerkbd`

GustavoSatig, 2023-12-06:
- sudo dkms install openrazer-driver/3.2.0:
   - Module openrazer-driver/3.2.0 already installed on kernel
   5.15.0-89-generic (x86_64).

   - sudo modprobe razerkbd
   - nothing

z3ntu, 2023-12-06:
So far looks good.

`ls -al /sys/bus/hid/drivers/razermouse/0*/` ? And `systemctl --user status openrazer-daemon.service`

GustavoSatig, 2023-12-06:
ls -al /sys/bus/hid/drivers/razermouse/0*/
'/sys/bus/hid/drivers/razermouse/0003:1532:007B.0002/':
total 0
drwxr-xr-x 5 root plugdev    0 dez  6 08:56 .
drwxr-xr-x 5 root root       0 dez  6 08:56 ..
--w--w---- 1 root plugdev 4096 dez  6 08:56 charge_colour
--w--w---- 1 root plugdev 4096 dez  6 08:56 charge_effect
-r--r----- 1 root plugdev 4096 dez  6 08:56 charge_level
-rw-rw---- 1 root plugdev 4096 dez  6 08:56 charge_low_threshold
-r--r----- 1 root plugdev 4096 dez  6 08:56 charge_status
-r--r--r-- 1 root plugdev 4096 dez  6 08:56 country
-rw-rw---- 1 root plugdev 4096 dez  6 08:56 device_idle_time
-rw-rw---- 1 root plugdev 4096 dez  6 08:56 device_mode
-r--r----- 1 root plugdev 4096 dez  6 08:56 device_serial
-r--r----- 1 root plugdev 4096 dez  6 08:56 device_type
-rw-rw---- 1 root plugdev 4096 dez  6 08:58 dpi
lrwxrwxrwx 1 root plugdev    0 dez  6 08:56 driver ->
../../../../../../../bus/hid/drivers/razermouse
-r--r----- 1 root plugdev 4096 dez  6 08:56 firmware_version
drwxr-xr-x 3 root plugdev    0 dez  6 08:56 hidraw
drwxr-xr-x 3 root plugdev    0 dez  6 08:56 input
-rw-rw---- 1 root plugdev 4096 dez  6 10:41 logo_led_brightness
--w--w---- 1 root plugdev 4096 dez  6 08:56 logo_matrix_effect_breath
--w--w---- 1 root plugdev 4096 dez  6 08:56 logo_matrix_effect_none
--w--w---- 1 root plugdev 4096 dez  6 08:56 logo_matrix_effect_reactive
--w--w---- 1 root plugdev 4096 dez  6 08:56 logo_matrix_effect_spectrum
--w--w---- 1 root plugdev 4096 dez  6 08:56 logo_matrix_effect_static
--w--w---- 1 root plugdev 4096 dez  6 08:56 matrix_custom_frame
--w--w---- 1 root plugdev 4096 dez  6 08:56 matrix_effect_custom
-r--r--r-- 1 root plugdev 4096 dez  6 08:56 modalias
-rw-rw---- 1 root plugdev 4096 dez  6 08:58 poll_rate
drwxr-xr-x 2 root plugdev    0 dez  6 08:56 power
-r--r--r-- 1 root plugdev 4096 dez  6 08:56 report_descriptor
lrwxrwxrwx 1 root plugdev    0 dez  6 08:56 subsystem ->
../../../../../../../bus/hid
--w--w---- 1 root plugdev 4096 dez  6 08:56 test
-rw-r--r-- 1 root plugdev 4096 dez  6 08:56 uevent
-r--r----- 1 root plugdev 4096 dez  6 08:56 version

'/sys/bus/hid/drivers/razermouse/0003:1532:007B.0003/':
total 0
drwxr-xr-x 5 root plugdev    0 dez  6 08:56 .
drwxr-xr-x 5 root root       0 dez  6 08:56 ..
-r--r--r-- 1 root plugdev 4096 dez  6 08:56 country
lrwxrwxrwx 1 root plugdev    0 dez  6 08:56 driver ->
../../../../../../../bus/hid/drivers/razermouse
drwxr-xr-x 3 root plugdev    0 dez  6 08:56 hidraw
drwxr-xr-x 3 root plugdev    0 dez  6 08:56 input
-r--r--r-- 1 root plugdev 4096 dez  6 08:56 modalias
drwxr-xr-x 2 root plugdev    0 dez  6 08:56 power
-r--r--r-- 1 root plugdev 4096 dez  6 08:56 report_descriptor
lrwxrwxrwx 1 root plugdev    0 dez  6 08:56 subsystem ->
../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 dez  6 08:56 uevent

'/sys/bus/hid/drivers/razermouse/0003:1532:007B.0004/':
total 0
drwxr-xr-x 5 root plugdev    0 dez  6 08:56 .
drwxr-xr-x 5 root root       0 dez  6 08:56 ..
-r--r--r-- 1 root plugdev 4096 dez  6 08:56 country
lrwxrwxrwx 1 root plugdev    0 dez  6 08:56 driver ->
../../../../../../../bus/hid/drivers/razermouse
drwxr-xr-x 3 root plugdev    0 dez  6 08:56 hidraw
drwxr-xr-x 3 root plugdev    0 dez  6 08:56 input
-r--r--r-- 1 root plugdev 4096 dez  6 08:56 modalias
drwxr-xr-x 2 root plugdev    0 dez  6 08:56 power
-r--r--r-- 1 root plugdev 4096 dez  6 08:56 report_descriptor
lrwxrwxrwx 1 root plugdev    0 dez  6 08:56 subsystem ->
../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 dez  6 08:56 uevent

systemctl --user status openrazer-daemon.service
● openrazer-daemon.service - Daemon to manage razer devices in userspace
     Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service;
enabled; vendor preset: enabled)
     Active: active (running) since Wed 2023-12-06 08:58:54 -03; 1h 46min
ago
       Docs: man:openrazer-daemon(8)
   Main PID: 1659 (openrazer-daemo)
      Tasks: 5 (limit: 18914)
     Memory: 14.2M
        CPU: 3.696s
     CGroup: ***@***.***
/app.slice/openrazer-daemon.service
             └─1659 openrazer-daemon "" "" "" "" "" "" "" "" "" "" "" "" ""
"" "" "" "" "" "" "" "" "" "" "" "" "" "" "" ""

dez 06 08:58:54 gustavosatig-OptiPlex-3040 openrazer-daemon[1659]:
2023-12-06 08:58:54 | razer                          | INFO     |
Initialising Daemon (v3.2.0). Pid: 1659
dez 06 08:58:54 gustavosatig-OptiPlex-3040 openrazer-daemon[1659]:
2023-12-06 08:58:54 | razer.screensaver              | INFO     |
Initialising DBus Screensaver Monitor
dez 06 08:58:54 gustavosatig-OptiPlex-3040 openrazer-daemon[1659]:
2023-12-06 08:58:54 | razer                          | INFO     | Found
device.0: 0003:1532:007B.0002
dez 06 08:58:55 gustavosatig-OptiPlex-3040 openrazer-daemon[1659]:
2023-12-06 08:58:55 | razer.device0                  | INFO     |
Initialising device.0 RazerViperUltimateWireless
dez 06 08:58:55 gustavosatig-OptiPlex-3040 openrazer-daemon[1659]:
2023-12-06 08:58:55 | razer.device0.batterynotifier  | WARNING  | Failed to
init notification daemon, err: org.freedesktop.DBus.Error.ServiceUnknown:
The name org.freedes>
dez 06 08:58:55 gustavosatig-OptiPlex-3040 openrazer-daemon[1659]:
2023-12-06 08:58:55 | razer                          | INFO     | Serving
DBus
dez 06 10:24:38 gustavosatig-OptiPlex-3040 openrazer-daemon[1659]:
2023-12-06 10:24:38 | razer.device0                  | INFO     |
Suspending RazerViperUltimateWireless
dez 06 10:24:52 gustavosatig-OptiPlex-3040 openrazer-daemon[1659]:
2023-12-06 10:24:52 | razer.device0                  | INFO     | Resuming
RazerViperUltimateWireless
dez 06 10:31:29 gustavosatig-OptiPlex-3040 openrazer-daemon[1659]:
2023-12-06 10:31:29 | razer.device0                  | INFO     |
Suspending RazerViperUltimateWireless
dez 06 10:41:51 gustavosatig-OptiPlex-3040 openrazer-daemon[1659]:
2023-12-06 10:41:51 | razer.device0                  | INFO     | Resuming
RazerViperUltimateWireless

z3ntu, 2023-12-06:
That's interesting, also the daemon is detecting the device correctly.

Just to confirm, this line will print a device also? `python3 -c "from openrazer.client import DeviceManager; a = DeviceManager(); print(a.devices)"`

Next, could you launch `razergenie` from the terminal and check the output it prints?

GustavoSatig, 2023-12-06:
`python3 -c "from openrazer.client import DeviceManager; a = DeviceManager(); print(a.devices)"`
[<RazerMouse 632208H12216924>]

I launched razerGenie now, and it showed my mouse 

[image]
I just reboot my pc and started razerGenie and it showed

GustavoSatig, 2023-12-06:
> `python3 -c "from openrazer.client import DeviceManager; a = DeviceManager(); print(a.devices)"` [<RazerMouse 632208H12216924>]
>
> I launched razerGenie now, and it showed my mouse [...]

oops, sorry, i missclick to close the issue, btw i think its not a problem anymore