openrazer/openrazer #2179: Razer Cobra Pro

garykruger, 2023-12-06:
Using Razer Cobra Pro mouse via 2.4gz wireless works fine and is working well with Polychromatic. However I prefer to use my mouse with usb via direct usb connection which has an ?Unrecognised: 1532:00AF as reported by Polychromatic. Any help would be appreciated.

garykruger, 2023-12-07:
gary@OptiPlex:~$ lsusb -d 1532: -v

Bus 001 Device 003: ID 1532:0a24 Razer USA, Ltd Razer BlackWidow V3 Tenkeyless
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0a24 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer BlackWidow V3 Tenkeyless
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x006d
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
[... 206 lines cut ...]
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     327
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
gary@OptiPlex:~$

z3ntu, 2023-12-28:
Could you please test this branch? https://github.com/openrazer/openrazer/commits/cobra-wired/

See https://github.com/openrazer/openrazer/wiki/Building-a-package, and feel free to ask if you have any questions.

garykruger, 2023-12-29:
Sorry, this seems too complicated for me to figure out.

By the way, I use Linux Mint

z3ntu, 2023-12-30:
You just need to run the commands listed on the wiki page and replace `TARGET_BRANCH` with `cobra-wired`. If you can't test it, unfortunately I cannot merge the changes.

genft-robot, 2023-12-30:
trying to test this and when i install i am getting this:

Note, selecting 'openrazer-daemon' instead of './dist/openrazer-daemon_3.7.0-0_all.deb'
Note, selecting 'openrazer-doc' instead of './dist/openrazer-doc_3.7.0-0_all.deb'
Note, selecting 'openrazer-driver-dkms' instead of './dist/openrazer-driver-dkms_3.7.0-0_all.deb'
Note, selecting 'openrazer-meta' instead of './dist/openrazer-meta_3.7.0-0_all.deb'
Note, selecting 'python3-openrazer' instead of './dist/python3-openrazer_3.7.0-0_all.deb'
 how can i make sure apt (ubuntu) uses the *.debs in ./dist?

z3ntu, 2023-12-30:
Try `sudo dpkg -i ./dist/*.deb` then

genft-robot, 2023-12-30:
still getting 
[image]

z3ntu, 2023-12-30:
Did you reboot after installing the new packages?

z3ntu, 2023-12-30:
Actually I missed something in the changes, please do the steps again (including git clone, you can delete the old directory you `git clone`'d) and then reboot. Hopefully the device shows up then.

garykruger, 2023-12-30:
I get this error when doing this commnad:

sudo apt install debhelper dh-python linux-headers-generic python3-setuptools lsb-release dkms # build dependencies
./scripts/build_debs.sh

[image]

build.log:

[image]

z3ntu, 2023-12-30:
`sudo apt install build-essential` and then try the `build_debs.sh` step again

garykruger, 2023-12-30:
Ran your command and then these commands with no errors and then rebooted:

cd openrazer
sudo apt install debhelper dh-python linux-headers-generic python3-setuptools lsb-release dkms # build dependencies
./scripts/build_debs.sh
sudo apt install ./dist/*.deb

Now in Polychromatic:

[image]

garykruger, 2023-12-30:
Is it possible my system is using the incorrect version of openrazer:

[image]

webcoderz, 2023-12-30:
Okay will try in a bit (this is my other account that's like my CI account on my other PC )

webcoderz, 2023-12-30:
Yep looks to work now 👍

garykruger, 2023-12-30:
So I removed all openrazer and polychromatic packages and folders and run through the wiki commands to install the cobra-wired version and also polychromatic. After a reboot it now seems to work. Awesome!

z3ntu, 2023-12-30:
So clicking around in polychromatic/RazerGenie shows that everything works as expected? Any unexpected issues?

garykruger, 2023-12-30:
Mouse colors work fine. Lights turn off in screen saver. But don't come back on when waking from screen saver.  I have to got to polychromatic to turn on colors again.

webcoderz, 2023-12-30:
I don't have a screensaver / suspend but everything seemed to work from Lock Screen for me

z3ntu, 2023-12-31:
@garykruger Weird, this happens every time? Anything from razermouse in dmesg then?

garykruger, 2023-12-31:
Consistently no rgb after waking from screen saver. Here is my dmesg:

gary@OptiPlex:~$ dmesg
[    0.000000] Linux version 5.15.0-91-generic (buildd@lcy02-amd64-045) (gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #101-Ubuntu SMP Tue Nov 14 13:30:08 UTC 2023 (Ubuntu 5.15.0-91.101-generic 5.15.131)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.15.0-91-generic root=UUID=00c26390-457b-4dcb-941a-0f3e22ab5e60 ro quiet splash
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009c3ff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009c400-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000bff98fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bff99000-0x00000000bff99fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000bff9a000-0x00000000bffe3fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bffe4000-0x00000000c003cfff] usable
[    0.000000] BIOS-e820: [mem 0x00000000c003d000-0x00000000c083dfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000c083e000-0x00000000cef65fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000cef66000-0x00000000cf1bdfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000cf1be000-0x00000000cf20bfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000cf20c000-0x00000000cf9ebfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000cf9ec000-0x00000000cfffefff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000cffff000-0x00000000cfffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000d0000000-0x00000000d7ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000d8000000-0x00000000d85fffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000d8600000-0x00000000d97fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x00000002247fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 3.0.0 present.
[    0.000000] DMI: Dell Inc. OptiPlex 7040/0HD5W2, BIOS 1.24.0 07/14/2022
[    0.000000] tsc: Detected 3400.000 MHz processor
[    0.000000] tsc: Detected 3399.906 MHz TSC
[    0.000747] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000750] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000760] last_pfn = 0x224800 max_arch_pfn = 0x400000000
[    0.000879] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.001462] last_pfn = 0xd8600 max_arch_pfn = 0x400000000
[    0.008225] found SMP MP-table at [mem 0x000fcde0-0x000fcdef]
[    0.008237] Using GB pages for direct mapping
[    0.008589] RAMDISK: [mem 0x28fe3000-0x307e8fff]
[    0.008595] ACPI: Early table checksum verification disabled
[    0.008597] ACPI: RSDP 0x00000000000F05B0 000024 (v02 DELL  )
[    0.008601] ACPI: XSDT 0x00000000CF1DE0A0 0000C4 (v01 DELL   CBX3     01072009 AMI  00010013)
[    0.008606] ACPI: FACP 0x00000000CF200A70 00010C (v05 DELL   CBX3     01072009 AMI  00010013)
[    0.008611] ACPI: DSDT 0x00000000CF1DE1F8 022875 (v02 DELL   CBX3     01072009 INTL 20120913)
[    0.008613] ACPI: FACS 0x00000000CF9EBD80 000040
[    0.008616] ACPI: APIC 0x00000000CF200B80 0000BC (v03 DELL   CBX3     01072009 AMI  00010013)
[    0.008618] ACPI: FPDT 0x00000000CF200C40 000044 (v01 DELL   CBX3     01072009 AMI  00010013)
[    0.008621] ACPI: MCFG 0x00000000CF200C88 00003C (v01 DELL   CBX3     01072009 MSFT 00000097)
[    0.008624] ACPI: HPET 0x00000000CF200CC8 000038 (v01 DELL   CBX3     01072009 AMI. 0005000B)
[    0.008626] ACPI: SSDT 0x00000000CF200D00 00546C (v02 SaSsdt SaSsdt   00003000 INTL 20120913)
[    0.008629] ACPI: UEFI 0x00000000CF206170 000042 (v01                 00000000      00000000)
[    0.008632] ACPI: LPIT 0x00000000CF2061B8 000094 (v01 INTEL  SKL      00000000 MSFT 0000005F)
[    0.008634] ACPI: SSDT 0x00000000CF206250 000248 (v02 INTEL  sensrhub 00000000 INTL 20120913)
[    0.008637] ACPI: SSDT 0x00000000CF206498 002BAE (v02 INTEL  PtidDevc 00001000 INTL 20120913)
[    0.008640] ACPI: SSDT 0x00000000CF209048 000BE3 (v02 INTEL  Ther_Rvp 00001000 INTL 20120913)
[    0.008642] ACPI: SSDT 0x00000000CF209C30 0004A3 (v02 INTEL  zpodd    00001000 INTL 20120913)
[    0.008645] ACPI: DBGP 0x00000000CF20A0D8 000034 (v01 INTEL           00000000 MSFT 0000005F)
[    0.008647] ACPI: DBG2 0x00000000CF20A110 000054 (v00 INTEL           00000000 MSFT 0000005F)
[    0.008650] ACPI: SSDT 0x00000000CF20A168 000611 (v02 INTEL  DELL_SFF 00000000 INTL 20120913)
[    0.008653] ACPI: SSDT 0x00000000CF20A780 000E73 (v02 CpuRef CpuSsdt  00003000 INTL 20120913)
[    0.008655] ACPI: MSDM 0x00000000CF20B5F8 000055 (v03 DELL   CBX3     06222004 AMI  00010013)
[... 893 lines cut ...]
[... cut ...]
[   18.347541] audit: type=1400 audit(1703970433.485:26): apparmor="DENIED" operation="capable" profile="/usr/sbin/cups-browsed" pid=1663 comm="cups-browsed" capability=23  capname="sys_nice"
[31984.902791] audit: type=1400 audit(1704002400.111:27): apparmor="DENIED" operation="capable" profile="/usr/sbin/cupsd" pid=6780 comm="cupsd" capability=12  capname="net_admin"
[31984.918570] audit: type=1400 audit(1704002400.127:28): apparmor="DENIED" operation="capable" profile="/usr/sbin/cups-browsed" pid=6781 comm="cups-browsed" capability=23  capname="sys_nice"
gary@OptiPlex:~$

z3ntu, 2024-01-02:
Hm there's not, weird. What effect do you set before locking the screen, also what desktop environment are you using? Can you send a screenshot of polychromatic/RazerGenie *after* unlocking the screen when it's black (re-open the app and select the device so brightness/effect settings for the device are visible)? I'm curious what it's thinking the current effect is.

garykruger, 2024-01-02:
I set mouse preferences as static green at 50% brightness.  I am using linux mint 21.2 with cinnamon desktop. Here is polychromatic screenshot:

[image]

garykruger, 2024-01-02:
Screenshot is after screensaver when mouse has no RGB before any changes are made.

z3ntu, 2024-01-06:
If you manually change brightness again, does the device light up again?

I can definitely see in the daemon log that it's supposed to set the brightness to 0 when locking the screen, and then restoring the brightness once you unlock
```
2024-01-06 11:42:49 | razer.screensaver              | DEBUG    | Received screensaver active signal
2024-01-06 11:42:49 | razer.device0                  | INFO     | Suspending RazerCobraProWired
2024-01-06 11:42:49 | razer.device0                  | DEBUG    | DBus call set_brightness
2024-01-06 11:42:49 | razer.device0                  | DEBUG    | DBus call set_logo_brightness
2024-01-06 11:42:49 | razer.device0                  | DEBUG    | DBus call set_scroll_brightness
2024-01-06 11:42:51 | razer.screensaver              | DEBUG    | Received screensaver inactive signal
2024-01-06 11:42:51 | razer.device0                  | INFO     | Resuming RazerCobraProWired
2024-01-06 11:42:51 | razer.device0                  | DEBUG    | DBus call set_brightness
2024-01-06 11:42:51 | razer.device0                  | DEBUG    | DBus call set_logo_brightness
2024-01-06 11:42:51 | razer.device0                  | DEBUG    | DBus call set_scroll_brightness
```

garykruger, 2024-01-06:
So what I usually do after waking from screen saver is set the color from the polychromatic applet icon on the panel by the system clock and that works. BUT If I open the polychromatic app and try to set the brightness or color it does NOT work. It is ONLY when I select static, wave, etc. the the RGB lights come on.

z3ntu, 2024-03-03:
For anyone else following, please read https://github.com/openrazer/openrazer/pull/2190#issuecomment-1975129080

PR #2190 (2024-04-06): Add support for Razer Cobra Pro (Wired)
Rename the Wireless variant to Wireless.

Fixes #2179
closed 2024-04-06 by z3ntu