openrazer/openrazer #2180: RAZER DEATHADDER V2 PRO - Side buttons not registering events with XEV - EVTEST - ETC

maurokren, 2023-12-11:
Hello I am using Debian 12. Recently switched to Linux.

I have installed OpenRazer and I am able to switch DPI - Colors etc from my mouse using razer commander or razer genie without any issues.

However both thumb side buttons  and dpi buttons don't work at all.

When testing inputs through evtest or xev I don't even get an event registered from these buttons. Rest of buttons are ok.

When searching from xinput device "razer deathadder v2 pro" - side buttons show up listed as EVENT CODE 275 / 276, but still don't register or emite codes when testing with xev or evtest.

Any advice or fix would be appreciated.

z3ntu, 2023-12-11:
Can you check if they work when you stop the daemon with `systemctl --user stop openrazer-daemon.service` ?

maurokren, 2023-12-11:
When I type on terminal without root

systemctl --user stop openrazer-daemon.service - it does nothing and buttons still don't work

z3ntu, 2023-12-12:
Okay, at least it's consistent that way ;)

Unfortunately I can't do much myself without having the device in my hands, would be interesting to see if you switch the device to driver mode (`setDeviceMode(3, 0)`) what kind of events the kernel driver receives in `razer_raw_event`, some of those bytes should be the side buttons.

closed 2024-05-11 by github-actions[bot]