openrazer/openrazer #2181: Support for Huntsman V3 Pro Mini

AsessionGuy, 2023-12-14:
### Device Info

* Name:  Huntsman V3 Pro Mini
* Product Number: RZ03-04990200-R3U1
* Model Number: RZ03-0499

### Device Information
```
Please post the output of
    lsusb -d 1532: -v

    Bus 005 Device 002: ID 1532:02b0 Razer USA, Ltd Razer Huntsman V3 Pro Mini
```

```
Please post the output of
    ls -lah /dev/input/by-id/

    lrwxrwxrwx 1 root root   9 Dec 13 19:40 usb-Razer_Razer_Huntsman_V3_Pro_Mini-event-if01 -> ../event5
    lrwxrwxrwx 1 root root   9 Dec 13 19:40 usb-Razer_Razer_Huntsman_V3_Pro_Mini-event-kbd -> ../event3
    lrwxrwxrwx 1 root root   9 Dec 13 19:40 usb-Razer_Razer_Huntsman_V3_Pro_Mini-if01-event-kbd -> ../event4
    lrwxrwxrwx 1 root root   9 Dec 13 19:40 usb-Razer_Razer_Huntsman_V3_Pro_Mini-if02-event-mouse -> ../event6
    lrwxrwxrwx 1 root root   9 Dec 13 19:40 usb-Razer_Razer_Huntsman_V3_Pro_Mini-if02-mouse -> ../mouse0
```

<details>

```
Please post the output of
    usbhid-dump -m 1532 -ed

005:002:005:DESCRIPTOR         1702518062.822227
 06 00 FF 09 00 A1 01 15 00 26 FF 00 19 00 2A FF
 00 75 08 95 40 09 00 81 02 09 00 91 02 C0

005:002:004:DESCRIPTOR         1702518062.822595
 05 59 09 01 A1 01 85 01 09 02 A1 02 09 03 15 00
 27 FF FF 00 00 75 10 95 01 B1 03 09 04 09 05 09
 06 09 07 09 08 15 00 27 FF FF FF 7F 75 20 95 05
 B1 03 C0 85 02 09 20 A1 02 09 21 15 00 27 FF FF
 00 00 75 10 95 01 B1 02 C0 85 03 09 22 A1 02 09
 21 15 00 27 FF FF 00 00 75 10 95 01 B1 02 09 23
 09 24 09 25 09 27 09 26 15 00 27 FF FF FF 7F 75
 20 95 05 B1 02 09 28 09 29 09 2A 09 2B 09 2C 09
 2D 15 00 26 FF 00 75 08 95 06 B1 02 C0 85 04 09
 50 A1 02 09 03 09 55 15 00 25 08 75 08 95 02 B1
 02 09 21 15 00 27 FF FF 00 00 75 10 95 08 B1 02
 09 51 09 52 09 53 09 54 09 51 09 52 09 53 09 54
 09 51 09 52 09 53 09 54 09 51 09 52 09 53 09 54
 09 51 09 52 09 53 09 54 09 51 09 52 09 53 09 54
 09 51 09 52 09 53 09 54 09 51 09 52 09 53 09 54
 15 00 26 FF 00 75 08 95 20 B1 02 C0 85 05 09 60
 A1 02 09 55 15 00 25 08 75 08 95 01 B1 02 09 61
 09 62 15 00 27 FF FF 00 00 75 10 95 02 B1 02 09
 51 09 52 09 53 09 54 15 00 26 FF 00 75 08 95 04
 B1 02 C0 85 06 09 70 A1 02 09 71 15 00 25 01 75
 08 95 01 B1 02 C0 C0

005:002:002:DESCRIPTOR         1702518062.824001
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 06 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 C0

005:002:001:DESCRIPTOR         1702518062.824725
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 29 A0 15 00 25 01
 75 01 95 A0 81 02 75 08 95 1A 81 01 05 08 19 01
 29 03 15 00 25 01 75 01 95 03 91 02 95 05 91 01
 C0 05 0C 09 01 A1 01 85 02 19 00 2A 3C 02 15 00
 26 3C 02 95 01 75 10 81 00 75 08 95 2D 81 01 C0
 05 01 09 80 A1 01 85 03 19 81 29 83 15 00 25 01
 75 01 95 03 81 02 95 05 81 01 75 08 95 2E 81 01
 C0 05 01 09 00 A1 01 85 04 09 03 15 00 26 FF 00
 35 00 46 FF 00 75 08 95 2F 81 00 C0 05 01 09 00
 A1 01 85 05 09 03 15 00 26 FF 00 35 00 46 FF 00
 75 08 95 2F 81 00 C0 05 01 09 00 A1 01 85 07 09
 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95 2F 81
 00 C0 05 01 09 00 A1 01 85 0B 09 03 15 00 26 FF
 00 35 00 46 FF 00 75 08 95 2F 81 00 C0

005:002:000:DESCRIPTOR         1702518062.825722
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

001:004:003:DESCRIPTOR         1702518062.829276
 06 00 FF 09 00 A1 01 85 01 19 00 29 3F 15 00 26
 FF 00 75 08 95 3F 81 00 19 00 29 3F 15 00 26 FF
 00 75 08 95 3F 91 00 C0 05 0C 09 01 A1 01 85 02
 09 E9 09 EA 09 E2 75 01 95 03 15 00 25 01 81 02
 75 05 95 01 81 03 09 B6 09 CD 09 B5 75 01 95 03
 81 02 75 05 95 01 81 03 09 CF 75 01 95 01 81 02
 75 0F 95 01 81 03 C0

```

</details>

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment
I have no screenshots because I don't use Windows but I'd happy to provide them from a vw if it works and anything else you need

z3ntu, 2023-12-14:
Thanks for opening the issue! You can also take a look at https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md how you could add support for this device yourself.

mrhoomanwastaken, 2024-01-07:
If we are able to get support for the normal Huntsman v3 pro, will adding support for the mini be easier?

z3ntu, 2024-02-09:
@mrhoomanwastaken Most devices are quite similar so it wouldn't change much, you need to find what values work with this device and use them in OpenRazer.

If you can provide packet captures of Razer Synapse on Windows doing some communication, I can at least point you in the right direction where to start. https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

Anonymous17372, 2025-05-19:
Hey anyone working on this, still doesn't work to this date.

AsessionGuy, 2025-05-19:
> Hey anyone working on this, still doesn't work to this date.

I don't have the time nor the tools to work on this but as #2229 we might be closer?

Anonymous17372, 2025-05-19:
> > Hey anyone working on this, still doesn't work to this date.
> 
> I don't have the time nor the tools to work on this but as [#2229](https://github.com/openrazer/openrazer/issues/2229) is clsoed we might be closer?

Is there a way I could personally do it myself with a tutorial or is it not that straight forward :/

mrhoomanwastaken, 2025-05-19:
> Is there a way I could personally do it myself with a tutorial or is it not that straight forward :/

the info should all be here 
https://github.com/openrazer/openrazer/wiki

Sonbul001, 2025-12-28:
Hello, any updates on this?