openrazer/openrazer #2184: Support for RAZER BLACKSHARK V2 PRO (2023)

alansenairj, 2023-12-19:
### Device Info

* Name: RAZER BLACKSHARK V2 PRO (2023)
* Product Number: RZ04-04530100-R3U1
* Model Number: RZ04-0453

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
Bus 001 Device 008: ID 1532:0555 Razer USA, Ltd [unknown]
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0555 [unknown]
  bcdDevice            0.01
  iManufacturer           0 
  iProduct                2 Razer BlackShark V2 Pro 2.4
  iSerial                 3 O001000007
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00e0
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0046
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0201 Microphone
        bAssocTerminal          0
        bNrChannels             1
        wChannelConfig     0x0000
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                 8
[... 218 lines cut ...]
lrwxrwxrwx     9 root 19 Dec 14:54  ├── usb-Logitech_USB_Receiver-event-kbd -> ../event8
lrwxrwxrwx     9 root 19 Dec 14:54  ├── usb-Logitech_USB_Receiver-if01-event-mouse -> ../event9
lrwxrwxrwx     9 root 19 Dec 14:54  └── usb-Logitech_USB_Receiver-if01-mouse -> ../mouse1

```
Please post the output of
    usbhid-dump -m 1532 -ed
```

001:008:003:DESCRIPTOR         1703009198.031110
 05 0C 09 01 A1 01 85 01 15 00 25 01 09 CD 09 B5
 09 B6 09 B7 75 01 95 04 81 02 15 00 25 01 09 B0
 09 B1 09 B3 09 B4 75 01 95 04 81 22 15 00 25 01
 09 E9 09 EA 09 E2 75 01 95 03 81 22 75 05 95 01
 81 01 C0 06 00 FF 09 01 A1 01 85 02 15 00 26 FF
 00 09 02 75 08 95 3F 81 02 15 00 26 FF 00 09 03
 75 08 95 3F 91 02 C0

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

z3ntu, 2023-12-20:
Thanks for opening the issue! You can also take a look at https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md how you could add support for this device yourself.

alansenairj, 2023-12-20:
> Thanks for opening the issue! You can also take a look at https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md how you could add support for this device yourself.

It is really difficult to develop a driver for me. It is not my skill. I can write shellscripts and I know very little about python but I really don't know how to start. I read this link and it is about dev. environment setup. 

The problem is mic via USB dongle is not working. It can't be detected. 

By the way I can share my experience here using this product. It's output works well in blue tooth and USB dongle well. I just configure KDE output. In bluetooth I choose a HD codec because default configuration was nasty. 

Dongle USB configuration: 
[image]

BT configuration

[image]

alansenairj, 2023-12-20:
I am suspecting this issue is a problem in my  KDE Fedora instalation. When a do all steps to capture USB traffic I tested mic using Fedora 39 with Gnome and I can use mic. 

[image]

So I even need to capture nothing. I will do some update here and go back with results. 
[image]

alansenairj, 2023-12-20:
I got it. The configurations are pretty confuse using KDE. It is tricky to make it work. 

First i list the device 
arecord -l

[image]

Then I config setting like this

[image]

[image]

alansenairj, 2023-12-20:
So the problem was solved and it was a profile configuration issue at KDE advanced audio part.

closed 2023-12-20 by alansenairj

marcosilva85, 2024-01-24:
@alansenairj 
I assume you have the blackshark v2 pro 2023 version, which linux distro do you use?

    does the mic performs in [high-quality like in this video](https://youtu.be/7_xjpBYwm5g?t=307)
    how well does this model worked for you in linux, is it good in music and films or do you think the Windows software (synapse 3) would be required for this model to perform well by default?

alansenairj, 2024-01-25:
I Marcos. Are you Brazilian? 

I tested it in Fedora 38/39 upgraded. I installed this driver. 
I am using now a fresh Fedora 39 install. On fresh install I did nothing. It is detected by system and I am using it with dongle. 

The mic is the best. It is almost the same as a good dynamic mic I used with my soundcard later.

So if you want to deep dive in configs it is mandatory to install easy effects flatpak app. Despite it is a little awkward it is intuitive and you just get Linux DAW plugins working with your razer. Those plugins are pro and they are better than Razer's software. 

I am using   this setup and this one autoloads with system:

[image]

Askme anythig. I am daily user. You must adapt a little to use with teams. Sometimes I get some problems. It is not rock solid but Teams is not a rock solid software too.

marcosilva85, 2024-01-25:
Portuguese :)
Nice! I've seen customers so many problems with this 2023 pro version in Windows that hearing that it works fine in Linux is uplifting!

The dynamic mic is why I want to get it! Do you get impressive mic-sound by default, or do yid you had to equalize it?
(As I have need to use it in MacOS at work and company does not allow eq. software)

Did you had to update firmwares for some reason? Which firmware version are you currently running of dongle/headset?

alansenairj, 2024-02-07:
I Marcos, sorry for delay. 
Cara, eu tive um desastre aqui e tive que formatar o Fedora por um backup corrompido de OS que não funcionou. Mas temos boas e más notícias. A boa é que o dongle USB funcionou nativamente. A ruim é que o dongle muito provavelmente está acordando o PC da suspensão. 

Eu ia gravar um vídeo comparando o audio, mas não estou com tempo. O que o pessoal do trabalho me falou é que o som é o mesmo. 

Mais pra frente eu faço o comparativo e atualizo em Inglês. O fone é bom o mic é foda, mas tem esses problemas de acordar o PC no Fedora 39

guilhermetk, 2024-04-05:
@alansenairj 

can you share which app is that one you posted on which you can control Denoise, Voice detection, etc...? also, another BR here :)

alansenairj, 2024-04-08:
> @alansenairj
> 
> can you share which app is that one you posted on which you can control Denoise, Voice detection, etc...? also, another BR here :)

saudações Guilherme, use o easyeffects. 

Deixa eu passar um feedback. O funcionamento do fone não é tão estável assim. Já tive problemas loucos aqui durante o uso. Gostaria que fosse mais confiável. Mas funciona bem no geral. 

Aqui é Fedora, e tem pipewire no soim, isso é importante. Funcionou sem precisar do driver. às vezes eu preciso desconectar o dongle para ele reencontrar. 

O software para carregar os plugins é muito bom. é flexível e carrega rápido. Mas nas novas versões do proprio teams eu tenho observado que o filtro de barulho tem funcionado a contento. 

no geral, não é essa maravilha, gostaria que fosse mais confiável, tipo, ligou funcionou, mas tem que conviver né. 

vlww e boa sorte 

[image]

[image]

makarov-roman, 2024-06-06:
V2 Pro (2023) ,Fedora 40/Gnome  - I can see the mic, but it doesn't work. installed openrazer with no effect. Any ideas? how to fix it?

Xpiravit, 2024-09-05:
@makarov-roman did you get it working?

alansenairj, 2024-09-07:
me too. Fedora 39 mic worked but Fedora 40 not. I got some issues on sound but it is related to my /home reusing. 

My mic is not working. Even using it connect in Bluetooth .

makarov-roman, 2024-09-25:
@Xpiravit yeah, but I don't understand how. At some point it just started to work as expected. I removed openrazer and didn't install any additional components.
Make sure to turn on physical switch, select Analog output in sound settings (without that volume loudness is limited for me).

alansenairj, 2024-09-25:
Hi! I am using now with Bluetooth. Mic is working. I am not using with dongle anymore. 
About Fedora 40, there was a lot of problems in audio reusing /home because of kde upgrade from 5 to 6 differs a lot from each other. But beside that my system is stable now using bluetooth. I hope it helps you in some way.

xela-zone, 2025-08-09:
i'd like to be able to get the battery levle of my headset - any idea how to do that?

alansenairj, 2025-08-10:
> i'd like to be able to get the battery levle of my headset - any idea how to do that?

KDE gives the level by default

[image]