openrazer/openrazer #2187: Are there some way to use use this driver in Fedora 39 upgrade? I got some problems doing system upgrade. Does it support Fedora 39?

alansenairj, 2023-12-26:
### Distribution

Fedora 38

### Support

- [ ] Have you tried rebooting? not related
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?
not related

--- problem ----
I think it has a problem with newest python in Fedora 39. Does it support this version ? 
# command
sudo dnf system-upgrade download --releasever=39 

--- results ---
 Problem 3: problem with installed package openrazer-daemon-3.7.0-1.1.noarch
  - package openrazer-daemon-3.7.0-1.1.noarch from @System requires python3.11dist(daemonize) >= 2.4.7, but none of the providers can be installed
  - package openrazer-daemon-3.7.0-1.1.noarch from hardware_razer requires python3.11dist(daemonize) >= 2.4.7, but none of the providers can be installed
  - python3-daemonize-2.5.0-15.fc38.noarch from @System  does not belong to a distupgrade repository
 Problem 4: problem with installed package python3-openrazer-3.7.0-1.1.noarch
  - package python3-openrazer-3.7.0-1.1.noarch from @System requires python3.11dist(dbus-python) >= 1.2, but none of the providers can be installed
  - package python3-openrazer-3.7.0-1.1.noarch from hardware_razer requires python3.11dist(dbus-python) >= 1.2, but none of the providers can be installed
  - python3-dbus-1.3.2-2.fc38.x86_64 from @System  does not belong to a distupgrade repository
 Problem 5: problem with installed package openrazer-meta-3.7.0-1.1.noarch
  - package openrazer-meta-3.7.0-1.1.noarch from @System requires openrazer-daemon, but none of the providers can be installed
  - package openrazer-meta-3.7.0-1.1.noarch from hardware_razer requires openrazer-daemon, but none of the providers can be installed
  - package openrazer-daemon-3.7.0-1.1.noarch from @System requires python3.11dist(pygobject) >= 3.20, but none of the providers can be installed
  - package openrazer-daemon-3.7.0-1.1.noarch from hardware_razer requires python3.11dist(pygobject) >= 3.20, but none of the providers can be installed
  - python3-gobject-base-3.44.2-1.fc38.x86_64 from @System  does not belong to a distupgrade repository

z3ntu, 2023-12-26:
Fedora 39 packages should be available the same as 38 https://build.opensuse.org/projects/hardware:razer/packages/openrazer/repositories/Fedora_39/binaries

Maybe you need to edit the url for the repo manually on your system, I think normally the release upgrade should do that automatically

alansenairj, 2023-12-27:
So I got the problem. As recommended I am using latest installation with added repo as instructions said. It was not an  official repo, so installation cannot update it. 

I just upgrade using --allowerasing and reinstalled it again before upgrade to 39 finished and rebooted. 

Thanx for your help.

closed 2023-12-27 by alansenairj

MrAn0nym, 2024-11-02:
Got the same problem whilst upgrading to fedora 41. Managed to fix it by changing the repo to fedora 41 before upgrade (`sudo dnf config-manager --add-repo https://download.opensuse.org/repositories/hardware:/razer/Fedora_41/hardware:razer.repo`).
Dunno if that's just a problem on my end though :shrug:

z3ntu, 2024-11-14:
@MrAn0nym Please remove the file `/etc/yum.repos.d/hardware:razer.repo` and run the add-repo command from the website https://openrazer.github.io again.