openrazer/openrazer #2188: Cobra mouse PID shows not supported on arch git package

Kn4ughty, 2023-12-27:
### Distribution
Arch Linux 6.6.8-arch1-1

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I first installed the release package from pacman, then saw it did not work in polychromatic, I followed the troubleshooting guide and found my mouse has had support added but only on git. I went through the steps for that. (installed openrazer-meta-git package on aur via paru) and it still says my PID is not supported. 
```
$ pacman -Q | grep openrazer                 
libopenrazer 0.2.0-1
openrazer-daemon 3.7.0-1
openrazer-driver-dkms 3.7.0-1
openrazer-meta-git 3.7.0.r4.g85e81ae3-1
python-openrazer 3.7.0-1
$ sudo dkms install openrazer-driver/3.7.0
Module openrazer-driver/3.7.0 already installed on kernel 6.6.8-arch1-1 (x86_64), skip. You may override by specifying --force.
$ groups
plugdev wheel d
$ sudo modprobe razerkbd
$
```
[image]

z3ntu, 2023-12-27:
You also need to install `openrazer-daemon-git`, `openrazer-driver-dkms-git` and `python-openrazer-git` for it to work. If you run `makepkg -sri` where the PKGBUILD is, it should install all subpackages automatically.

Kn4ughty, 2023-12-27:
That fixed it! Thank you very much for the help

closed 2023-12-27 by Kn4ughty