openrazer/openrazer #2192: Razer Basilisk X Hyperspeed

dancadar, 2024-01-09:
### Distribution

opensuse leap 15.5

### Support

- [ x] Have you tried rebooting?
- [ x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I have an issue with my mouse that whenever is unused and entering sleep mode (as expected), when i start using it again it will take a long time to reconnect, or in some extreme cases it won't reconnect at all. The most annoying thing is that sometimes it will enter sleep after just a few seconds and trying to set "sleep mode after X seconds" doesn't seem to have any effect on it. To exclude the possibility that there is something wrong with the hardware, i've checked that the firmware of both the dongle and the mouse are up to date; also it has the same behavior on macos (the macos app uses openrazer) but at the same time it works without any issue on windows, which makes me believe that there is somewhere a bug within openrazer. Could not find anything in the logs apart from:
```
[Tue Jan  9 11:55:38 2024] razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 16, command_class: 00, command_id.id: 82 Params: 00000000000000000000000000000000 .
[Tue Jan  9 11:55:39 2024] razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 16, command_class: 00, command_id.id: 82 Params: 00000000000000000000000000000000 .
[Tue Jan  9 11:55:39 2024] razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 16, command_class: 00, command_id.id: 82 Params: 00000000000000000000000000000000 .
[Tue Jan  9 11:55:39 2024] razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 16, command_class: 00, command_id.id: 82 Params: 00000000000000000000000000000000 .
[Tue Jan  9 11:55:39 2024] razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 16, command_class: 00, command_id.id: 82 Params: 00000000000000000000000000000000 .
[Tue Jan  9 11:55:39 2024] razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 07, command_class: 04, command_id.id: 05 Params: 01070807080000000000000000000000 .
[Tue Jan  9 11:55:39 2024] razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 01, command_class: 00, command_id.id: 05 Params: 02000000000000000000000000000000 .
[Tue Jan  9 11:55:39 2024] razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[Tue Jan  9 12:05:07 2024] razermouse: Response doesn't match request. status: 02 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 00, command_id.id: 81 Params: 01020000000000000000000000000000 .
[Tue Jan  9 12:05:40 2024] razermouse: Response doesn't match request. status: 02 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00eb0000000000000000000000000000 .
```

z3ntu, 2024-02-09:
> trying to set "sleep mode after X seconds" doesn't seem to have any effect on it

Where do you set this?

> the macos app uses openrazer

In any case, macOS is not supported by this project - but reading your message makes me believe you're also seeing the behavior in your openSUSE install.

The error messages in the log are somewhat expected in case the device is asleep but the kernel still wants to send it messages, then they error out like this. Generally nothing to be worried about on that part, so not the smoking gun.

dancadar, 2024-02-09:
I've set it  with polychromatic.

I did see that the behaviour in openSuse; I was only pointing out that: 
- on another platform (macos) with software that is based on openrazer (https://github.com/1kc/razer-macos) it behaves in the same manner
- on yet another platform (windows) with software that is not based on openrazer (Synapse3) it works as it should, without disconnecting
These different behaviours made me believe that there is nothing wrong with the hardware itself, but more like some bug with openrazer. Unfortunatelly, i could not find anything in the logs, apart from those timeout messages. 

Anyway, in the meantime I got another mouse (Basilisk v3 Hyperspeed) and while i get the same messages in the logs, it works without issues. I guess I will use the old mouse on Windows, as it has absolutely no issues in Windows.

z3ntu, 2024-03-03:
Since the mouse should work without OpenRazer installed (as in mouse functionality, without RGB settings), does the mouse behave the same way without openrazer installed?

closed 2024-08-01 by github-actions[bot]