openrazer/openrazer #2196: Unsupported Device: 1532:00A3

lixiaolong2007, 2024-01-19:
I'm using archlinux I tried openrazer-daemon and openrazer-meta-git but in polychromatic it says device unsupported, I ran the troubleshooting and it worked fine, then I used the cobra-wired branch and it says device unsupported.

```
[vega9500@vega9500 ~]$ lsusb -d 1532: -v
Bus 001 Device 002: ID 1532:00a3 Razer USA, Ltd Razer Cobra
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x00a3 Razer Cobra
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Cobra
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

```
[vega9500@vega9500 ~]$     ls -lah /dev/input/by-id/
总计 0
drwxr-xr-x 2 root root 220  1月19日 20:24 .
drwxr-xr-x 4 root root 620  1月19日 20:24 ..
lrwxrwxrwx 1 root root  10  1月19日 20:24 usb-RAPOO_Rapoo_2.4G_Wireless_Device-event-if01 -> ../event12
lrwxrwxrwx 1 root root  10  1月19日 20:24 usb-RAPOO_Rapoo_2.4G_Wireless_Device-event-mouse -> ../event10
lrwxrwxrwx 1 root root  10  1月19日 20:24 usb-RAPOO_Rapoo_2.4G_Wireless_Device-if02-event-kbd -> ../event13
lrwxrwxrwx 1 root root   9  1月19日 20:24 usb-RAPOO_Rapoo_2.4G_Wireless_Device-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9  1月19日 20:24 usb-Razer_Razer_Cobra-event-if01 -> ../event6
lrwxrwxrwx 1 root root   9  1月19日 20:24 usb-Razer_Razer_Cobra-event-mouse -> ../event4
lrwxrwxrwx 1 root root   9  1月19日 20:24 usb-Razer_Razer_Cobra-if01-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9  1月19日 20:24 usb-Razer_Razer_Cobra-if02-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9  1月19日 20:24 usb-Razer_Razer_Cobra-mouse -> ../mouse0

```
[image]

lah7, 2024-01-19:
Looks like your device was recently added 85e81ae3ba08f2af33031e8a08a4f0ecc6adee91, but is not released yet. For now,  install all the [`-git` versions](https://aur.archlinux.org/pkgbase/openrazer-git) of those packages from the AUR:

    openrazer-daemon-git python-openrazer-git openrazer-driver-dkms-git

After installing these, it should work after a reboot. Installing `openrazer-meta-git` on its own might not have installed the other packages (it depends on the AUR tool)

lixiaolong2007, 2024-01-19:
This is strange. Other devices can be used normally, and I have installed and tried Meta, but it still doesn't work. I thought it was an issue with Aur, so I built the package myself, but it didn't work either

lixiaolong2007, 2024-01-19:
I am certain that my mouse is genuine and can be recognized properly on Windows

lah7, 2024-01-20:
Should work, the latest commit uses the same device ID `1532:00A3`. What versions of the AUR `-git` packages do you have installed?

    pacman -Q | grep openrazer

The version string should mention `85e81ae`. There should be two Cobras in the driver sources:

    grep -r 'Cobra' /usr/src/openrazer-driver*/

 A reboot (if you haven't already) will load the newer driver and daemon if it is the latest master branch code.

lixiaolong2007, 2024-01-21:
Finally, I chose to give up arch and switch to Ubuntu, as it is now working properly

closed 2024-01-21 by lah7