openrazer/openrazer #2197: Bazilisk V3 charging not detected anymore when connected via USB on Fedora 39 and kernel 6.6.11

ad-on-is, 2024-01-20:
When I connect the mouse, razer-cli shows the following info. As can be seen, charging isn't detected. This used to work before I updated my system recently. The charge was detected correctly and charging was set to true.

```
Razer Basilisk V3 Pro (Wireless):
   type: mouse
   DPI: (0, 0)
   max DPI: 26000
   DPI stages: (0, [])
   polling rate: 500
   battery:
      charge: 0
      charging: False
      idle delay 0 seconds
   generic zone:
      colors: ['00FF00', '00FFFF', '0000FF']
      effect: spectrum
      speed: 1
      wave_dir: 1
      effects available: none, spectrum, static, wave, multicolor
   logo zone:
      brightness: 30.0
      colors: ['00FF00', '00FFFF', '0000FF']
      effect: spectrum
      speed: 1
      wave_dir: 1
      effects available: brightness, none, spectrum, static, wave
   scroll_wheel zone:
      brightness: 30.0
      colors: ['00FF00', '00FFFF', '0000FF']
      effect: spectrum
      speed: 1
      wave_dir: 1
      effects available: brightness, none, spectrum, static, wave
   serial: PMXXXXXXXX4
   firmware version: v0.0
   driver version: 3.7.0
```

When disconnected, it is shown correctly.

```
Razer Basilisk V3 Pro (Wireless):
   type: mouse
   DPI: (800, 800)
   max DPI: 26000
   DPI stages: (3, [(400, 400), (600, 600), (800, 800), (1000, 1000)])
   polling rate: 500
   battery:
      charge: 27
      charging: False
      idle delay 300 seconds
   generic zone:
      colors: ['00FF00', '00FFFF', '0000FF']
      effect: spectrum
      speed: 1
      wave_dir: 1
      effects available: none, spectrum, static, wave, multicolor
   logo zone:
      brightness: 30.0
      colors: ['00FF00', '00FFFF', '0000FF']
      effect: spectrum
      speed: 1
      wave_dir: 1
      effects available: brightness, none, spectrum, static, wave
   scroll_wheel zone:
      brightness: 30.0
      colors: ['00FF00', '00FFFF', '0000FF']
      effect: spectrum
      speed: 1
      wave_dir: 1
      effects available: brightness, none, spectrum, static, wave
   serial: PMXXXXXXXX4
   firmware version: v1.1
   driver version: 3.7.0
```

z3ntu, 2024-02-09:
Since OpenRazer v3.7.0 was released November 1st, I assume you weren't on e.g. OpenRazer v3.6.1 before updating your system?

Do you see any errors from razermouse in dmesg when this happens?

ad-on-is, 2024-02-09:
I do regular updates, so I was on 3.7.0 for the time being.

But, I think this can be closed now, as it somehow miraculously fixed itself.

closed 2024-03-03 by z3ntu