openrazer/openrazer #2199: Build Instructions for Ubuntu result in `error: option --install-layout not recognized`

alichtman, 2024-01-21:
I have this [repo cloned](https://github.com/AndersKlint/openrazer-add-ornata-v3-tkl), which has one commit difference from this one. I'm trying to test out [#2183](https://github.com/openrazer/openrazer/pull/2183).

I'm using Ubuntu 23.10, and have all the dependencies installed. I am following these instructions: https://github.com/openrazer/openrazer/wiki/Building-a-package#ubuntu--linux-mint

```
openrazer-chroma-ornata-v3-tkl on  master [?] via 💎 v3.2.2 took 7ms
09:13:17 PM ➜ ./scripts/build_debs.sh
Temp DIR: /tmp/tmp.ItL8Vzy8YG_deb_build_tmp
Copying source to temporary build directory
Building deb packages

Failed to generate deb files. Check /tmp/tmp.ItL8Vzy8YG_deb_build_tmp/build.log for more details
```

Log file:

```
dpkg-buildpackage: info: source package openrazer
dpkg-buildpackage: info: source version 3.7.0-0
dpkg-buildpackage: info: source distribution jammy
dpkg-buildpackage: info: source changed by Luca Weiss <luca@z3ntu.xyz>
 dpkg-source --before-build .
dpkg-buildpackage: info: host architecture amd64
 fakeroot debian/rules clean
dh clean --with python3,dkms
   debian/rules override_dh_auto_clean
make[1]: Entering directory '/tmp/tmp.YB6yHqSeV9_deb_build_tmp/razer'
echo "No need to clean DKMS modules at packages build time."
No need to clean DKMS modules at packages build time.
echo "DKMS modules are built at packages installation time."
DKMS modules are built at packages installation time.
make[1]: Leaving directory '/tmp/tmp.YB6yHqSeV9_deb_build_tmp/razer'
   dh_clean
 dpkg-source -b .
dpkg-source: info: using source format '3.0 (quilt)'
dpkg-source: info: building openrazer using existing ./openrazer_3.7.0.orig.tar.gz
dpkg-source: info: building openrazer in openrazer_3.7.0-0.debian.tar.xz
dpkg-source: info: building openrazer in openrazer_3.7.0-0.dsc
 debian/rules build
dh build --with python3,dkms
   dh_update_autotools_config
   dh_autoreconf
   dh_auto_configure
   debian/rules override_dh_auto_build
make[1]: Entering directory '/tmp/tmp.YB6yHqSeV9_deb_build_tmp/razer'
echo "No need to build DKMS modules at packages build time."
No need to build DKMS modules at packages build time.
echo "DKMS modules are built at packages installation time."
DKMS modules are built at packages installation time.
make[1]: Leaving directory '/tmp/tmp.YB6yHqSeV9_deb_build_tmp/razer'
   dh_auto_test
   create-stamp debian/debhelper-build-stamp
 fakeroot debian/rules binary
dh binary --with python3,dkms
   dh_testroot
   dh_prep
   debian/rules override_dh_auto_install
make[1]: Entering directory '/tmp/tmp.YB6yHqSeV9_deb_build_tmp/razer'
/usr/bin/make ubuntu_install install-systemd UDEV_PREFIX=/ DESTDIR=/tmp/tmp.YB6yHqSeV9_deb_build_tmp/razer/debian/tmp
make[2]: Entering directory '/tmp/tmp.YB6yHqSeV9_deb_build_tmp/razer'
-e 
::[34m Installing DKMS files[0m
=====================================================
install -m 644 -v -D Makefile /tmp/tmp.YB6yHqSeV9_deb_build_tmp/razer/debian/tmp/usr/src/openrazer-driver-3.7.0/Makefile
install: creating directory '/tmp/tmp.YB6yHqSeV9_deb_build_tmp/razer/debian/tmp'
install: creating directory '/tmp/tmp.YB6yHqSeV9_deb_build_tmp/razer/debian/tmp/usr'
install: creating directory '/tmp/tmp.YB6yHqSeV9_deb_build_tmp/razer/debian/tmp/usr/src'
install: creating directory '/tmp/tmp.YB6yHqSeV9_deb_build_tmp/razer/debian/tmp/usr/src/openrazer-driver-3.7.0'
'Makefile' -> '/tmp/tmp.YB6yHqSeV9_deb_build_tmp/razer/debian/tmp/usr/src/openrazer-driver-3.7.0/Makefile'
install -m 644 -v -D install_files/dkms/dkms.conf /tmp/tmp.YB6yHqSeV9_deb_build_tmp/razer/debian/tmp/usr/src/openrazer-driver-3.7.0/dkms.conf
[... 54 lines cut ...]
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: option --install-layout not recognized
make[3]: *** [Makefile:25: ubuntu_install] Error 1
make[2]: *** [Makefile:68: ubuntu_daemon_install] Error 2
make[2]: Leaving directory '/tmp/tmp.YB6yHqSeV9_deb_build_tmp/razer'
make[1]: *** [debian/rules:19: override_dh_auto_install] Error 2
make[1]: Leaving directory '/tmp/tmp.YB6yHqSeV9_deb_build_tmp/razer'
make: *** [debian/rules:8: binary] Error 2
dpkg-buildpackage: error: fakeroot debian/rules binary subprocess returned exit status 2
```

I can't figure out what this `--install-layout` option is supposed to do. Can someone help me out here?

### Distribution

Ubuntu 23.10

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

lah7, 2024-01-21:
Looks like you've figured it out in #2198, Ubuntu 23.04 and newer now need a new `dk-dkms` package installed when building. Wiki updated.

Double checked in a Ubuntu 23.10 container, can confirm that branch builds and installs. All the best with testing!

closed 2024-01-21 by lah7

alichtman, 2024-01-21:
Unfortunately I filed those issues in the opposite order :) 

I have not been able to compile this successfully on Ubuntu yet

lah7, 2024-01-21:
Whoops, sorry! Feel free to re-open the issue.

For your convenience, here are the resulting builds from that repo (master branch). Built in a Ubuntu 23.04 Docker container: [openrazer-2183-debs.zip](https://github.com/openrazer/openrazer/files/14001965/openrazer-2183-debs.zip)

JeffreyLind22, 2024-06-05:
Just in case anyone else runs into this problem AND has a conda environment active, try disabling the conda environment, then running `./scripts/build_debs.sh `. Have a feeling there is some versioning / access problem with setuptools with pure pip v.s. conda pip.