openrazer/openrazer #2200: error messages in log: command timed out. status: 04 transaction_id.id: 1f

damnms, 2024-01-21:
### Distribution
debian trixie

### Support

- [ x] Have you tried rebooting?
- [ x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I have a pro click and a ultra type. The pro click should be supported, the keyboard not (yet).
In my dmesg i see those error(?) messages:
```
   32.359561] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 16, command_class: 00, command_id.id: 82 Params: 00000000000000000000000000000000 .
[   32.494572] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 16, command_class: 00, command_id.id: 82 Params: 00000000000000000000000000000000 .
[   32.514595] r8169 0000:0f:00.0 enp15s0: Link is Up - 1Gbps/Full - flow control rx/tx
[   32.629571] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 16, command_class: 00, command_id.id: 82 Params: 00000000000000000000000000000000 .
[   32.764562] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 16, command_class: 00, command_id.id: 82 Params: 00000000000000000000000000000000 .
[   32.899575] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 16, command_class: 00, command_id.id: 82 Params: 00000000000000000000000000000000 .
[   33.292562] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 07, command_class: 04, command_id.id: 05 Params: 01070807080000000000000000000000 .
[   33.328567] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 01, command_class: 00, command_id.id: 05 Params: 02000000000000000000000000000000 .
[   33.392561] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[   33.437561] usb 5-1.2: reset high-speed USB device number 4 using xhci_hcd

```
And i see very often the popup, that the battery has 0%. But its fully loaded and working. Not sure if its related and really a bug or if the device goes into sleep after a few seconds and then the popup appears. But i dont think so, because that would be some sort of ultra aggressive suspend.

z3ntu, 2024-02-09:
See https://github.com/openrazer/openrazer/issues/2203#issuecomment-1925411515

damnms, 2024-02-09:
> See [#2203 (comment)](https://github.com/openrazer/openrazer/issues/2203#issuecomment-1925411515)

thanks, i probably forgot to mention that i already set this to False

z3ntu, 2024-02-12:
If you've disabled the battery notifications in the config (correctly) then there should be no notifications popping up from OpenRazer.

About the 0% / "Command timed out" errors when the device is asleep, that's a different topic.

spikespaz, 2024-02-16:
https://github.com/openrazer/openrazer/pull/2160#issuecomment-1832451622

closed 2024-07-17 by github-actions[bot]