openrazer/openrazer #2201: Need to remove the reference of previous version of driver in dkms.conf

PlamenStilyianov, 2024-01-27:
(base) orajava@suse15l:~> sudo ls -lart /var/lib/dkms/openrazer-driver/
total 20
lrwxrwxrwx 1 root root  40 Jul 28 12:41 kernel-5.14.21-150500.55.7-default-x86_64 -> 3.6.1/5.14.21-150500.55.7-default/x86_64
lrwxrwxrwx 1 root root  41 Aug 15 10:18 kernel-5.14.21-150500.55.19-default-x86_64 -> 3.6.1/5.14.21-150500.55.19-default/x86_64
drwxr-xr-x 1 root root  60 Sep  7 10:29 ..
drwxr-xr-x 1 root root 122 Nov  7 23:46 3.6.1
lrwxrwxrwx 1 root root  41 Nov  7 23:48 kernel-5.14.21-150500.55.28-default-x86_64 -> 3.7.0/5.14.21-150500.55.28-default/x86_64
lrwxrwxrwx 1 root root  35 Nov  7 23:55 kernel-6.4.0-150600.1-default-x86_64 -> 3.7.0/6.4.0-150600.1-default/x86_64
drwxr-xr-x 1 root root 156 Jan 27 01:04 3.7.0
lrwxrwxrwx 1 root root  35 Jan 27 01:04 kernel-6.4.0-150600.4-default-x86_64 -> 3.7.0/6.4.0-150600.4-default/x86_64
drwxr-xr-x 1 root root 414 Jan 27 01:04 .
(base) orajava@suse15l:~> uname -r
6.4.0-150600.4-default
(base) orajava@suse15l:~> **sudo dkms status**
**_Error! Could not locate dkms.conf file._**
File: /var/lib/dkms/openrazer-driver/3.6.1/source/dkms.conf does not exist.

z3ntu, 2024-02-08:
Normally the hooks in the package manager should do that I think. Do you have any other DKMS modules installed maybe that behave correctly? Hard to say what openSUSE expects a package to do here.

nathancurry, 2024-03-02:
I have this issue after a Fedora upgrade 38 > 39 > 40 on a system I hadn't used since last Summer.  It seems to be the result of uninstalling after the upgrade.

I believe packages were installed per instructions here:
https://openrazer.github.io/#fedora
 
I'm still seeing valid symlinks to an old kernel:

```
root@eeyore:~# ls -lart /var/lib/dkms/openrazer-driver/
total 12
lrwxrwxrwx. 1 root root  35 Jun  2  2023 kernel-6.2.14-300.fc38.x86_64-x86_64 -> 3.6.1/6.2.14-300.fc38.x86_64/x86_64
lrwxrwxrwx. 1 root root  34 Jun  2  2023 kernel-6.3.4-201.fc38.x86_64-x86_64 -> 3.6.1/6.3.4-201.fc38.x86_64/x86_64
lrwxrwxrwx. 1 root root  34 Jun 10  2023 kernel-6.3.6-200.fc38.x86_64-x86_64 -> 3.6.1/6.3.6-200.fc38.x86_64/x86_64
drwxr-xr-x. 1 root root  60 Jan 23 19:00 ..
drwxr-xr-x. 1 root root 222 Feb 19 06:49 .
drwxr-xr-x. 1 root root 140 Feb 19 06:49 3.6.1
```

Here's the dnf transaction:

```
root@eeyore:~# dnf history info 56
Transaction ID : 56
Begin time     : Mon 19 Feb 2024 06:49:11 AM EST
Begin rpmdb    : f295e4e1fcbc0220fd2dc5e6ef86d3b97a223f757ccc8dbdaff1b043b81b8e1c
End time       : Mon 19 Feb 2024 06:49:37 AM EST (26 seconds)
End rpmdb      : 2f9136328241955e0b2314a812ff187f87bbd4f82bea440cdd261c995574a1e9
User           : nc <nc>
Return-Code    : Success
Releasever     : 38
Command Line   : remove openrazer*
Comment        : 
Packages Altered:
    Removed openrazer-daemon-3.6.1-1.1.noarch              @@System
    Removed openrazer-kernel-modules-dkms-3.6.1-1.1.noarch @@System
    Removed openrazer-meta-3.6.1-1.1.noarch                @@System
    Removed python3-openrazer-3.6.1-1.1.noarch             @@System
Scriptlet output:
   1 Module openrazer-driver-3.6.1 for kernel 6.3.8-200.fc38.x86_64 (x86_64).
   2 Before uninstall, this module version was ACTIVE on this kernel.
   3 
   4 razerkbd.ko.xz:
   5  - Uninstallation
   6    - Deleting from: /lib/modules/6.3.8-200.fc38.x86_64/extra/
   7  - Original module
   8    - No original module was found for this module on this kernel.
   9    - Use the dkms install command to reinstall any previous module version.
  10 
  11 razermouse.ko.xz:
  12  - Uninstallation
  13    - Deleting from: /lib/modules/6.3.8-200.fc38.x86_64/extra/
  14  - Original module
  15    - No original module was found for this module on this kernel.
  16    - Use the dkms install command to reinstall any previous module version.
  17 
  18 razerkraken.ko.xz:
  19  - Uninstallation
  20    - Deleting from: /lib/modules/6.3.8-200.fc38.x86_64/extra/
  21  - Original module
  22    - No original module was found for this module on this kernel.
  23    - Use the dkms install command to reinstall any previous module version.
  24 
  25 razeraccessory.ko.xz:
  26  - Uninstallation
  27    - Deleting from: /lib/modules/6.3.8-200.fc38.x86_64/extra/
  28  - Original module
  29    - No original module was found for this module on this kernel.
  30    - Use the dkms install command to reinstall any previous module version.
  31 depmod.....
  32 Module openrazer-driver-3.6.1 for kernel 6.4.9-200.fc38.x86_64 (x86_64).
[... 54 lines cut ...]
  87 razeraccessory.ko.xz:
  88  - Uninstallation
  89    - Deleting from: /lib/modules/6.7.4-100.fc38.x86_64/extra/
  90  - Original module
  91    - No original module was found for this module on this kernel.
  92    - Use the dkms install command to reinstall any previous module version.
  93 depmod.....
```

I was able to get the dkms cli utility to run normally by downloading the 3.6.1 tarball and placing the dkms.conf file in the indicated location.

More output:

```
root@eeyore:~# dkms status
openrazer-driver/3.6.1: added
root@eeyore:~# dkms remove -m openrazer-driver/3.6.1 --all --force --verbose
root@eeyore:~# dkms uninstall -m openrazer-driver/3.6.1 --all --force --verbose
root@eeyore:~# dkms status --verbose
openrazer-driver/3.6.1: added
```

I force installed 3.7.0 by downloading the rpm and running `rpm -ihv --nodeps --nofiledigest $rpms`, and it then shows up in the `dkms status` output.  When I removed those rpms, 3.7.0 no longer appears in the `dkms status` output.

I would prefer to clean this up, but it doesn't seem to be a pressing issue at the moment.

closed 2024-08-01 by github-actions[bot]