openrazer/openrazer #2202: RazerGenie can't find daemon

Vulpes-Nix, 2024-02-02:
### Distribution
Fedora 39
BlackWidow v4
RazerViper Ultimate

### Support

- [ x] Have you tried rebooting?
- [ x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?
- [ x] Tried problem solving suggestions from https://github.com/openrazer/openrazer/issues/1345

I've went through the installation steps, I think I've set the headers correctly and the OpenRazer version is also updated, yet for some reason applications like RazerGenie tell me that the OpenRazer daemon is not installed or too old. 

Other applications are also not able to recognize the M1-6 keys ether. 

**`lsusb | grep 1532`**
Bus 007 Device 003: ID 1532:007b Razer USA, Ltd RC30-0305 Gaming Mouse Dongle [Viper Ultimate (Wireless)]
Bus 007 Device 004: ID 1532:0287 Razer USA, Ltd Razer BlackWidow V4

1532:0287 does not appear [in this list](https://github.com/openrazer/openrazer/tree/stable#device-support) but [the patchnotes for 3.7.0 ](https://github.com/openrazer/openrazer/releases/tag/v3.7.0)state that you added support for the BlackWidow v4. 

Maybe I got too excited too fast?

**`sudo dkms install openrazer-driver/3.7.0`**
Module openrazer-driver/3.7.0 already installed on kernel 6.6.13-200.fc39.x86_64 (x86_64), skip. You may override by specifying --force.

**`sudo modprobe razerkbd`**
_actually shows nothing in the console_

**`openrazer-daemon -Fv`**

Starting daemon.
2024-02-02 15:47:23 | razer                          | CRITICAL | User is not a member of the plugdev group
2024-02-02 15:47:23 | razer                          | CRITICAL | Please run the command 'sudo gpasswd -a $USER plugdev' and then reboot!
Stopping daemon.
Exception ignored in atexit callback: <bound method Daemonize.exit of <daemonize.Daemonize object at 0x7f33ca9f6ff0>>
Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/daemonize.py", line 72, in exit
    sys.exit(0)
SystemExit: 0

Help is much appreciated and also I'm kinda new to Fedora and Linux in general. 
Big thanks in advance.

lah7, 2024-02-02:
> 2024-02-02 15:47:23 | razer | CRITICAL | User is not a member of the plugdev group

This is the culprit. To fix, open a Terminal and run:

    sudo gpasswd -a $USER plugdev

Then log out or reboot the system for the change to take effect.

This step is needed because of how the driver works. Since  _[everything is a file](https://en.wikipedia.org/wiki/Everything_is_a_file)_ in Linux, the OpenRazer daemon will be able to read/write to OpenRazer's driver files if the user have a group in common, in this case, the  `plugdev` group.

Afterwards, RazerGenie should work with your Viper Ultimate mouse. Unfortunately, your BlackWidow V4 `1532:0287` isn't supported yet (#2163). Apologies, it seems there was an outsight in the release notes since the commit 1278b22b67e0288c58cda9a1eb1145ef3a3f6c85 missed out the "Pro" part. The README is correct:

> Razer BlackWidow V4 Pro 	1532:028D

I'll amend the release notes, even though it's a little late now (news outlets may have copied and pasted this list)

Vulpes-Nix, 2024-02-02:
> > 2024-02-02 15:47:23 | razer | CRITICAL | User is not a member of the plugdev group
> 
> This is the culprit. To fix, open a Terminal and run:
> 
> ```
> sudo gpasswd -a $USER plugdev
> ```
> 
> Then log out or reboot the system for the change to take effect.
> 
> This step is needed because of how the driver works. Since _[everything is a file](https://en.wikipedia.org/wiki/Everything_is_a_file)_ in Linux, the OpenRazer daemon will be able to read/write to OpenRazer's driver files if the user have a group in common, in this case, the `plugdev` group.
> 
> Afterwards, RazerGenie should work with your Viper Ultimate mouse. Unfortunately, your BlackWidow V4 `1532:0287` isn't supported yet (#2163). Apologies, it seems there was an outsight in the release notes since the commit [1278b22](https://github.com/openrazer/openrazer/commit/1278b22b67e0288c58cda9a1eb1145ef3a3f6c85) missed out the "Pro" part. The README is correct:
> 
> > Razer BlackWidow V4 Pro 	1532:028D
> 
> I'll amend the release notes, even though it's a little late now (news outlets may have copied and pasted this list)

Thank you so much for the quick reply and it absolutely worked.
I'll keep my eyes peeled for the v4 to be added some time in the future. 

In my eyes this ticket can be closed. Thanks again!

closed 2024-02-03 by lah7