openrazer/openrazer #2204: Target not found. Openrazer Daemon install SteamOS

GodPhase777, 2024-02-03:
Openrazor-Daemon

- [Yes] Have you tried rebooting?
- [Yes] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Install command on page doesn't work.
[image]

Sorry if this isn't a real issue, I'm fairly new to linux and need a bit of help.

lah7, 2024-02-03:
Steam Deck runs Arch Linux, so try running this command first:

    sudo pacman -Sy

This command will syncs repositories. Afterwards, try `sudo pacman -S openrazer-daemon` again.

If it still says `target not found`, it's likely that the repository for the OpenRazer package is disabled, which is `extra`. This can be fixed by editing `/etc/pacman.conf` and removing the `#` for these lines (or similiar):

```
[extra]
Include = /etc/pacman.d/mirrorlist
```

GodPhase777, 2024-02-04:
Did something different but still an error.

[image]

z3ntu, 2024-02-04:
To my knowledge Steam OS doesn't build openrazer in their repos, but you should be able to find guides online how to install it on SteamOS by building it yourself

lah7, 2024-02-04:
Take a look at this Reddit comment, these instructions are more recent and should do the trick for SteamOS:

https://www.reddit.com/r/SteamDeck/comments/16k0tap/comment/k0uqvx3/

I noticed there are a lot of older guides that mention the AUR but might not work any more.

GodPhase777, 2024-02-04:
> Take a look at this Reddit comment, these instructions are more recent and should do the trick for SteamOS:
> 
> https://www.reddit.com/r/SteamDeck/comments/16k0tap/comment/k0uqvx3/
> 
> I noticed there are a lot of older guides that mention the AUR but might not work any more.

[image]

[image]

Doesn't seem to work either, can't get passed 5. Can't seem to do -S or-Sy at all without some annoying error like this.

lah7, 2024-02-04:
Be sure to undo the change made in `/etc/pacman.conf` suggested by the earlier comment - either delete the following lines or add a `#` at the start of them:

    [extra]
    Include = /etc/pacman.d/mirrorlist

Then you should be able to run `sudo pacman -Sy` and resume from step 5. SteamOS calls it `extra-3.5` instead of `extra` as it's known in regular Arch Linux. I didn't know, my bad.

You definitely need step 5 - to install `linux-neptune-61-headers`. Otherwise, the driver won't install properly. It's stuck at the moment as we are telling it there's a non-existent `extra` repository.

lah7, 2024-02-04:
Also, just checking the instructions. The latest kernel [in their repository](https://steamdeck-packages.steamos.cloud/archlinux-mirror/jupiter-main/os/x86_64/) for an up-to-date Steam Deck is now `linux-neptune-65-headers`, use that package.

If ever in doubt, this lists the current running kernel version:

    uname -r

This would list kernel packages currently installed:

    pacman -Q | grep linux-neptune

GodPhase777, 2024-02-05:
> Also, just checking the instructions. The latest kernel [in their repository](https://steamdeck-packages.steamos.cloud/archlinux-mirror/jupiter-main/os/x86_64/) for an up-to-date Steam Deck is now [...]

I'm still new to anything linux, could you explain this a bit simpler for me?

GodPhase777, 2024-02-05:
> Be sure to undo the change made in `/etc/pacman.conf` suggested by the earlier comment - either delete the following lines or add a `#` at the start of them:
> 
> ```
> [extra]
> Include = /etc/pacman.d/mirrorlist
> ```
> 
> Then you should be able to run `sudo pacman -Sy` and resume from step 5. SteamOS calls it `extra-3.5` instead of `extra` as it's known in regular Arch Linux. I didn't know, my bad.
> 
> You definitely need step 5, to install the Linux headers. Otherwise, the driver won't install properly. It's stuck at the moment as we are telling it there's a non-existent `extra` repository.

Seems that worked, I'll move on to the next steps now.

lah7, 2024-02-05:
Not many devices in OpenRazer support over Bluetooth, but it _should_ support over 2.4GHz dongle or wired. That said, I don't know specifically which device you have - this command will tell us:

    lsusb | grep 1532

 For example:

    Bus 001 Device 006: ID 1532:0214 Razer USA, Ltd BlackWidow Ultimate 2016
                           ^^^^^^^^^

To know for sure,  [check the ID against the README file](https://github.com/openrazer/openrazer/tree/stable?tab=readme-ov-file#device-support). If that specific ID is not in the list, seeing "No devices" is normal for RazerGenie. This USB ID might change depending on how it's connected. It might not appear at all over Bluetooth but I'm not 100% certain as I don't have any Bluetooth Razer stuff.

----

> Should I be concerned about this error after installing the three packages?

Yep, it's not installed correctly. Foreshadowing the 5 months old instructions in [that Reddit comment](https://www.reddit.com/r/SteamDeck/comments/16k0tap/comment/k0uqvx3/), there might be a newer 🐧  kernel version.

It's because to install this kind of driver (a DKMS module), the right "headers" need to be installed that match the version of the Linux kernel. 🐧 ⚙️  So, if SteamOS provides the Linux kernel in a package named `linux-neptune-61`, you'll need `linux-neptune-61-headers`. (In that case, my comment about needing `65` is incorrect, sorry)

We can find out what kernel and package is installed with these commands:

    uname -r 
    pacman -Q | grep linux-neptune

Installing the headers package should automatically build the driver for the kernel. We'll know for sure with this command:

    sudo dkms status

    openrazer-driver/3.7.0, 6.7.3-arch1-2, x86_64: installed    (for example)

GodPhase777, 2024-02-05:
[image]
[image]
First command works fine, second displays nothing.

lah7, 2024-02-06:
I get the impression that SteamOS has a read only mode for system files. You may need to toggle it on/off as we work through this.

```bash
sudo steamos-readonly disable    # when making modifications
sudo steamos-readonly enable     # when finished
```

So, we have the kernel packages we need 👍🏼  This is what the `ERROR: Missing [...] kernel modules tree` was complaining about earlier.  The driver ("DKMS module") now just needs to be built, Try:

    sudo dkms install openrazer-driver/3.7.0

If it's successful, restart the Steam Deck. For best chances, try using the mouse over a wired or 2.4 GHz connection first, so we can be sure it is working.

GodPhase777, 2024-02-07:
> I get the impression that SteamOS has a read only mode for system files. You may need to toggle it on/off as we work through this.
> 
> ```shell
> sudo steamos-readonly disable    # when making modifications
> sudo steamos-readonly enable     # when finished
> ```
> 
> So, we have the kernel packages we need 👍🏼 This is what the `ERROR: Missing [...] kernel modules tree` was complaining about earlier. The driver ("DKMS module") now just needs to be built, Try:
> 
> ```
> sudo dkms install openrazer-driver/3.7.0
> ```
> 
> If it's successful, restart the Steam Deck. For best chances, try using the mouse over a wired or 2.4 GHz connection first, so we can be sure it is working.

[image]
The system was in read write mode already. The command just errors out.

lah7, 2024-02-07:
Let's have a look:

## ❓ Modules won't be signed

From what I looked up, Steam Deck doesn't have [Secure Boot](https://en.wikipedia.org/wiki/UEFI#Secure_Boot) enabled, so this message seems **harmless**. If secure boot was enabled, the driver still wouldn't load because it's unsigned. It's a complicated process to sign & install certificates onto hardware, so disabling secure boot is the easier option.

I don't think we need to do anything - but just wanted to point it out in case.

## ⚠️ Kernel headers cannot be found

That's odd since the package for them was installed earlier. Try reinstalling:

    sudo pacman -S linux-neptune-61-headers

Reinstalling should trigger the `dkms install` command, similar to the output of this [other Reddit comment](https://www.reddit.com/r/SteamDeck/comments/17lj4j6/comment/k7ejqwu/) (except hopefully, without the error)

There should be a `6.1.52-value-9-1-neptune` (or similar) folder here:

    ls /usr/lib/modules/

## ℹ️ SteamOS Updates will revert changes

Should point this out, since we're modifying files on an immutable root filesystem. That might be why we're not having much luck. So, if SteamOS updates, the changes get wiped out and need repeating - until the day OpenRazer becomes driverless.

----

Seems like we're close - it's just the driver that's not working. An alternate method could be to try this blog's script - the comments indicate the post was updated in the past month:

https://daniele.tech/2022/12/how-to-get-openrazer-working-easily-on-steam-deck/

GodPhase777, 2024-02-08:
> Seems like we're close - it's just the driver that's not working. An alternate method could be to try this blog's script - the comments indicate the post was updated in the past month:
> 
> https://daniele.tech/2022/12/how-to-get-openrazer-working-easily-on-steam-deck/

Funny you picked that post. It was one of the first things I tried before asking on Git. Just errored out a bunch.

GodPhase777, 2024-02-16:
Any update? Do you know how to get it fully working?

closed 2024-07-17 by github-actions[bot]

Mte90, 2024-09-25:
I am the author of the guide to install openrazer on the steam deck with a script that does everything (https://daniele.tech/2022/12/how-to-get-openrazer-working-easily-on-steam-deck/) and I am always working on updating.
Let me know if it works for you!