openrazer/openrazer #2205: Bluetooth peripherals not recognized

Kaukov, 2024-02-08:
### Distribution

### Support

- [x ] Have you tried rebooting?
- [ x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I have a Naga Pro and a BlackWidow V3 Mini. Both work flawlessly when connected via cable or via their respective dongles.

However, when I connect any of them via bluetooth, OpenRazer no longer sees them and can't control them at all. This means that I can't control their backlight, effects, or synchronize them with my other Razer devices.

`lsusb` doesn't display any of the bluetooth-connected devices. I think this is normal since they're not connected via USB at all.

I have a Windows partition on the same laptop and I can debug anything you need, I just need instructions on how to do it.

Running Gentoo with kernel 6.7.3 currently. Kernel modules and everything is working except for this issue.

z3ntu, 2024-02-12:
> lsusb doesn't display any of the bluetooth-connected devices. I think this is normal since they're not connected via USB at all.

For most devices they show up there, according to https://github.com/openrazer/openrazer/issues/1544#issuecomment-872965040 also the BlackWidow V3 Mini Hyperspeed should show up there.

Perhaps you're missing some kernel config option (since you're running Gentoo)?

z3ntu, 2024-02-12:
Check that you've enabled CONFIG_UHID and the module is loaded.

Kaukov, 2024-02-12:
> Check that you've enabled CONFIG_UHID and the module is loaded.

Here's what I got:
```sh
~ > zgrep -i uhid /proc/config.gz
CONFIG_UHID=m
~ > lsmod | grep -i uhid
uhid                   16384  1
```

And the devices are still not visible via OpenRazer or `lsusb`.

I also found the module `btusb` but it was loaded by default:
```sh
~ > lsmod | grep -i btusb
btusb                  73728  0
btrtl                  28672  1 btusb
btintel                49152  1 btusb
btbcm                  20480  1 btusb
btmtk                  12288  1 btusb
bluetooth             884736  48 btrtl,btmtk,btintel,btbcm,bnep,btusb,rfcomm
```

z3ntu, 2024-03-03:
Might be worth checking on e.g. an Ubuntu livecd image, connect the Bluetooth mouse there and check if it shows up in lsusb. I don't own any Razer Bluetooth devices myself so it's difficult to judge how it really behaves in practice.

Kaukov, 2024-03-03:
`lsusb` doesn't see the bluetooth-connected device. But `bluetoothctl devices` shows it.

Kaukov, 2024-04-05:
I'm currently using another laptop with OpenSUSE Tumbleweed on it. Testing with the Razer Naga Pro mouse.

When connecting via bluetooth, the mouse appears in `lsusb` and is shown in Polychromatic as Razer Naga Pro (Wireless), but the battery is shown as 0% and I can't control any of the mouse's properties like brightness, polling, etc.

Is this a Polychromatic issue or an OpenRazer one?

z3ntu, 2024-04-06:
I just saw there's `HID_BLUETOOTH_DEVICE` in the kernel sources, so maybe that's somehow useful for matching on 1532:0092 for example. Also pretty sure e.g. `razer_get_usb_response` in the driver wouldn't work on Bluetooth, not sure what other code we'd need to have there instead. Udev rules actually look like they could/should also be able to match on these Bluetooth devices.

In short, somebody with such a device in hand will need to try to hack on it and add some code for it. I can't do it without such device.

hockeymikey, 2025-01-03:
I have the orochi 2015 (or 2016?) if that helps to figure this out.