openrazer/openrazer #2206: Hyperspeed Multi-Device - second device not visible

Kaukov, 2024-02-08:
### Distribution

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I have a BlackWidow V3 Mini and a Naga Pro. On Windows, I was able to pair them both to a single dongle. And Synapse sees them as separate devices, I can control them separately, or sync them to my other devices.

OpenRazer doesn't see the second device though. Doesn't matter if I pair the mouse to the keyboard's dongle, or vice-versa, the device whose dongle it isn't, isn't visible in OpenRazer and can't be controlled in any way.

I currently have the keyboard paired to the mouse's dongle because when it's the opposite, the mouse loses the profile functionality on 2.4GHz mode and I can't switch profiles. This is a Razer issue, not an OpenRazer one.

Here's the output of `lsusb`. It correctly shows only the mouse and the keyboard is nowhere to be seen:

```
~ > lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 1532:0090 Razer USA, Ltd Razer Naga Pro
Bus 003 Device 003: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 003 Device 005: ID 04f2:b75c Chicony Electronics Co., Ltd FHD Webcam
Bus 003 Device 006: ID 8087:0026 Intel Corp. AX201 Bluetooth
Bus 003 Device 007: ID 0fce:320d Sony Ericsson Mobile Communications AB XQ-DQ54
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 004 Device 002: ID 05e3:0620 Genesys Logic, Inc. GL3523 Hub
Bus 004 Device 003: ID 0bda:0316 Realtek Semiconductor Corp. Card Reader
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 005 Device 003: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 005 Device 004: ID 1532:007e Razer USA, Ltd RC30-030502 Mouse Dock
Bus 005 Device 005: ID 1532:48f0 Razer USA, Ltd Razer Base Station V2 Chroma
Bus 005 Device 006: ID 1532:0f20 Razer USA, Ltd Razer Base Station V2 Chroma
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 006 Device 002: ID 05e3:0620 Genesys Logic, Inc. GL3523 Hub
Bus 006 Device 003: ID 090c:2000 Silicon Motion, Inc. - Taiwan (formerly Feiya Technology Corp.) USB DISK
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 002: ID 1532:0520 Razer USA, Ltd Kraken Tournament Edition
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 009 Device 002: ID 1532:0f1a Razer USA, Ltd Core X Chroma
Bus 010 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 010 Device 002: ID 0b95:1790 ASIX Electronics Corp. AX88179 Gigabit Ethernet
```

And only Razer devices:
```
~ > lsusb | grep 1532
Bus 003 Device 002: ID 1532:0090 Razer USA, Ltd Razer Naga Pro
Bus 005 Device 004: ID 1532:007e Razer USA, Ltd RC30-030502 Mouse Dock
Bus 005 Device 005: ID 1532:48f0 Razer USA, Ltd Razer Base Station V2 Chroma
Bus 005 Device 006: ID 1532:0f20 Razer USA, Ltd Razer Base Station V2 Chroma
Bus 007 Device 002: ID 1532:0520 Razer USA, Ltd Kraken Tournament Edition
Bus 009 Device 002: ID 1532:0f1a Razer USA, Ltd Core X Chroma
```

Kaukov, 2024-02-10:
An update:

It seems that even though the mouse is recognized, I cannot control any of its properties. When I try to change the brightness or its effect, nothing happens with the mouse. It cannot sync to other peripherals and is acting just like the keyboard.

z3ntu, 2024-03-03:
So can we close this issue?

Kaukov, 2024-03-03:
The original issue is still there - the device whose dongle isn't used isn't shown up in `lsusb` or OpenRazer and therefore can't be controlled/synced.

I just tested on Windows and Razer Synapse sees all devices and reports the battery state, BUT it seems that on the keyboard the RGB lighting control is broken and I can't change the backlight, Chroma doesn't work, etc. So there's an underlying issue with Razer Synapse as well. I'll probably report this issue to Razer directly and close this issue until they do something about it. If I connect the keyboard via USB, it works as expected on both Linux and Windows.

closed 2024-03-03 by Kaukov