openrazer/openrazer #2207: Razer Thunderbolt 4 Chroma dock not recognized

Calavera00, 2024-02-09:
### Distribution

Razer Blade 15 (Mid 2019) Mercury
Linux Mint 21.3 x86_64
Kernel: 6.5.0-17-generic

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I'm using a Razer blade 15 Mercury and just got a Razer Thunderbolt 4 Chroma dock. I am able to use the monitor connected to the dock, but apart from the laptop itself, openrazer does not recognize any connected devices like the razer wireless mouse charging dock, and the lights don't work on the thunderbolt dock. I had exactly the same problem on Windows until I installed thunderbolt 3 drivers, so I am suspecting something similar is the problem for me on Linux.

I ran through the troubleshooter but couldn't find any issues. Any advice or suggestions for what I can try next would be greatly appreciated!

```
$ lsusb | grep -i 1532
Bus 001 Device 004: ID 1532:0245 Razer USA, Ltd Blade 15 (Mid 2019) Mercury
``` 

```
$ ls -las /sys/bus/hid/drivers  
total 0
0 drwxr-xr-x 6 root root 0 Feb  8 18:20 .
0 drwxr-xr-x 4 root root 0 Feb  8 18:20 ..
0 drwxr-xr-x 2 root root 0 Feb  8 18:21 apple
0 drwxr-xr-x 2 root root 0 Feb  8 18:20 hid-generic
0 drwxr-xr-x 2 root root 0 Feb  8 18:20 hid-multitouch
0 drwxr-xr-x 2 root root 0 Feb  8 18:20 razerkbd
```

```
cat ~/.local/share/openrazer/logs/razer.log
2024-02-08 18:17:18 | razer                          | INFO     | Stopping daemon on signal 15
2024-02-08 18:18:04 | razer                          | INFO     | Initialising Daemon (v3.7.0). Pid: 2038
2024-02-08 18:18:04 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2024-02-08 18:18:04 | razer                          | INFO     | Found device.0: 0003:1532:0245.0004
2024-02-08 18:18:04 | razer                          | CRITICAL | Could not access /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0245.0004/device_type, file is not owned by plugdev
2024-02-08 18:18:04 | razer                          | INFO     | Serving DBus
2024-02-08 18:19:08 | razer                          | INFO     | Stopping daemon on signal 15
2024-02-08 18:19:11 | razer                          | INFO     | Initialising Daemon (v3.7.0). Pid: 4427
2024-02-08 18:19:11 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2024-02-08 18:19:11 | razer                          | INFO     | Found device.0: 0003:1532:0245.0004
2024-02-08 18:19:11 | razer                          | CRITICAL | Could not access /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0245.0004/device_type, file is not owned by plugdev
2024-02-08 18:19:11 | razer                          | INFO     | Serving DBus
2024-02-08 18:20:26 | razer                          | INFO     | Stopping daemon on signal 15
2024-02-08 18:21:08 | razer                          | INFO     | Initialising Daemon (v3.7.0). Pid: 2037
2024-02-08 18:21:08 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2024-02-08 18:21:08 | razer                          | INFO     | Found device.0: 0003:1532:0245.0004
2024-02-08 18:21:08 | razer.device0                  | INFO     | Initialising device.0 RazerBladeMid2019Mercury
2024-02-08 18:21:08 | razer.device0                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2024-02-08 18:21:08 | razer                          | INFO     | Serving DBus

```

z3ntu, 2024-02-09:
The dock should show up in lsusb as `1532:0F21` (or some other `1532:xxxx` value), otherwise OpenRazer can't do anything with the device.

Maybe check dmesg output what messages appear when you connect the dock?

Calavera00, 2024-02-09:
oh interesting, dmesg actually seem to find the dock when I connect it. I wonder why it doesn't recognize it in Polychromatic or RazerGenie, any ideas?
```
[  +0.059545] wlan0: Limiting TX power to 21 (24 - 3) dBm as advertised by 9c:0b:05:7a:a7:86
[Feb 9 07:00] workqueue: pm_runtime_work hogged CPU for >10000us 4 times, consider switching to WQ_UNBOUND
[  +4.157308] thunderbolt 0-1: new device found, vendor=0x127 device=0x8
[  +0.000004] thunderbolt 0-1: Razer Thunderbolt 4 Dock
```
Connecting my Deathadder mouse over bluetooth, it shows up at DA V2 and works just fine but also doesn't show up in Polychromatic or RazerGenie. 
```
[Feb 9 07:06] input: DA V2 Pro Mouse as /devices/virtual/misc/uhid/0005:1532:008E.0008/input/input40
[  +0.000269] input: DA V2 Pro Consumer Control as /devices/virtual/misc/uhid/0005:1532:008E.0008/input/input41
[  +0.000118] input: DA V2 Pro System Control as /devices/virtual/misc/uhid/0005:1532:008E.0008/input/input42
[  +0.000122] input: DA V2 Pro as /devices/virtual/misc/uhid/0005:1532:008E.0008/input/input43
[  +0.000084] input: DA V2 Pro Keyboard as /devices/virtual/misc/uhid/0005:1532:008E.0008/input/input44
[  +0.000053] hid-generic 0005:1532:008E.0008: input,hidraw4: BLUETOOTH HID v0.01 Mouse [DA V2 Pro] on 3c:f0:11:da:43:db
```

z3ntu, 2024-02-12:
The Thunderbolt dock is also supposed to show up in lsusb, not sure why it doesn't do that for you.

With the DeathAdder connected, could you also run `lsusb | grep 1532` again?

Calavera00, 2024-02-13:
lsusb shows the same device as Polychromatic (The laptop itself only) even though I am using the Deathadder to move the cursor and the Thunderbolt Dock to use my two monitors.
```
$ lsusb | grep 1532                                                                                                                                                                                                                                                                                                                                                                 
Bus 001 Device 004: ID 1532:0245 Razer USA, Ltd Blade 15 (Mid 2019) Mercury
```
The razer mouse charger seems dead though even though it's connected to the Thunderbolt dock. The lights are off and it doesn't seem to be charging the DA mouse. I had the same problem on Windows until I installed the correct Thunderbolt drivers. Is there a way to do a clean uninstall/reinstall of openrazer and could it make a difference?

z3ntu, 2024-03-03:
OpenRazer shouldn't influence the Thunderbolt functionality at all, it just operates on the USB device already discovered by the kernel and uses that for its functionality (which is mostly limited to RGB settings and things like that).

So feels more like something more in general (not openrazer-related) seems to be not working for you.

Calavera00, 2024-03-03:
Thanks for pointing me in this direction, I randomly ran into a solution. I was trying out Gnome desktop environment, and when I connected the thunderbolt connection while running Gnome, I got a popup telling me to approve the connection. Turns out that Gnome comes with Thunderbolt settings as part of the DE but I couldn't find anything like it in Cinnamon. After opening the thunderbolt settings and approving the connection, everything works as expected and I now see the connected devices in RazerGenie, it even works back in Cinnamon. I was suspecting something like this because I went through a very similar experience on Windows.

Since it seems to be a system wide setting, I wonder if there's a way to troubleshoot this with CLI commands. ChatGPT tells me I can use `boltctl list` to display connected devices and then `botlctl enroll <device id>`. I might try that next time.

closed 2024-03-03 by Calavera00