openrazer/openrazer #2210: modprobe: ERROR: could not insert 'razerkbd': Invalid argument

kx9080, 2024-02-27:
Hello, I have been encountering this error with the Kernel Module. I have tried all of the troubleshooting steps and the only logs to return anything were the razer_mount log which returned 
`Feb 26 18:34:26 caden-MS-7B86 razer_mount[770]: Driver razermouse
Feb 26 18:34:26 caden-MS-7B86 razer_mount[771]: Device_ID 0003:1532:009C.0004
Feb 26 18:34:26 caden-MS-7B86 razer_mount[774]: Modprobing razermouse
Feb 26 18:34:26 caden-MS-7B86 systemd-udevd[572]: 0003:1532:009C.0004: Process 'razer_mount razermouse 0003:1532:009C.0004' failed with exit code 1.
`
Daemon log which returns this over and over again
 `2024-02-26 18:36:00 | razer                          | DEBUG    | DBus called get_serial_list
`
I am using Linux Mint 21.3 and Razer Kernel v3.7.0 with polychromatic and the Razer Deathadder V2 x Hyperspeed

z3ntu, 2024-02-27:
Did you check our Troubleshooting page? https://github.com/openrazer/openrazer/wiki/Troubleshooting

kx9080, 2024-02-27:
Yes I did, my secure boot is off and every other step returned nothing other than the logs and modprobe. Polychromatic knows that openrazer is installed and confirms that the kernel driver exists.

lah7, 2024-02-27:
Please try this command and let us know the output. It should print each line of the `razer_mount` script and might reveal which stage causes it to exit with an error.

    sudo bash -xe /usr/lib/udev/razer_mount razermouse 0003:1532:009C.0004

kx9080, 2024-02-28:
> Please try this command and let us know the output. It should print each line of the `razer_mount` script and might reveal which stage causes it to exit with an error.
> 
> ```
> sudo bash -xe /usr/lib/udev/razer_mount razermouse 0003:1532:009C.0004
> ```

It returns this
`
+ set -e
+ PATH=/sbin:/bin:/usr/sbin:/usr/bin
+ '[' -x /usr/bin/logger ']'
+ LOGGER=/usr/bin/logger
+ '[' -t 1 -a -z /usr/bin/logger ']'
+ '[' '!' -e /dev/log ']'
+ '[' -t 1 ']'
+ DRIVER=razermouse
+ DEVICE_ID=0003:1532:009C.0004
+ mesg 'Driver razermouse'
+ echo 'Driver razermouse'
Driver razermouse
+ /usr/bin/logger -t 'razer_mount[16509]' 'Driver razermouse'
+ mesg 'Device_ID 0003:1532:009C.0004'
+ echo 'Device_ID 0003:1532:009C.0004'
Device_ID 0003:1532:009C.0004
+ /usr/bin/logger -t 'razer_mount[16509]' 'Device_ID 0003:1532:009C.0004'
+ '[' '!' -d /sys/bus/hid/drivers/razermouse ']'
+ mesg 'Modprobing razermouse'
+ echo 'Modprobing razermouse'
Modprobing razermouse
+ /usr/bin/logger -t 'razer_mount[16509]' 'Modprobing razermouse'
+ modprobe razermouse
modprobe: ERROR: could not insert 'razermouse': Invalid argument
`

z3ntu, 2024-02-28:
Does `dmesg | tail` show anything new/interesting when running `sudo modprobe razermouse` ?

z3ntu, 2024-02-29:
From what I can find online, when this "Invalid argument" error appears, there should really be some content in dmesg to say what happened. So I'm not really sure what's going on on your PC.

quanticle, 2024-03-11:
I'm having the same issue on Fedora 39. My kernel version is `6.7.7-200.fc39.x86_64`, and I followed the instructions from the [installation page](https://openrazer.github.io/#apps) to install openrazer and razergenie:
- `sudo dnf install kernel-devel`
- `sudo dnf config-manager --add-repo "https://download.opensuse.org/repositories/hardware:/razer/Fedora_$(rpm -E %fedora)/hardware:razer.repo`
- `sudo dnf install openrazer-meta`
- `sudo gpasswd -a $USER plugdev`

After running those commands and rebooting, I ran `razergenie` and got the following error:
```
The daemon didn't detect a device that is connected
Linux detected connected devices but the daemon didn't. This could be either due to a permission problem or a kernel module problem.
```

I figured it wasn't a permission problem, as I'd run the `gpasswd` command to add my user to the `plugdev` group, so I looked into the kernel modules. Sure enough,  when I attempted to `modprobe razermouse`, I got `modprobe: ERROR: could not insert 'razermouse': Required key not available`. To fix that, I signed the kernel modules with the following procedure:

```
openssl req -new -x509 -newkey rsa:2048 -keyout "razer_sign.priv" -outform "DER" -out "razer_sign.der" -nodes -days 36500 -subj "/CN=Razer/"
sudo "/usr/src/kernels/$(uname -r)/scripts/sign-file" "sha256" "./razer_sign.priv" "./razer_sign.der" "$(modinfo -n razerkbd)"
sudo "/usr/src/kernels/$(uname -r)/scripts/sign-file" "sha256" "./razer_sign.priv" "./razer_sign.der" "$(modinfo -n razermouse)"
sudo "/usr/src/kernels/$(uname -r)/scripts/sign-file" "sha256" "./razer_sign.priv" "./razer_sign.der" "$(modinfo -n razeraccessory)"
sudo "/usr/src/kernels/$(uname -r)/scripts/sign-file" "sha256" "./razer_sign.priv" "./razer_sign.der" "$(modinfo -n razerkraken)"
```
All of the above commands succeeded. Then I imported the signing key with:
```
sudo mokutil --import "./razer_sign.der"
```
and rebooted again. After rebooting, when I tried to run `sudo modprobe razermouse`, I got the invalid argument error that OP got:
```
modprobe: ERROR: could not insert 'razermouse': Invalid argument
```

Grepping `dmesg` for `razer` yields:
```
923:[    2.520341] usb 1-4.1: Product: Razer DeathAdder Essential
924:[    2.520346] usb 1-4.1: Manufacturer: Razer
925:[    2.529980] input: Razer Razer DeathAdder Essential as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.1/1-4.1:1.0/0003:1532:0098.0004/input/input11
926:[    2.530557] hid-generic 0003:1532:0098.0004: input,hidraw3: USB HID v1.11 Mouse [Razer Razer DeathAdder Essential] on usb-0000:00:14.0-4.1/input0
927:[    2.532477] input: Razer Razer DeathAdder Essential Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.1/1-4.1:1.1/0003:1532:0098.0005/input/input12
929:[    2.585199] input: Razer Razer DeathAdder Essential as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.1/1-4.1:1.1/0003:1532:0098.0005/input/input13
930:[    2.585676] hid-generic 0003:1532:0098.0005: input,hidraw4: USB HID v1.11 Keyboard [Razer Razer DeathAdder Essential] on usb-0000:00:14.0-4.1/input1
931:[    2.586834] input: Razer Razer DeathAdder Essential as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.1/1-4.1:1.2/0003:1532:0098.0006/input/input14
933:[    2.640508] hid-generic 0003:1532:0098.0006: input,hidraw5: USB HID v1.11 Keyboard [Razer Razer DeathAdder Essential] on usb-0000:00:14.0-4.1/input
```
which doesn't seem all that helpful. I have attached the full `dmesg` log as well, in case there's some other stuff there that `grep` missed.
[dmesg.log](https://github.com/openrazer/openrazer/files/14564659/dmesg.log)

quanticle, 2024-03-11:
Running `bash -xe /usr/lib/udev/razer_mount razermouse 0003:1532:009C.0004` yields:
```
+ set -e
+ PATH=/sbin:/bin:/usr/sbin:/usr/bin
+ '[' -x /usr/bin/logger ']'
+ LOGGER=/usr/bin/logger
+ '[' -t 1 -a -z /usr/bin/logger ']'
+ '[' '!' -e /dev/log ']'
+ '[' -t 1 ']'
+ DRIVER=razermouse
+ DEVICE_ID=0003:1532:009C.0004
+ mesg 'Driver razermouse'
+ echo 'Driver razermouse'
Driver razermouse
+ /usr/bin/logger -t 'razer_mount[6672]' 'Driver razermouse'
+ mesg 'Device_ID 0003:1532:009C.0004'
+ echo 'Device_ID 0003:1532:009C.0004'
Device_ID 0003:1532:009C.0004
+ /usr/bin/logger -t 'razer_mount[6672]' 'Device_ID 0003:1532:009C.0004'
+ '[' '!' -d /sys/bus/hid/drivers/razermouse ']'
+ mesg 'Modprobing razermouse'
+ echo 'Modprobing razermouse'
Modprobing razermouse
+ /usr/bin/logger -t 'razer_mount[6672]' 'Modprobing razermouse'
+ modprobe razermouse
modprobe: ERROR: could not insert 'razermouse': Invalid argument
```

lah7, 2024-03-11:
@quanticle I'm guessing it works if you temporarily boot with secure boot disabled?

These lines from `dmesg` look interesting:

```
[    0.000000] secureboot: Secure boot enabled
[    0.000000] Kernel is locked down from EFI Secure Boot mode; see man kernel_lockdown.7
```

As a guess, maybe `Invalid argument` is because the kernel is too locked down - it "trusted" your self-signed module to attempt insertion, but the kernel might be in this "hardened" state in that OpenRazer still isn't _trustworthy/secure enough_ to be running kernel code. [This answer](https://stackoverflow.com/a/60104165) is interesting, but I'm not familiar with kernel development so it's just a guess it might be related. This is the man file: [kernel_lockdown.7](https://www.man7.org/linux/man-pages/man7/kernel_lockdown.7.html). 

Not sure if there's a way to pull more info out of `modprobe` - there is a verbose parameter:

    sudo modprobe -v razermouse

quanticle, 2024-03-11:
I rebooted with Secure Boot disabled, and it didn't seem to make a difference. I'm still getting the same error. I don't think either kernel lockdown or secure boot is an issue because I was able to build and load the [VMWare kernel host modules](https://github.com/mkubecek/vmware-host-modules) using the same process that I described above, without errors.

Running `modprobe -v razermouse` does give more information:
```
insmod /lib/modules/6.7.7-200.fc39.x86_64/extra/razermouse.ko.xz 
modprobe: ERROR: could not insert 'razermouse': Invalid argument
```
but it doesn't tell me anything that I wasn't able to find out from `lsmod` or `modinfo -n`.

quanticle, 2024-03-11:
I was also able to get some more detailed logs by running `journalctl -b -0 | grep -i "razer"`:
```
884:Mar 11 17:32:16 fedora kernel: usb 1-4.1: Product: Razer DeathAdder Essential
885:Mar 11 17:32:16 fedora kernel: usb 1-4.1: Manufacturer: Razer
886:Mar 11 17:32:16 fedora kernel: input: Razer Razer DeathAdder Essential as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.1/1-4.1:1.0/0003:1532:0098.0004/input/input11
887:Mar 11 17:32:16 fedora kernel: hid-generic 0003:1532:0098.0004: input,hidraw3: USB HID v1.11 Mouse [Razer Razer DeathAdder Essential] on usb-0000:00:14.0-4.1/input0
888:Mar 11 17:32:16 fedora kernel: input: Razer Razer DeathAdder Essential Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.1/1-4.1:1.1/0003:1532:0098.0005/input/input12
892:Mar 11 17:32:16 fedora kernel: input: Razer Razer DeathAdder Essential as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.1/1-4.1:1.1/0003:1532:0098.0005/input/input13
893:Mar 11 17:32:16 fedora kernel: hid-generic 0003:1532:0098.0005: input,hidraw4: USB HID v1.11 Keyboard [Razer Razer DeathAdder Essential] on usb-0000:00:14.0-4.1/input1
894:Mar 11 17:32:16 fedora kernel: input: Razer Razer DeathAdder Essential as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.1/1-4.1:1.2/0003:1532:0098.0006/input/input14
918:Mar 11 17:32:16 fedora kernel: hid-generic 0003:1532:0098.0006: input,hidraw5: USB HID v1.11 Keyboard [Razer Razer DeathAdder Essential] on usb-0000:00:14.0-4.1/input2
1329:Mar 11 17:32:22 fedora razer_mount[798]: Driver razermouse
1330:Mar 11 17:32:22 fedora razer_mount[802]: Device_ID 1-4.1
1331:Mar 11 17:32:22 fedora razer_mount[803]: Modprobing razermouse
1336:Mar 11 17:32:22 fedora (udev-worker)[782]: 1-4.1: Process 'razer_mount razermouse 1-4.1' failed with exit code 1.
1342:Mar 11 17:32:22 fedora razer_mount[817]: Driver razermouse
1343:Mar 11 17:32:22 fedora razer_mount[820]: Driver razermouse
1344:Mar 11 17:32:22 fedora razer_mount[819]: Driver razermouse
1345:Mar 11 17:32:22 fedora razer_mount[821]: Device_ID 1-4.1:1.0
1346:Mar 11 17:32:22 fedora razer_mount[822]: Device_ID 1-4.1:1.1
1347:Mar 11 17:32:22 fedora razer_mount[823]: Device_ID 1-4.1:1.2
1348:Mar 11 17:32:22 fedora razer_mount[826]: Modprobing razermouse
1349:Mar 11 17:32:22 fedora razer_mount[824]: Modprobing razermouse
1350:Mar 11 17:32:22 fedora razer_mount[825]: Modprobing razermouse
1352:Mar 11 17:32:22 fedora (udev-worker)[772]: 1-4.1:1.0: Process 'razer_mount razermouse 1-4.1:1.0' failed with exit code 1.
1353:Mar 11 17:32:22 fedora (udev-worker)[782]: 1-4.1:1.2: Process 'razer_mount razermouse 1-4.1:1.2' failed with exit code 1.
1355:Mar 11 17:32:22 fedora (udev-worker)[766]: 1-4.1:1.1: Process 'razer_mount razermouse 1-4.1:1.1' failed with exit code 1.
1358:Mar 11 17:32:22 fedora razer_mount[837]: Driver razermouse
1359:Mar 11 17:32:22 fedora razer_mount[840]: Device_ID 0003:1532:0098.0006
1360:Mar 11 17:32:22 fedora razer_mount[841]: Driver razermouse
1361:Mar 11 17:32:22 fedora razer_mount[842]: Modprobing razermouse
1362:Mar 11 17:32:22 fedora razer_mount[843]: Device_ID 0003:1532:0098.0004
1363:Mar 11 17:32:22 fedora razer_mount[845]: Modprobing razermouse
1365:Mar 11 17:32:22 fedora (udev-worker)[782]: 0003:1532:0098.0006: Process 'razer_mount razermouse 0003:1532:0098.0006' failed with exit code 1.
1369:Mar 11 17:32:22 fedora (udev-worker)[772]: 0003:1532:0098.0004: Process 'razer_mount razermouse 0003:1532:0098.0004' failed with exit code 1.
1370:Mar 11 17:32:22 fedora razer_mount[852]: Driver razermouse
1374:Mar 11 17:32:22 fedora razer_mount[856]: Device_ID 0003:1532:0098.0005
1375:Mar 11 17:32:22 fedora razer_mount[858]: Modprobing razermouse
1379:Mar 11 17:32:22 fedora (udev-worker)[766]: 0003:1532:0098.0005: Process 'razer_mount razermouse 0003:1532:0098.0005' failed with exit code 1.
1817:Mar 11 17:32:24 fedora systemd-logind[1074]: Watching system buttons on /dev/input/event10 (Razer Razer DeathAdder Essential Keyboard)
1818:Mar 11 17:32:24 fedora systemd-logind[1074]: Watching system buttons on /dev/input/event13 (Razer Razer DeathAdder Essential)
2727:Mar 11 17:33:01 fedora systemd[2331]: Starting openrazer-daemon.service - Daemon to manage razer devices in userspace...
2730:Mar 11 17:33:01 fedora openrazer-daemon[3510]: Starting daemon.
2731:Mar 11 17:33:01 fedora systemd[2331]: Started openrazer-daemon.service - Daemon to manage razer devices in userspace.
2732:Mar 11 17:33:01 fedora openrazer-daemon[3510]: 2024-03-11 17:33:01 | razer                          | INFO     | Initialising Daemon (v3.7.0). Pid: 3510
2733:Mar 11 17:33:01 fedora openrazer-daemon[3510]: 2024-03-11 17:33:01 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2734:Mar 11 17:33:02 fedora openrazer-daemon[3510]: 2024-03-11 17:33:02 | razer                          | INFO     | Serving DBus
```
but again, all it really says is that it wasn't able to load the `razermouse` kernel module.

quanticle, 2024-03-11:
I made a bit more progress. I cloned the repo into `~/openrazer/` and ran `make driver`. Then, I went into `openrazer/driver/` and ran `sudo insmod ./razermouse.ko`. That command completed without errors. Then, I restarted the userspace daemon by running `systemctl --user restart openrazer-daemon`. Then, when I ran `systemctl --user status openrazer-daemon`, I got the following output:
```
● openrazer-daemon.service - Daemon to manage razer devices in userspace
     Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; disabled; preset: disabled)
    Drop-In: /usr/lib/systemd/user/service.d
             └─10-timeout-abort.conf
     Active: active (running) since Mon 2024-03-11 18:28:40 CDT; 10s ago
       Docs: man:openrazer-daemon(8)
   Main PID: 5889 (openrazer-daemo)
      Tasks: 4 (limit: 28366)
     Memory: 15.3M
        CPU: 159ms
     CGroup: /user.slice/user-1000.slice/user@1000.service/app.slice/openrazer-daemon.service
             └─5889 openrazer-daemon

Mar 11 18:28:40 fedora systemd[2263]: Starting openrazer-daemon.service - Daemon to manage razer devices in userspace...
Mar 11 18:28:40 fedora openrazer-daemon[5889]: Starting daemon.
Mar 11 18:28:40 fedora systemd[2263]: Started openrazer-daemon.service - Daemon to manage razer devices in userspace.
Mar 11 18:28:40 fedora openrazer-daemon[5889]: 2024-03-11 18:28:40 | razer                          | INFO     | Initialising Daemon (v3.7.0). Pid: 5889
Mar 11 18:28:40 fedora openrazer-daemon[5889]: 2024-03-11 18:28:40 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
Mar 11 18:28:40 fedora openrazer-daemon[5889]: 2024-03-11 18:28:40 | razer                          | INFO     | Found device.0: 0003:1532:0098.0004
Mar 11 18:28:40 fedora openrazer-daemon[5889]: 2024-03-11 18:28:40 | razer                          | CRITICAL | Could not access /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.1/1-4.1:1.0/0003:1532:0098.0004/device_type, file is not owned by plugdev
Mar 11 18:28:40 fedora openrazer-daemon[5889]: 2024-03-11 18:28:40 | razer                          | INFO     | Serving DBus
```
The most interesting bit is the line: `Could not access /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.1/1-4.1:1.0/0003:1532:0098.0004/device_type, file is not owned by plugdev`. And indeed, if I go to `/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.1/1-4.1:1.0/0003:1532:0098.0004/`, I see that all the files and directories within are owned by root.

So, let's fix that! I ran: 
```
sudo chgrp plugdev country device_mode device_serial device_type dpi firmware_version logo_led_brightness logo_matrix_effect_breath logo_matrix_effect_none logo_matrix_effect_static modalias poll_rate report_descriptor test uevent version
```
and then I restarted the daemon once again with `systemctl --user restart openrazer-daemon`. And now, when I run Polychromatic, I see my mouse and I'm able to alter its settings. 

This is an ugly, ugly hack, but I was able to get my new Razer Deathadder to stop glowing like a neon sign. I think what's going on is that for some reason the kernel that the Fedora package builds against isn't the same as the kernel that is currently shipping with Fedora, and the mismatch is causing the `Invalid argument` error.

lah7, 2024-03-12:
> I rebooted with Secure Boot disabled, and it didn't seem to make a difference. 
>
>  I think what's going on is that for some reason the kernel that the Fedora package builds against isn't the same as the kernel that is currently shipping with Fedora, and the mismatch is causing the Invalid argument error.

Thanks for investigating. In that case, it could be a problem with DKMS, which is supposed to build the driver each time a system update installs a new kernel.

Not familiar with Fedora, there _should_ be two packages installed - the kernel and a kernel "headers" package. These should match the same version / kernel.

    rpm -qa | grep kernel

_(Might be `kernel-devel`)_

In Linux Mint (for original poster, @kx9080), it's probably `linux-generic` and `linux-headers-generic`. 

    apt list --installed | grep linux

This command will list what drivers have been built:

    dkms status

And the kernel version currently in use:

    uname -r

There should be a match. This could reveal if the system's booting an older kernel against a newer driver; or if a newer kernel has an older driver. Or maybe a different kind of kernel (of same version)

**Example:**

```bash
user@linuxmint$  dkms status
openrazer-driver/3.7.0, 5.15.0-100-generic, x86_64: installed
openrazer-driver/3.7.0, 5.15.0-88-generic, x86_64: installed

user@linuxmint$  uname -r
5.15.0-100-generic
```

This command should install for all kernels it finds, I believe:

    sudo dkms install openrazer-driver/3.7.0

Hopefully, the output of this will reveal any problems. If some are already installed, try `sudo dkms remove`'ing them first.

By the way, the `root` ownership when inserting modules is normal. When the device is plugged in, `udev` is responsible for setting the permission to the `plugdev` group, so replugging could also work.

kx9080, 2024-03-13:
> ```shell
> dkms status
> ```

Weirdly enough I cannot seem to get a DKMS status, it just throws this:
`
Error! Could not locate dkms.conf file.
File: /var/lib/dkms/amdgpu/6.2.4-1683306.22.04/source/dkms.conf does not exist.
`
But when I try to reinstall openrazer drivers I get:
`
Module openrazer-driver/3.7.0 already installed on kernel 5.15.0-100-generic (x86_64).
`
Any ideas? I am pretty young and not too experienced delving this deep into linux.

lah7, 2024-03-15:
@quanticle Does `dkms status` output a similar error message for you?

---

@kx9080 No worries! It seems like something else on your system might be borking DKMS from working correctly, depending on how that `amdgpu` driver was installed (or not properly cleaned up, if it was removed).

Not being familiar AMD's GPU driver or lacking expertise in Linux Mint, you might get better advice [asking on their beginner friendly forums.](https://forums.linuxmint.com)

If it was me, I'd focus on fixing the cause of the `dkms status` error first, ensure the entire system is up-to-date and then reinstall the OpenRazer driver to see if that fixes it. Besides that, I don't know the _complete_ fix as the `Invalid argument` error is something not seen before (unless, it is because of the above findings that kernel versions are being mismatched).

kx9080, 2024-04-13:
> @quanticle Does `dkms status` output a similar error message for you?
>
> ---
>
> @kx9080 No worries! It seems like something else on your system might be borking DKMS from working correctly, depending on [...]

Hey, I just did a fresh install of Linux Mint and it just turns out I messed up my dkms install but thank you for all of the help and support, I have now successfully installed open razer

closed 2024-04-13 by kx9080