openrazer/openrazer #2214: Support for DeathAdder V3 PRO (2024)

Moatimus, 2024-03-06:
### Device Info

* Name: DeathAdder V3 PRO
* Product Number: RZ01-04630300
* Model Number: RZ01-0463

RazerGenie cannot detect the device. I noticed the PID is different from the currently supported version of this device.

### Device Information
```
Wireless:
Bus 003 Device 002: ID 1532:00c3 Razer USA, Ltd Razer DeathAdder V3 Pro
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x00c3 Razer DeathAdder V3 Pro
  bcdDevice            1.01
  iManufacturer           1 Razer
  iProduct                2 Razer DeathAdder V3 Pro
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              2 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
[... 410 lines cut ...]
 19 81 29 83 15 00 25 01 75 01 95 03 81 02 95 05
 81 01 75 08 95 0E 81 01 C0 05 01 09 00 A1 01 85
 04 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 05 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0

001:007:000:DESCRIPTOR         1709757682.884747
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 06 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0

```

</details>

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

z3ntu, 2024-04-04:
You can read https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md and add your device with PID 00c3 just like 00b6/00b7.

Also do you have the mouse connected wired or wireless? What's the `lsusb | grep 1532` output when you connect it the other way?

ryan-1337, 2024-04-26:
Hi, I'm trying to add the device but I can't test to see if it works, I've added it as it should but it's not detected by polychromatic, can I make a PR and test it on your machine?

z3ntu, 2024-04-27:
@ryan-1337 If you have some code work-in-progress feel free to share somewhere, either in a branch in a fork on github or open a PR with it (marked that it doesn't work yet if it doesn't), I can take a look to see if there's anything obvious missing, but all testing can only be done with the actual device and not with another unfortunately - that's why it's basically always up to the owners of a device to add support for it.

Moatimus, 2024-04-27:
I've been wanting to try to add support myself, but I haven't had time to. If you create a branch/fork I can take a look and see if it works when I get the chance. Also, I did include both the wired and wireless for each section. I probably should have put a more clear marker for it. When wired it shows two device information sections. One has the 00c3 PID and the other has 00c2.

ryan-1337, 2024-04-27:
hi, its pushed

ryan-1337, 2024-05-08:
after disabling secure boot I can run the commands 
rmmod razermouse && insmod razermouse.ko

and I now get an error message (see screen)

[image]

I tried to give plugdev permission with these commands :

 ```sudo chown -R root:plugdev /sys/bus/hid/devices```
 ```sudo chown -R root:plugdev /sys/bus/hid/driver```

but I still get the error

ryan-1337, 2024-05-08:
previous problem solved now I just have to manage the battery

[image]

ryan-1337, 2024-05-08:
everything's fine, just a formatting problem I can't solve

z3ntu, 2024-05-09:
Please keep the state of the PR in #2239, otherwise it's hard to track.

PR #2239 (2024-07-02): Add support for DeathAdder V3 Pro (00c2/00c3)
Fixes #2214
closed 2024-07-02 by z3ntu