openrazer/openrazer #2215: Click input delay on Razer DeathAdder V3 Pro

AmbreKC, 2024-03-08:
### Manjaro

Razer DeathAdderV3 Pro and openrazer v3.7.0

### Support

Hello, I've been having this issue where sometimes left and right click do not register until I press it several times.
Let me preface this by saying that the mouse in question is second-hand so it could be a hardware issue but I only get unregistered clicks (on both left and right) and not any other kind of click issue like double-clicks and such that could indicate a degradation of the switches.

This also seems to only happen in games and at first I thought it was only happening in CS2 when alt-tabbing back into CS2 at the start of a round but I've seen it happen in other games now and even when I haven't tabbed out for a long time and it's possible that it also happens outside of games but I don't notice unlike when I can't hit a headshot.
I've had this happen in wired mode and with two different dongles (a standard one and the hyperpolling one). Though I have to say I didn't try as much in wired mode.

I have tried the different steps on the troubleshooting page but everything seems in order, the one thing that looks out of the ordinary is when looking at dmesg I do get what looks like potential error messages.

```
sudo dmesg | grep razer                    
[sudo] password for trans_fem_zboomer: 
[  131.789117] razermouse 0003:1532:00B3.0008: input,hidraw4: USB HID v1.00 Mouse [Razer Razer HyperPolling Wireless Dongle] on usb-0000:03:00.0-2/input0
[  132.092443] razermouse 0003:1532:00B3.0009: input,hidraw5: USB HID v1.00 Keyboard [Razer Razer HyperPolling Wireless Dongle] on usb-0000:03:00.0-2/input1
[  132.305629] razermouse 0003:1532:00B3.000A: input,hidraw6: USB HID v1.00 Keyboard [Razer Razer HyperPolling Wireless Dongle] on usb-0000:03:00.0-2/input2
[  133.778864] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 16, command_class: 00, command_id.id: 82 Params: 00000000000000000000000000000000 .
[  133.945898] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 16, command_class: 00, command_id.id: 82 Params: 00000000000000000000000000000000 .
[  134.112918] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 16, command_class: 00, command_id.id: 82 Params: 00000000000000000000000000000000 .
[  134.280925] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 16, command_class: 00, command_id.id: 82 Params: 00000000000000000000000000000000 .
[  134.447942] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 16, command_class: 00, command_id.id: 82 Params: 00000000000000000000000000000000 .
[  134.618055] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 07, command_class: 04, command_id.id: 05 Params: 01070807080000000000000000000000 .
[  134.684954] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 00, command_id.id: 40 Params: 00100000000000000000000000000000 .
[  134.752360] razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 00, command_id.id: 40 Params: 01100000000000000000000000000000 .
[  134.820366] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[  135.051986] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 07, command_class: 04, command_id.id: 85 Params: 00000000000000000000000000000000 .
[  135.120994] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 00, command_id.id: 81 Params: 00000000000000000000000000000000 .
[  170.849502] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 00, command_id.id: 81 Params: 00000000000000000000000000000000 .
[  170.941515] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 00, command_id.id: 81 Params: 00000000000000000000000000000000 .
[  171.009507] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 84 Params: 00000000000000000000000000000000 .
[  171.077531] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[  171.148518] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 83 Params: 00000000000000000000000000000000 .
[  171.215525] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 07, command_class: 04, command_id.id: 85 Params: 00000000000000000000000000000000 .
[  171.282530] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 07, command_class: 04, command_id.id: 85 Params: 00000000000000000000000000000000 .
[  199.273095] razermouse 0003:1532:00B6.000C: input,hidraw8: USB HID v1.11 Mouse [Razer Razer DeathAdder V3 Pro] on usb-0000:03:00.0-5/input0
[  199.338587] razermouse 0003:1532:00B6.000D: input,hidraw9: USB HID v1.11 Keyboard [Razer Razer DeathAdder V3 Pro] on usb-0000:03:00.0-5/input1
[  199.408247] razermouse 0003:1532:00B6.000E: input,hidraw10: USB HID v1.11 Keyboard [Razer Razer DeathAdder V3 Pro] on usb-0000:03:00.0-5/input2
[  203.766728] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 00, command_id.id: 40 Params: 00020000000000000000000000000000 .
[  203.833718] razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 00, command_id.id: 40 Params: 01020000000000000000000000000000 .
[  209.700325] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 07, command_class: 04, command_id.id: 05 Params: 01064006400000000000000000000000 .
[  209.767299] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 07, command_class: 04, command_id.id: 05 Params: 01064006400000000000000000000000 .
[  213.718685] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 00, command_id.id: 81 Params: 00000000000000000000000000000000 .
[  213.824696] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 84 Params: 00000000000000000000000000000000 .
[  213.891726] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[  221.503467] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 83 Params: 00000000000000000000000000000000 .
[  221.570468] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 07, command_class: 04, command_id.id: 85 Params: 00000000000000000000000000000000 .
[  221.637474] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 07, command_class: 04, command_id.id: 85 Params: 00000000000000000000000000000000 .
[  223.169629] razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 07, command_class: 04, command_id.id: 05 Params: 01064006400000000000000000000000 .
[... cut ...]
 05 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0

001:038:000:DESCRIPTOR         1709890979.070047
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 0B 81 03
 05 0C 0A 38 02 15 81 25 7F 75 08 95 01 81 06 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0
```

I'm not sure if you need packet capture for this, I'm currently downloading a VM from Microsoft's website in case it's needed.

z3ntu, 2024-03-10:
Can you try uninstalling openrazer and see if this issue persists? OpenRazer really shouldn't cause any issues like that.

AmbreKC, 2024-03-15:
Hey so since then, I changed the sleet mode setting to 15 mins (was 2 mins, which shouldn't be an issue in a game I don't go two mins without clicking but it fixed it somehow).
To be more precise, it fixed it in Apex Legends, but not counter-strike but I feel like it's probably an issue on the game's part since there are no other areas where this happens now (thought it didn't happen with the mouse I had before), the linux release is riddled with problems and I often get weird window focus loss problems so the issue could even be unrelated to the mouse altogether and just this specific game tbh.

AmbreKC, 2024-03-16:
Haha ! In fact I just found that I'm not the only person suffering of this issue with CS specifically -> https://github.com/ValveSoftware/csgo-osx-linux/issues/3566

The people encountering this issue also have similar setups to me (both in terms of hardware (no mention of their mouse though) and software) so I think I'm just going to report that I have this issue in Valve's repo.

I'm leaving whether or not to close this issue up to you because there is still the fact that it was also happening in Apex Legends until I changed the sleep mode setting. This happened a lot less though and it could just be that I'm not as fast as I thought or I had lag spikes, etc...

dm17, 2024-09-19:
Same:

```
razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 00, command_id.id: 40 Params: 01040000000000000000000000000000 .
...
razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
```

closed 2025-01-18 by AmbreKC