openrazer/openrazer #2216: Support for Razer Basilisk v3 X Hyperspeed

kubo12312, 2024-03-08:
### Device Info

* Name: Razer Basilisk v3 X Hyperspeed
* Product Number: RZ01-04870100-R3U1
* Model Number: RZ01-04870100

### Device Information
```
Please post the output of
    lsusb -d 1532: -v

Bus 001 Device 002: ID 1532:00b9 Razer USA, Ltd Razer Basilisk V3 X HyperSpeed
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x00b9 
  bcdDevice            1.00
  iManufacturer           1 Razer
  iProduct                2 Razer Basilisk V3 X HyperSpeed
  iSerial                 3 000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          4 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      97
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     314
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  E
```

```
Please post the output of
    ls -lah /dev/input/by-id/

drwxr-xr-x 2 root root 220 mar  8 12:54 .
drwxr-xr-x 4 root root 760 mar  8 12:54 ..
lrwxrwxrwx 1 root root  10 mar  8 12:44 usb-ITE_Tech._Inc._ITE_Device_8258_-event-if00 -> ../event17
lrwxrwxrwx 1 root root  10 mar  8 12:44 usb-ITE_Tech._Inc._ITE_Device_8258_-event-kbd -> ../event16
lrwxrwxrwx 1 root root  10 mar  8 12:54 usb-Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000-event-if01 -> ../event14
lrwxrwxrwx 1 root root  10 mar  8 12:54 usb-Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000-event-mouse -> ../event15
lrwxrwxrwx 1 root root  10 mar  8 12:54 usb-Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000-if01-event-kbd -> ../event12
lrwxrwxrwx 1 root root  10 mar  8 12:54 usb-Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000-if01-event-mouse -> ../event13
lrwxrwxrwx 1 root root   9 mar  8 12:54 usb-Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000-if01-mouse -> ../mouse2
lrwxrwxrwx 1 root root  10 mar  8 12:54 usb-Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000-if02-event-kbd -> ../event11
lrwxrwxrwx 1 root root   9 mar  8 12:54 usb-Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000-mouse -> ../mouse3

```

<details>

```
Please post the output of
    usbhid-dump -m 1532 -ed
001:002:002:DESCRIPTOR         1709898857.637819
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

001:002:001:DESCRIPTOR         1709898857.638840
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 0E 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0 05 01 09
 02 A1 01 85 0A 09 01 A1 00 05 09 19 01 29 06 15
 00 25 01 75 01 95 06 81 02 75 01 95 02 81 03 06
 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05 01
 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09 31
 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 C0 05 0C
 09 01 A1 01 85 02 19 00 2A 3C 02 15 00 26 3C 02
 95 01 75 10 81 00 75 08 95 0D 81 01 C0 05 01 09
 80 A1 01 85 03 19 81 29 83 15 00 25 01 75 01 95
 03 81 02 95 05 81 01 75 08 95 0E 81 01 C0 05 01
 09 00 A1 01 85 04 09 03 15 00 26 FF 00 35 00 46
 FF 00 75 08 95 0F 81 00 C0 05 01 09 00 A1 01 85
 05 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 08 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0 05
 01 09 00 A1 01 85 09 09 03 15 00 26 FF 00 35 00
 46 FF 00 75 08 95 0F 81 00 C0

001:002:000:DESCRIPTOR         1709898857.639508
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 0B 81 03
 06 00 FF 05 0C 0A 38 02 15 81 25 7F 75 08 95 01
 81 06 05 01 09 38 15 81 25 7F 75 08 95 01 81 06
 09 30 09 31 16 01 80 26 FF 7F 75 10 95 02 81 06
 C0 06 00 FF 09 02 15 00 25 01 75 08 95 5A B1 01
 C0

```

</details>

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

z3ntu, 2024-03-10:
This device is already supported in the master branch. Either you can wait for the next release or you can check https://github.com/openrazer/openrazer/wiki/Building-a-package if you want to build it from source now.

closed 2024-03-10 by z3ntu

omidforoqi, 2024-03-23:
I build the master branch from source in Arch/Manjaro with AUR, but after that my device is not supported agian.

Erothas, 2024-03-28:
> This device is already supported in the master branch. Either you can wait for the next release or you can check https://github.com/openrazer/openrazer/wiki/Building-a-package if you want to build it from source now.

Hey there. I saw this reply of yours about a week ago, and I figured a release will not be coming any time soon. However, I really need to get my mouse's functionality unlocked, so I built openrazer with the latest commit available. The result was my mouse remaining as unsupported in polychromatic. While I wasn't sure if that was needed or not, but I just did the same, and built polychromatic from source as well. But still, my mouse is unsupported.

Is there anything I am doing wrong?

z3ntu, 2024-03-28:
@Erothas What distro are you using? What commands did you run to build OpenRazer from source?

Erothas, 2024-03-28:
@z3ntu NixOS on my main system, and Arch Linux in a VM for sanity check. This is the module I build openrazer with on Nix:
```
let
  openrazer_overlay = self: super: {
    openrazer = super.openrazer.overrideAttrs (old: {
      version = "git";
      src = super.fetchFromGitHub {
        owner = "openrazer";
        repo = "openrazer";
        rev = "479a8f330bf126abee6c54ca9ccfd3af7f4e2ed3"; 
        sha256 = "1xbhxvb03lpky7wjl2xd0hvkhnskkagy5bxg3lpvx52y1rwibvm6"; 
      };
    });
  };

  polychromatic_overlay = self: super: {
    polychromatic = super.polychromatic.overrideAttrs (old: {
      version = "git";
      src = super.fetchFromGitHub {
        owner = "polychromatic";
        repo = "polychromatic";
        rev = "2ee53dc567698608394f26f62ecf49229515e8f6"; 
        sha256 = "1h296x0qbq4vmmbfv8rldfl3wk902cwad9hpypk1s4crj1bgk4bi";
      };
    });
  };
in
{
  nixpkgs.overlays = [ openrazer_overlay polychromatic_overlay ];
  hardware.openrazer.enable = true;
  environment.systemPackages = with pkgs; [ openrazer-daemon polychromatic ];
  users.users.${vars.user}.extraGroups = [ "plugdev" ];
}

```
As for Arch linux, I firstly tried this method: https://github.com/openrazer/openrazer/wiki/Building-a-package#build-packages-from-a-different-github-repository
But then the AUR package did basically the same thing. So, I installed that instead.

z3ntu, 2024-03-29:
While I have no clue about Nix, can you check if whereever Nix installs the udev rules to (normally /lib/udev/rules.d/99-razer.rules I think) if that file contains `00b9`. Also check your `lsusb | grep 1532` output please.

Erothas, 2024-03-29:
So, it seems like there is a way to find out the udev rules set in NixOS. `sudo udevadm info --export-db | grep -i razer` should be the one. Here is the output:
```
E: ID_MODEL=Razer_Basilisk_V3_X_HyperSpeed
E: ID_MODEL_ENC=Razer\x20Basilisk\x20V3\x20X\x20HyperSpeed
E: ID_SERIAL=Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000
E: ID_VENDOR=Razer
E: ID_VENDOR_ENC=Razer
E: ID_USB_MODEL=Razer_Basilisk_V3_X_HyperSpeed
E: ID_USB_MODEL_ENC=Razer\x20Basilisk\x20V3\x20X\x20HyperSpeed
E: ID_USB_SERIAL=Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000
E: ID_USB_VENDOR=Razer
E: ID_USB_VENDOR_ENC=Razer
E: ID_VENDOR_FROM_DATABASE=Razer USA, Ltd
E: ID_VENDOR_FROM_DATABASE=Razer USA, Ltd
E: HID_NAME=Razer Razer Basilisk V3 X HyperSpeed
E: NAME="Razer Razer Basilisk V3 X HyperSpeed"
E: ID_MODEL=Razer_Basilisk_V3_X_HyperSpeed
E: ID_MODEL_ENC=Razer\x20Basilisk\x20V3\x20X\x20HyperSpeed
E: ID_SERIAL=Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000
E: ID_VENDOR=Razer
E: ID_VENDOR_ENC=Razer
E: ID_USB_MODEL=Razer_Basilisk_V3_X_HyperSpeed
E: ID_USB_MODEL_ENC=Razer\x20Basilisk\x20V3\x20X\x20HyperSpeed
E: ID_USB_SERIAL=Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000
E: ID_USB_VENDOR=Razer
E: ID_USB_VENDOR_ENC=Razer
S: input/by-id/usb-Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000-event-mouse
E: ID_MODEL=Razer_Basilisk_V3_X_HyperSpeed
E: ID_MODEL_ENC=Razer\x20Basilisk\x20V3\x20X\x20HyperSpeed
E: ID_SERIAL=Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000
E: ID_VENDOR=Razer
E: ID_VENDOR_ENC=Razer
E: ID_USB_MODEL=Razer_Basilisk_V3_X_HyperSpeed
E: ID_USB_MODEL_ENC=Razer\x20Basilisk\x20V3\x20X\x20HyperSpeed
E: ID_USB_SERIAL=Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000
E: ID_USB_VENDOR=Razer
E: ID_USB_VENDOR_ENC=Razer
E: DEVLINKS=/dev/input/by-path/pci-0000:0a:00.0-usbv2-0:8:1.0-event-mouse /dev/input/by-id/usb-Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000-event-mouse /dev/input/by-path/pci-0000:0a:00.0-usb-0:8:1.0-event-mouse
S: input/by-id/usb-Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000-mouse
E: ID_MODEL=Razer_Basilisk_V3_X_HyperSpeed
E: ID_MODEL_ENC=Razer\x20Basilisk\x20V3\x20X\x20HyperSpeed
E: ID_SERIAL=Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000
E: ID_VENDOR=Razer
E: ID_VENDOR_ENC=Razer
E: ID_USB_MODEL=Razer_Basilisk_V3_X_HyperSpeed
E: ID_USB_MODEL_ENC=Razer\x20Basilisk\x20V3\x20X\x20HyperSpeed
E: ID_USB_SERIAL=Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000
E: ID_USB_VENDOR=Razer
E: ID_USB_VENDOR_ENC=Razer
E: DEVLINKS=/dev/input/by-path/pci-0000:0a:00.0-usbv2-0:8:1.0-mouse /dev/input/by-path/pci-0000:0a:00.0-usb-0:8:1.0-mouse /dev/input/by-id/usb-Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000-mouse
E: ID_VENDOR_FROM_DATABASE=Razer USA, Ltd
E: HID_NAME=Razer Razer Basilisk V3 X HyperSpeed
E: NAME="Razer Razer Basilisk V3 X HyperSpeed"
E: ID_MODEL=Razer_Basilisk_V3_X_HyperSpeed
E: ID_MODEL_ENC=Razer\x20Basilisk\x20V3\x20X\x20HyperSpeed
E: ID_SERIAL=Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000
E: ID_VENDOR=Razer
E: ID_VENDOR_ENC=Razer
E: ID_USB_MODEL=Razer_Basilisk_V3_X_HyperSpeed
E: ID_USB_MODEL_ENC=Razer\x20Basilisk\x20V3\x20X\x20HyperSpeed
E: ID_USB_SERIAL=Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000
E: ID_USB_VENDOR=Razer
E: ID_USB_VENDOR_ENC=Razer
S: input/by-id/usb-Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000-event-if01
E: ID_MODEL=Razer_Basilisk_V3_X_HyperSpeed
E: ID_MODEL_ENC=Razer\x20Basilisk\x20V3\x20X\x20HyperSpeed
E: ID_SERIAL=Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000
E: ID_VENDOR=Razer
E: ID_VENDOR_ENC=Razer
E: ID_USB_MODEL=Razer_Basilisk_V3_X_HyperSpeed
[... 65 lines cut ...]
E: NAME="Razer Razer Basilisk V3 X HyperSpeed"
E: ID_MODEL=Razer_Basilisk_V3_X_HyperSpeed
E: ID_MODEL_ENC=Razer\x20Basilisk\x20V3\x20X\x20HyperSpeed
E: ID_SERIAL=Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000
E: ID_VENDOR=Razer
E: ID_VENDOR_ENC=Razer
E: ID_USB_MODEL=Razer_Basilisk_V3_X_HyperSpeed
E: ID_USB_MODEL_ENC=Razer\x20Basilisk\x20V3\x20X\x20HyperSpeed
E: ID_USB_SERIAL=Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000
E: ID_USB_VENDOR=Razer
E: ID_USB_VENDOR_ENC=Razer
S: input/by-id/usb-Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000-if02-event-kbd
E: ID_MODEL=Razer_Basilisk_V3_X_HyperSpeed
E: ID_MODEL_ENC=Razer\x20Basilisk\x20V3\x20X\x20HyperSpeed
E: ID_SERIAL=Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000
E: ID_VENDOR=Razer
E: ID_VENDOR_ENC=Razer
E: ID_USB_MODEL=Razer_Basilisk_V3_X_HyperSpeed
E: ID_USB_MODEL_ENC=Razer\x20Basilisk\x20V3\x20X\x20HyperSpeed
E: ID_USB_SERIAL=Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000
E: ID_USB_VENDOR=Razer
E: ID_USB_VENDOR_ENC=Razer
E: DEVLINKS=/dev/input/by-id/usb-Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000-if02-event-kbd /dev/input/by-path/pci-0000:0a:00.0-usbv2-0:8:1.2-event-kbd /dev/input/by-path/pci-0000:0a:00.0-usb-0:8:1.2-event-kbd
```
`lsusb | grep 1532` output also: Bus 001 Device 003: ID 1532:00b9 Razer USA, Ltd Razer Basilisk V3 X HyperSpeed

Erothas, 2024-04-03:
@z3ntu So, I have absolutely no idea if I should be doing this or not. But was I supposed to ping you with my last response for you to notice it? I just don't know how github works. So, I apologise for any inconvenience caused.

Erothas, 2024-04-04:
> This output is not very useful. Can you run `find / -name 99-razer.rules` to find where this file is?

Alright. That yields results. However:
```
# Mice
ATTRS{idProduct}=="0013|0016|0020|0024|0025|0029|002e|002f|0032|0034|0036|0037|0038|0039|003e|003f|0040|0041|0042|0043|0044|0045|0046|0048|004c|004f|0050|0053|0054|0059|005a|005b|005c|005e|0060|0062|0064|0065|0067|006a|006b|006c|006e|006f|0070|0071|0072|0073|0077|0078|007a|007b|007c|007d|0080|0083|0084|0085|0086|0088|008a|008c|008d|008f|0090|0091|0094|0095|0096|0098|0099|009a|009c|009e|009f|00a1|00a5|00a6|00aa|00ab|00b0|00b2|00b3|00b4|00b6|00b7|00b8", \
    ATTRS{idVendor}=="1532", \
    ENV{ID_RAZER_CHROMA}="1", ENV{RAZER_DRIVER}="razermouse"
```
No ```00b9``` in sight.

z3ntu, 2024-04-04:
Then it very much looks like you didn't install the version from git master. Compare with https://github.com/openrazer/openrazer/blob/master/install_files/udev/99-razer.rules (note it's not just that file of course, but a bunch of files)

Erothas, 2024-04-04:
That's certainly strange. I don't see anything wrong with my install. Anyway, I will look around and see what's going on, then report back with my new findings. Thanks.

nordbrg, 2024-04-05:
I had the same problem on Ubuntu 22.04. In my case the solution was to switch to the `daily` PPA
`sudo add-apt-repository ppa:openrazer/daily` then after reboot

Erothas, 2024-04-09:
So, you were right. The razer module that I created only fetched the openrazer-daemon from the github repo. However, the other part of the puzzle was left untouched; the kernel module. So, after hours of research with the lovely folks at NixOS's discord server, this is the complete module that works flawlessly:

```
{ config, pkgs, vars, ... }:

let
  openrazer_overlay = self: super: {
    openrazer-daemon = super.openrazer-daemon.overrideAttrs (old: rec {
      version = "git";
      src = super.fetchFromGitHub {
        owner = "openrazer";
        repo = "openrazer";
        rev = "fd1e9a705d65aa1c85ae577246a263ebd845198a";
        sha256 = "LDthdD0AlEWUdO0zY5guxyQFkVxjGpTfGNKaSCc7C7g=";
      };
      sourceRoot = "${src.name}/daemon";
    });
  };

  polychromatic_overlay = self: super: {
    polychromatic = super.polychromatic.overrideAttrs (old: {
      version = "git";
      src = super.fetchFromGitHub {
        owner = "polychromatic";
        repo = "polychromatic";
        rev = "2ee53dc567698608394f26f62ecf49229515e8f6";
        sha256 = "cZH5VpCZER3m9RempjgTIE0+qGs0o+1WrZvghUE3ScA=";
      };
    });
  };
in
{
  hardware.openrazer = {
    enable = true;
    users = ["${vars.user}"];
  };

  nixpkgs.overlays = [ openrazer_overlay polychromatic_overlay ];
  environment.systemPackages = with pkgs; [ openrazer-daemon polychromatic ];
  users.users.${vars.user}.extraGroups = [ "plugdev" ];
  boot.kernelPackages = pkgs.linuxPackages_xanmod_latest.extend (self: super: {
    openrazer = super.openrazer.overrideAttrs (old: rec {
      version = "git";
      src = pkgs.fetchFromGitHub {
        owner = "openrazer";
        repo = "openrazer";
        rev = "fd1e9a705d65aa1c85ae577246a263ebd845198a";
        sha256 = "LDthdD0AlEWUdO0zY5guxyQFkVxjGpTfGNKaSCc7C7g=";
      };
    });
  });

}
```
To my very unfortunate surprise though, I discovered that the openrazer in general CANNOT change any of the mice's button keybinds. I wasted so many hours on a mission that was impossible in the first place. "My disappointment is immeasurable, and my day is ruined". And no, before you suggest to make a windows VM, set the keys up, and it will be stored in the in-built memory. That does not work for the hypershift button. It's the only button that, for whatever reason, does not call up on the saved profile of the in-built memory of the mouse...

z3ntu, 2024-04-10:
For keybinds you can use this project https://github.com/sezanzeb/input-remapper , this topic is out of scope for OpenRazer.

Erothas, 2024-04-10:
> For keybinds you can use this project https://github.com/sezanzeb/input-remapper , this topic is out of scope for OpenRazer.

You should be the most experienced in the razer sphere to know that nothing (outside of synapse itself) can even detect the hypershift button in the first place as any kind of input. But I still went ahead in case I was horribly mistaken, and installed input-remapper. Yeah, it does not detect the button as I said...

Anyway, I apologise for the tangent.

z3ntu, 2024-04-10:
Can you explain this Hypershift feature? Is it a dedicated button on the device that just doesn't produce any input event right now? Or is this just a software concept a.k.a press some button to change the functionality of other buttons? I wonder if there's some hardware support necessary for this or not.

Erothas, 2024-04-10:
> Can you explain this Hypershift feature? Is it a dedicated button on the device that just doesn't produce any input event right now? Or is this just a software concept a.k.a press some button to [...]

Nope. It does not produce any input event in xev (x11) or wev (wayland). Hypershift is indeed a feature which when the assigned button is pressed, the other buttons "transform", and execute a different input. A quick and simple example would be if you have two side buttons bound to the number 1  and 2. When you press down the hypershift button, they become 3 and 4, until you lift your finger off.

The dedicated button for the hypershift feature can be bound to any other input, as per usual, though. However, it seems that said button behaves differently hardware and software wise? That's what I am seeing at least. I also found someone else reporting the same problem on some razer support page.

z3ntu, 2024-04-10:
Sounds pretty much like this could be implemented for any input device inside input-remapper, as far as I understand it now? So kind of conditional remapping, press one button then some remappings get active and when you release then the buttons do the regular thing again. Maybe ask in the issue tracker for input-remapper then for this functionality?

Erothas, 2024-04-10:
Oh, I would say the majority of people will not care one bit about hypershift functionality itself. It's just that its button is pretty much dead the moment you quit synapse. I, and other people from what I have gathered from my research online, were hoping that openrazer would expose its input event, and make the button functional once again on Linux. So that one can rebind it to just some other key, like the usual. Remapping keys can be done in a million ways anyway. Any bog standard window manager or desktop manager already has that built in. However, they are all pointless if the key is not detected in the first place.

As to whether that is even possible in the first place or not.. Well, you are the doctor here :D

z3ntu, 2024-04-11:
I don't have the device so I can't help there more than pointing you to e.g. razer_raw_event function in razermouse_driver.c where you can see if you get any data at all from the device when this button gets pressed. You can add some logging there for example and see what data arrives.