openrazer/openrazer #2221: Unit openrazer-daemon.service could not be found. Fedora 39

ynuske, 2024-03-14:
I installed using  -sudo dnf config-manager --add-repo https://download.opensuse.org/repositories/hardware:razer/Fedora_39/hardware:razer.repo
- sudo dnf install razergenie
but when I logged in it said "Unit openrazer-daemon.service could not be found."

and I also installed openrazer from the website using
sudo dnf install kernel-devel

[image]
sudo dnf config-manager --add-repo https://download.opensuse.org/repositories/hardware:/razer/Fedora_$(rpm -E %fedora)/hardware:razer.repo

sudo dnf install openrazer-meta

lah7, 2024-03-15:
Installing `openrazer-meta` should install `openrazer-daemon` too. Is the daemon working?

    openrazer-daemon -F

If not, try installing that package explicitly:

    sudo dnf install openrazer-daemon

ynuske, 2024-03-16:
Hi, I tried everything in Fedora, but I couldn't. So I decided to change distros and installed Arch Linux, but the same error happened.
[image]
[image]

z3ntu, 2024-03-16:
What's the output of `systemctl --user status openrazer-daemon.service`  ? And `ls -al /usr/lib/systemd/user/openrazer-daemon.service` ?

ynuske, 2024-03-18:
look 
[image]
[image]

z3ntu, 2024-03-19:
That part looks fine. On your new install you also installed RazerGenie via the package manager, so from the AUR? Can you run `razergenie` in the terminal and post the output here?

ynuske, 2024-03-19:
I installed rezergenie via flatpak, but I uninstalled it and installed it via aur and it still gives this error
[image]
[image]
[image]

z3ntu, 2024-03-20:
Are you using some weird application sandboxing thing? This is definitely not normal...

ynuske, 2024-03-20:
no, right after installing arch linux I downloaded openrazer and then rezergenie

lah7, 2024-03-20:
I tried installing `razergenie` (AUR) and `openrazer-daemon` and could reproduce the issue on Arch Linux.

To get it working:

    sudo gpasswd -a $USER plugdev
    systemctl --user enable openrazer-daemon

Then reboot and try opening RazerGenie again. Does that work for you?

ynuske, 2024-03-20:
bro, my luck is 
[image]

ynuske, 2024-03-20:
I researched and saw that I needed linux-headers, so after installing and restarting I think it worked
[image]

closed 2024-03-21 by z3ntu