openrazer/openrazer #2223: Razer DeathAdder V3. Wrong pooling rate after reboot.

VeilSilence, 2024-03-17:
### Distribution
OS: Arch Linux x86_64
Kernel: 6.7.9-arch1-1
Openrazer-driver-dkms 3.7.0-1

### Support

- [✔] Have you tried rebooting? 
- [✔] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page? 

###  Issue

Greetings. 
Got the problem with my DeathAdder V3. I installed openrazer-dkms, set plugdev group, rebooted, installed polychromatic app, plugged my mouse, set 8000Hz pooling rate.
Then I used `sudo evhz` app to verify that my mouse actually at, 8000Hz, which is correct. 
After pc reboot, `evhz` report that my mouse stuck at 1000Hz pooling rate, when in `polychromatic`polling rate set at 8000Hz. 

###  What I tried to fix the issue
After fresh boot, I changed from 8000Hz pooling rate to 1000Hz in `polychromatic`. 
At 8000Hz in `polychromatic`, `evhz` reports `Latest 1000Hz, Average 856Hz`. When I set to 1000Hz in `polychromatic`, `evhz` reports `Latest 125Hz, Average 116Hz` for example. 
`modprobe -r usbhid && modprobe usbhid` does not fix this issue. 
The only way I found to fix this issue, is unplug/replug my mouse. I tried to put in different usb ports -> reboot and verify my pooling rate. All usb ports behave the same. I also checked the logs, but i don't see anything suspicious. For me atleast.
In windows, all working as intended. 

[dmesg.razer.txt](https://github.com/openrazer/openrazer/files/14627436/dmesg.razer.txt)
[dmesg.usb.txt](https://github.com/openrazer/openrazer/files/14627438/dmesg.usb.txt)
[journalctl.razer.txt](https://github.com/openrazer/openrazer/files/14627439/journalctl.razer.txt)
[ls.al.txt](https://github.com/openrazer/openrazer/files/14627440/ls.al.txt)
[lsmod.razer.txt](https://github.com/openrazer/openrazer/files/14627441/lsmod.razer.txt)

VeilSilence, 2024-03-17:
Update: 6.8.1 kernel fixed this issue for me.
`modprobe -r xhci_pci && modprobe xhci-pci` also can be used for "workaround".

VeilSilence, 2024-04-04:
The problem still persists. 
I'll reopen the issue, so perhaps someone has an idea why it's happening. When I shut down the PC and then power it on, the polling rate is 8000Hz as expected. However, if I use systemctl reboot -i, the polling rate will be 1/8 of my settings.

z3ntu, 2024-04-04:
Honestly sounds like the mouse firmware might be doing some weird things. But you could try to uninstall OpenRazer to see if it makes any difference, because I don't think OpenRazer is causing this at all.

burd1, 2024-04-05:
https://github.com/openrazer/openrazer/issues/2122

VeilSilence, 2024-04-05:
> Honestly sounds like the mouse firmware might be doing some weird things. But you could try to uninstall OpenRazer to see if it makes any difference, because I don't think OpenRazer is causing this at all.

Yea.. looks like mouse firmware bug for me also, because after "warm boot" this issue appear. 

I checked #2122. After warm boot,  `cat /sys/bus/hid/drivers/razer*/*1532*/device_serial` return same serial ( PM2343H26801396 ) as it was before.
`cat .config/openrazer/persistence.conf`
```shell
[PM2343H26801396]
dpi_x = 3200
dpi_y = 3200
poll_rate = 8000
``` 
Looks same, but pooling rate 1/8.

I also moved from arch linux to NixOS, so i doubt reinstalling change something. Looks like i got faulty/unlucky mouse then..

lah7, 2024-04-05:
Out of curiosity, does the "USB reset" command cause (or correct) this firmware behaviour?

    sudo usbreset 1532:00B2

_(On Arch, `usbreset` is in the `usbutils` package)_

lah7, 2024-04-06:
Do you have Razer Synapse installed on the Windows side? Also, did you try uninstalling OpenRazer on NixOS?

I'd try isolating them to see if having both installed is the cause of the problem. Razer devices typically have a ["normal" and "driver" mode](https://github.com/openrazer/openrazer/wiki/Using-the-mouse-driver#device_mode---set-device-mode).

When the device is plugged in, it'll be in "normal" mode, until [`openrazer-daemon` or Synapse service] puts it into "driver" mode. Maybe the mouse is still in "driver" mode across reboots? There is a [sysfs file](https://github.com/openrazer/openrazer/wiki/Using-the-mouse-driver#device_mode---set-device-mode) to find out. Could try stopping `openrazer-daemon` (and check device is in "normal" mode) before rebooting. Or, check if `set_device_mode` impacts the Hz test?

VeilSilence, 2024-04-14:
Sorry for the delayed response; I've had some troubles to attend to this week. Well, this problem is really weird.

Last week, I decided to try plug my mouse in "Headphone Stand" with three USB ports in it, which I rarely used. Plugged mouse -> rebooted, and everything worked as it should. The polling rate stayed at 8000Hz when I rebooted or switched between Windows and NixOS. After that, i was busy, so i couldn't test it thoroughly.

I returned home three days ago, and this bug hasn't occurred in these three days anymore. Now, I decided to try plugging the mouse back into my first USB port, where this bug started happening... and again, everything is working. I tried four chain reboots in a row (NixOS -> reboot -> NixOS -> reboot -> NixOS). I also tried mixing between Windows -> reboot -> NixOS. Everything is working... for now.

I'll keep this bug open for now in case this occur again. Somehow, plugging mouse to usb hub ( Headphone Stand in my case ), fixed the issue?

lah7, 2024-04-14:
I'm not sure. I have the Base Station Chroma ("Headphone Stand") as well, which is just a regular USB 2.0 hub. Was your first port a USB 3.0 (blue one)?

VeilSilence, 2024-04-15:
> I'm not sure. I have the Base Station Chroma ("Headphone Stand") as well, which is just a regular USB 2.0 hub. Was your first port a USB 3.0 (blue one)?

Yes, blue one.

Also i remember that when this pooling rate bug happens, in KDE -> Settings -> USB devices Razer Deathadder "USB version" was 1.1 or 1.0. Right now it shows USB 2.0 always and pooling rate correct.

VeilSilence, 2024-05-11:
After a month without any issues, this bug has reappeared.

NixOS update changed related to Python/Glib things, and OpenRazer stopped working. Without the OpenRazer daemon, my mouse was running at a 1000Hz polling rate, according to `evhz`
<details><summary>This error I got after update nixos system</summary>
<p>

```shell
openrazer-daemon -Fv
Traceback (most recent call last):
  File "/nix/store/s8vsc4p2cyqc2kim2cjvvl85d9fdz54x-python3.11-openrazer-daemon-3.8.0/bin/.openrazer-daemon-wrapped", line 13, in <module>
    from openrazer_daemon.daemon import RazerDaemon, __version__
  File "/nix/store/s8vsc4p2cyqc2kim2cjvvl85d9fdz54x-python3.11-openrazer-daemon-3.8.0/lib/python3.11/site-packages/openrazer_daemon/daemon.py", line 20, in <module>
    from gi.repository import GLib
  File "/nix/store/40cddha2iiqzbqd6dpal3rlrgvb4p0yc-python3.11-pygobject-3.48.2/lib/python3.11/site-packages/gi/importer.py", line 133, in create_module
    raise ImportError('cannot import name %s, '
ImportError: cannot import name GLib, introspection typelib not found
``` 

</p>
</details> 

Anyway, I resolved python/Glib issue by myself, and openrazer-daemon/polychromatic working without errors now, but bug with pooling rate appear again. 

While pooling rate bugged, I tried:
Checked that pooling rate is indeed set at, 8000hz:
```shell
cat /sys/bus/hid/drivers/razer*/*1532*/poll_rate
8000
``` 
`cat /sys/bus/hid/drivers/razer*/*1532*/device_mode` returned nothing.
`echo -n -e "\x03\x00" > /sys/bus/hid/drivers/razer*/*1532*/device_mode` had no effect on this bug. Maybe I'm doing this wrong? Tried also `x00\00`. No effect whatsoever. 

`echo -n "1000" > /sys/bus/hid/drivers/razer*/*1532*/poll_rate` changed value to 1000, but mouse movement very sluggish and `evhz` report 125Hz.
Mouse USB speed reported was 12 Mbit/s while bugged. After replug 480Mbit/s.

Mouse replug once again fixed the issue for the current boot session. Note that mouse was connected for a month on back panel USB and worked correctly. I can try plug again mouse to "Headphone Stand" USB-Hub and see if it fix the issue again, but I will hold for now.

johnjohnlin, 2024-07-14:
Same issue here. I have DA v3, and after rebooting, my DA v3 is working at USB1.0(12M) mode. Re-plugging solves the problem. I also notice that when the bug happens, using RazerGenie still works; however, setting to X Hz **exactly** gives X/8 hz polling rate (reported by evhz). This happens whether I connect my mouse on motherboard, behind one, or two hub.

tfhavingfun, 2024-08-26:
@johnjohnlin Have you updated the [firmware](https://mysupport.razer.com/app/answers/detail/a_id/13817/kw/razer%20deathadder%20v3%20firmware)? It's a known issue for DA v3.

johnjohnlin, 2024-09-05:
@tfhavingfun Thank you! After updating the firmware and testing for a few reboot cycles, I think the firmware is the root cause of this issue.

dm17, 2024-10-15:
I get this as well. I stopped running the razer software because it has no option to configure the buttons, despite deathaddr v3 being on the supported devices list. And because anything over 2,000hz polling causes massive mouse stuttering (no idea why; ideas on how to debug that would be appreciated). Works fine in Windows, which is how I paired the hyperpolling dongle with the DA V3. Even without rebooting sometimes it reverts to 500hz after being set to 2k using razergenie (and confirmed to be 2k hz by evhz). Also curious how anyone has managed to change the keybindings on this "supported" mouse. I'm running the latest firmware on the dongle and mouse - I checked using Synapse yesterday on Windows.

closed 2025-03-19 by github-actions[bot]

lah7, 2025-03-19:
Was this still an issue?

Recently we discovered a BlackWidow V4 "lost" it's higher polling rate (8000 -> 1000) at the time of entering "driver mode", where the poll rate was set before driver mode.

* https://github.com/openrazer/openrazer/pull/2415
* https://github.com/openrazer/openrazer/pull/2416

Maybe this mouse has the same issue? Let me know if you need something to test that theory.

VeilSilence, 2025-03-28:
> Was this still an issue?
> 
> Recently we discovered a BlackWidow V4 "lost" it's higher polling rate (8000 -> 1000) at the time of entering "driver mode", where the poll rate was set before driver mode.
> 
>     * [openrazer-daemon not setting the correct polling rate after setting driver mode #2408](https://github.com/openrazer/openrazer/issues/2408)
> 
>     * [daemon: Add EARLY_DEVICE_MODE variable to fix BlackWidow V4 Pro 8000 Hz poll rate loss #2416](https://github.com/openrazer/openrazer/pull/2416)
> 
> 
> Maybe this mouse has the same issue? Let me know if you need something to test that theory.

Sorry for late respond. Yes, it's still the issue sadly.

I looked at your patch and modified a bit, but sadly without success.

<details><summary>Patch</summary>
<p>

```python
Subject: [PATCH] DeathAdderv3 fix? Backport from
 https://github.com/openrazer/openrazer/pull/2416/files

---
 daemon/openrazer_daemon/hardware/device_base.py | 10 ++++++++++
 daemon/openrazer_daemon/hardware/mouse.py       |  1 +
 2 files changed, 11 insertions(+)

diff --git a/openrazer_daemon/hardware/device_base.py b/openrazer_daemon/hardware/device_base.py
index 1f10fabe..ad314ee1 100644
--- a/openrazer_daemon/hardware/device_base.py
+++ b/openrazer_daemon/hardware/device_base.py
@@ -39,6 +39,7 @@ class RazerDevice(DBusService):
     MATRIX_DIMS = None
     POLL_RATES = None
     DPI_MAX = None
+    EARLY_DEVICE_MODE = False

     WAVE_DIRS = (1, 2)

@@ -322,6 +323,11 @@ class RazerDevice(DBusService):
         if 'get_battery' in self.METHODS:
             self._init_battery_manager()

+        # Some devices should be put into driver mode first to keep their desired persistence
+        if self.EARLY_DEVICE_MODE:
+            self.logger.info('Setting device to "driver" mode. Daemon will handle special functionality')
+            self.set_device_mode(0x03, 0x00)  # Driver mode
+
         self.restore_dpi_poll_rate()
         self.restore_brightness()

@@ -333,6 +339,10 @@ class RazerDevice(DBusService):
                 self.logger.debug("Restoring effect persistence again (dual boot quirk)")
                 self.restore_effect()

+        if not self.EARLY_DEVICE_MODE:
+            self.logger.info('Setting device to "driver" mode. Daemon will handle special functionality')
+            self.set_device_mode(0x03, 0x00)  # Driver mode
+
     def send_effect_event(self, effect_name, *args):
         """
         Send effect event
diff --git a/openrazer_daemon/hardware/mouse.py b/openrazer_daemon/hardware/mouse.py
index d326dc0c..7a3ed08a 100644
--- a/openrazer_daemon/hardware/mouse.py
+++ b/openrazer_daemon/hardware/mouse.py
@@ -1578,6 +1578,7 @@ class RazerDeathAdderV3(__RazerDevice):
     POLL_RATES = [125, 500, 1000, 2000, 4000, 8000]

     DEVICE_IMAGE = "https://dl.razerzone.com/src/6124/6124-1-en-v2.png"
+    EARLY_DEVICE_MODE = True

     DPI_MAX = 30000

--
2.48.1
```

</p>
</details>