openrazer/openrazer #2226: RHEL Community Support Repo No Longer Maintained

Provisional9448, 2024-03-23:
### Distribution

Red Hat Enterprise Linux 9.3

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I found the [community repo](https://github.com/moozhub/yum-repo-mooz) that supports RHEL is no longer being maintained. 
Is it possible to add support for RHEL in the [main repo](https://download.opensuse.org/repositories/hardware:/razer/)?

z3ntu, 2024-03-25:
There's no RHEL 8 or 9 support on build.opensuse.org unfortunately https://github.com/openSUSE/open-build-service/issues/10944

If you're interested in RHEL support for OpenRazer, please feel free to set up builds and maintain the package for OpenRazer!

z3ntu, 2024-03-26:
Ok so it does seem like there's EPEL 9 available https://build.opensuse.org/project/show/Fedora:EPEL:9 Is this a suitable target to build against? I am too disconnect from the RHEL ecosystem to judge.

z3ntu, 2024-04-29:
@Provisional9448 Bump?

closed 2025-07-19 by github-actions[bot]