openrazer/openrazer #2227: Razer Deathadder V2 X Hyperspeed not detected via Blueooth

botsunny, 2024-03-29:
### Distribution

OS: Arch Linux
Kernel: 6.8.2-arch2-1
OpenRazer version: 3.7.0-1

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

My Razer Deathadder V2 X Hyperspeed used to work fine with both Bluetooth and dongle mode on Linux, though it can never connect successfully via Bluetooth on Windows. A few weeks ago I had trouble connecting via Bluetooth on my Linux laptop so I decided to forget the device to pair it again. Now, it doesn't get detected at all, so I am stuck on dongle mode. Every other Bluetooth device works okay with my computer.

z3ntu, 2024-03-29:
So the Bluetooth connection doesn't work at all? Sounds like OpenRazer can't really have much to do with this, all Bluetooth handling is done by the rest of the kernel and the OS.

shellclear, 2024-08-02:
I had a similar issue using bluetooth on fedora 40. The mouse has been working fine until some days ago... now everytime it disconnect or the laptop is restarted I need to unpair/pair again to make it comes back work. I have no idea why it is passing now...

``` 
✔ ❯ 🔌 ❯ [12:43:56 08/02/24] ❯  .../lib/dkms ❯ rpm -qa|grep -Ei "razer|kernel-headers"
openrazer-kernel-modules-dkms-3.8.0-1.1.noarch
openrazer-daemon-3.8.0-1.1.noarch
python3-openrazer-3.8.0-1.1.noarch
openrazer-meta-3.8.0-1.1.noarch
kernel-headers-6.9.4-200.fc40.x86_64
 ✔ ❯ 🔌 ❯ [12:43:59 08/02/24] ❯  .../lib/dkms ❯
```