openrazer/openrazer #2229: Support for razer huntsman v3 pro

mrhoomanwastaken, 2024-03-31:
### Device Info

* Name: razer huntsman v3 pro
* Product Number: RZ03-04970200-R3U1
* Model Number: RZ03-0497

### Device Information
```
Please post the output of
    lsusb -d 1532: -v

Bus 001 Device 010: ID 1532:005c Razer USA, Ltd DeathAdder Elite
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x005c DeathAdder Elite
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer DeathAdder Elite
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
[... 717 lines cut ...]
 35 00 46 FF 00 75 08 95 2F 81 00 C0 05 01 09 00
 A1 01 85 05 09 03 15 00 26 FF 00 35 00 46 FF 00
 75 08 95 2F 81 00 C0 05 01 09 00 A1 01 85 07 09
 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95 2F 81
 00 C0 05 01 09 00 A1 01 85 0B 09 03 15 00 26 FF
 00 35 00 46 FF 00 75 08 95 2F 81 00 C0

001:006:000:DESCRIPTOR         1711858185.225282
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

```

</details>

### Packet Captures

i am ready to capture packets just tell me what info you need.

### Synapse Screenshots
[image]
[image]
[image]

z3ntu, 2024-04-04:
Thanks for opening! If you want you can probably implement support for this device yourself, see https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md . If you can provide some USB packet captures of Razer Synapse on Windows interacting with the device, I could give some hints what code paths you'd need to take in OpenRazer. https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

mrhoomanwastaken, 2024-04-07:
@z3ntu hello,
i have everything ready to capture packets, i just need to know what info you want. ie. what do you want me to do in razer Synapse app, as i do not know what is the proper procedure for that part.

z3ntu, 2024-04-08:
Start Wireshark capture. Open Synapse, click around a bit, changing effects, brightness and whatever you see in the UI that does something on the device. Stop capture. That should be enough :)

halanm, 2024-05-06:
Any updates on this?

Sainan, 2024-07-22:
For what it's worth, on my Huntsman V3 Pro, this works to set the device mode:
```c
struct razer_report report = get_razer_report(0x00, 0x04, 0x02);
report.transaction_id.id = 0x1F;
report.arguments[0] = 0x03; /* Mode */
report.arguments[1] = 0x00;
report.crc = razer_calculate_crc(&report);
```

bior12, 2024-07-25:
Has there been any traction on this? I just picked up a V3 Pro Full size a couple days ago and would love to get this up and running. @mrhoomanwastaken did you ever get around to capturing packets? Or is anyone else actively working on this project? if not I can start.

mrhoomanwastaken, 2024-08-12:
> Has there been any traction on this? I just picked up a V3 Pro Full size a couple days ago and would love to get this up and running. @mrhoomanwastaken did you ever get around to capturing packets? Or is anyone else actively working on this ticket? if not I can start sometime in the next week or so.

hello. I have not started it because I just keep forgetting. also I have school starting so I will not be able to do anything.

AlexAxion, 2024-11-02:
has there been any updates on this

mrhoomanwastaken, 2024-11-11:
> has there been any updates on this

no and im between distros rn and also school.

antarean, 2024-12-15:
Same here. Got the keyboard recently. Got some coding experience.
If someone can let me know what is needed, I will do my best to organize it.

mrhoomanwastaken, 2024-12-17:
hello @antarean and @rarestofheroes 
so from what I understand is that you need 3 things.
1. wireshark can be downloaded via `sudo apt-get install wireshark`
2. Virtualbox from [https://www.virtualbox.org/](url)
3. and a windows dev image it would normally be here [https://developer.microsoft.com/en-us/windows/downloads/virtual-machines/](url) but it is down at the moment

after that all you need to do is boot up windows on the vm and install razer synapse. then just mess around and record the packets via Wireshark.
more info is here https://github.com/openrazer/openrazer/wiki

ps. I am not a maintainer and I barely know how any of this works, so I might be getting it wrong.

mrhoomanwastaken, 2024-12-17:
from what I can tell the drivers are written in c.
I do not know how to program in c only python.

bior12, 2024-12-17:
Hey guys, sorry for going radio silent, work picked up quite a bit, I should have some time to look into this in the new year, it's been quite awhile since working with C but I'm sure it's nothing chatgpt can't help with haha

closed 2025-02-25 by z3ntu