openrazer/openrazer #2230: Unable to Install

StarkCraft, 2024-03-31:
### Distribution

Open Razer Daemon, N/A

### Support

- [NO] Have you tried rebooting?
- [YES] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I am unable to install the necessary files, unfoirtunaty my issue isnt listed in trouble shooting and i am yet to find another user with a similar issue, i simply dont know whats wrong so ill attach the screenshot.

[image]

lah7, 2024-03-31:
From what we tried in #2204, we currently don't know how to install OpenRazer on the Steam Deck.

For your error, SteamOS uses an immutable root filesystem, meaning that it would be overwritten when SteamOS updates. It can be made read/write using these commands:

```
sudo steamos-readonly disable    # to make modifications
sudo steamos-readonly enable     # when finished
```

There is [this Reddit post](https://www.reddit.com/r/SteamDeck/comments/16k0tap/comment/k0uqvx3/) from 7 months ago of commands which you can try - but our help will be limited as we don't have confirmation of _what_ actually works.

StarkCraft, 2024-03-31:
I see.. well thank you for your swift reply, to clarify I have tinkered with dual booting and all that nonsense so my deck is writeable, however if it's something with the steam os branch of arch then there's no helping it, I'll just have to use the windows install to configure my onboard saved profiles for the time being, again thank you very much for the swift reply!

Mte90, 2024-09-25:
I am the author of the guide to install openrazer on the steam deck with a script that does everything (https://daniele.tech/2022/12/how-to-get-openrazer-working-easily-on-steam-deck/) and I am always working on updating.
Let me know if it works for you!

closed 2025-07-16 by github-actions[bot]