openrazer/openrazer #2234: Blackwidow V3 (1532:024E) Media Key not working with open windows

ofma-tinker, 2024-04-14:
Hello.

I'm using linux mint - 21.3 Cinnamon w/ 6.5.0-27-generic kernel and the daily update version of openrazer.

Running xev in the terminal, OS recognizes the media-key presses, but none of the software (spotify linux client, firefox, chrome/chromium) respond to keypresses. This has been going on for at least a month -- the mediakey function used to work fine.

I also have a razer basilisk mouse as well.

z3ntu, 2024-04-14:
You mean like the play/pause/volume-up/volume-down buttons? Or some other buttons?

ofma-tinker, 2024-04-19:
Yes. Sorry for the late response, I was on work travel.

z3ntu, 2024-04-23:
From the product page https://www.razer.com/eu-en/gaming-keyboards/razer-blackwidow-v3 :

[image]

> MULTI-FUNCTION ROLLER WHEEL AND MEDIA KEY

> Configure them to pause, play, skip and tweak everything from brightness to volume—the ultimate convenience as you enjoy your entertainment.

What's this button supposed to do normally? The product page is super vague about any of this.

ofma-tinker, 2024-04-25:
previously I was able to pause, go back a song, and go forward a song in the spotify client in Linux. It also worked in the browser to stop and start videos. Now, it doesn't nothing. Even though the key press is still registered.

z3ntu, 2024-04-25:
How do you do this different functionality using only one button?

Also try `systemctl --user stop openrazer-daemon.service` to see if that changes anything.

ofma-tinker, 2024-04-26:
You use a double or triple press for additionally functionality. Double Press is next song, Triple press in quick succession is previous. The " systemctl --user stop openrazer-daemon.service " didn't work.

z3ntu, 2024-04-27:
Interesting, thanks for sharing!

Could you try this please? https://github.com/openrazer/openrazer/issues/1466#issuecomment-804066364 - the part about patching the driver and loading the new version and then press the button and see if anything appears in dmesg when you do that? Let me know if you have any questions or need more detailed help.

HidingCherry, 2025-03-10:
> You use a double or triple press for additionally functionality. Double Press is next song, Triple press in quick succession is previous. The " systemctl --user stop openrazer-daemon.service " didn't work.

That works.
Although, I use another keyboard, the buttons are the same.
Whenever openrazer-daemon is running, the media keys are not working.
Will this be fixed?
Can I assist you somehow?

z3ntu, 2025-03-15:
@HidingCherry As written above:

> Could you try this please? [#1466 (comment)](https://github.com/openrazer/openrazer/issues/1466#issuecomment-804066364) - the part about patching the driver and loading the new version and then press the button and see if anything appears in dmesg when you do that? Let me know if you have any questions or need more detailed help.

HidingCherry, 2025-03-15:
@z3ntu 
The code has changed.

We have now:
static int razer_raw_event_standard ### line 3525
and
static int razer_raw_event_bitfield ### line 3621

I have applied the changes to the first one.

There you go, I hope you can use it. I also recorded the profile buttons on the back, nearby the 2.4G/off/BT switch.

```
one press - media pause/play

kernel: razerkbd event: size 22 - data
kernel:  razerkbd: 02 cd 00 00 00 00 00 00 00 00 00 00 00 00 00 00
kernel:  razerkbd: 00 00 00 00 00 00
kernel: razerkbd event: size 22 - data
kernel:  razerkbd: 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
kernel:  razerkbd: 00 00 00 00 00 00

double press - media next

kernel: razerkbd event: size 22 - data
kernel:  razerkbd: 02 b5 00 00 00 00 00 00 00 00 00 00 00 00 00 00
kernel:  razerkbd: 00 00 00 00 00 00
kernel: razerkbd event: size 22 - data
kernel:  razerkbd: 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
kernel:  razerkbd: 00 00 00 00 00 00

tripple press - media previous

kernel: razerkbd event: size 22 - data
kernel:  razerkbd: 02 b6 00 00 00 00 00 00 00 00 00 00 00 00 00 00
kernel:  razerkbd: 00 00 00 00 00 00
kernel: razerkbd event: size 22 - data
kernel:  razerkbd: 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
kernel:  razerkbd: 00 00 00 00 00 00

volume press - silent

kernel: razerkbd event: size 22 - data
kernel:  razerkbd: 02 e2 00 00 00 00 00 00 00 00 00 00 00 00 00 00
kernel:  razerkbd: 00 00 00 00 00 00
kernel: razerkbd event: size 22 - data
kernel:  razerkbd: 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
kernel:  razerkbd: 00 00 00 00 00 00

back 1

kernel: razerkbd event: size 22 - data
kernel:  razerkbd: 04 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00
kernel:  razerkbd: 00 00 00 00 00 00
kernel: razerkbd event: size 22 - data
kernel:  razerkbd: 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
kernel:  razerkbd: 00 00 00 00 00 00

back 2

kernel: razerkbd event: size 22 - data
kernel:  razerkbd: 04 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00
kernel:  razerkbd: 00 00 00 00 00 00
kernel: razerkbd event: size 22 - data
kernel:  razerkbd: 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
kernel:  razerkbd: 00 00 00 00 00 00

back 3

kernel: razerkbd event: size 22 - data
kernel:  razerkbd: 04 42 00 00 00 00 00 00 00 00 00 00 00 00 00 00
kernel:  razerkbd: 00 00 00 00 00 00
kernel: razerkbd event: size 22 - data
kernel:  razerkbd: 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
kernel:  razerkbd: 00 00 00 00 00 00
```

HidingCherry, 2025-04-29:
@z3ntu any progress on this?

z3ntu, 2025-04-29:
No sorry, I don't have any time to look into this in more detail right now.

If you like some tinkering, I'm sure you can figure out how these events work on your keyboard and what you may need to change!

HidingCherry, 2025-04-30:
@z3ntu do you have a hint, which keyboard might have those keys (mainly media keys) already working?
I'm looking at [driver/razerkbd_driver.c](https://github.com/openrazer/openrazer/blob/master/driver/razerkbd_driver.c) atm, but I'm completely lost, what I should look for and how it should be correctly set up.

I just need a model, I think I would find the rest myself.

HidingCherry, 2025-04-30:
Alright, I found another issue.
media keys are working, but only in specific circumstances.
I will raise another issue for this, to not mix up things.