openrazer/openrazer #2236: Razer Naga Pro RZ01-0342 mouse wheel scroll down triggering mouse wheel right-tilt click

highhbrid, 2024-04-20:
### Distribution

Ubuntu 23.10 Mantic

### Support

- [ ] Have you tried rebooting?
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Razer Naga Pro RZ01-0342 mouse wheel scroll down is triggering mouse wheel right-tilt click about 1/3 of the time. This does not occur on Windows 10 22H2. Updated device firmware and wireless dongle to v2.03.02_r1 (latest) but did not resolve the issue. Using mouse in wired or wireless mode does not yield a difference. This may have something to do with the front-ends saying "This supported device was not detected by the daemon" but unsure.

Happy to add more information if necessary. Thanks for your support.

lah7, 2024-04-21:
Please try updating to the latest OpenRazer, as yours is a few versions behind. While it might not specifically fix this misclick issue, some of the other fixes may yield different results.

Ubuntu 23.10 ships with [v3.4.0](https://packages.ubuntu.com/mantic/openrazer-daemon).  The dongle was first supported in [v3.6.0](https://github.com/openrazer/openrazer/releases/tag/v3.6.0). There's been [more releases](https://github.com/openrazer/openrazer/releases) too! [Use the "Ubuntu" instructions](https://openrazer.github.io/#ubuntu) to add the stable PPA, then update the system and reboot.

Has this just recently started happening, or has it always happened under Linux?

highhbrid, 2024-04-21:
This is the version I have installed. I am pulling from the stable repository.

openrazer-daemon/mantic,mantic,now 3.8.0~ubuntu23.10.1 all [installed,automatic]
openrazer-driver-dkms/mantic,mantic,now 3.8.0~ubuntu23.10.1 all [installed,automatic]
openrazer-meta/mantic,mantic,now 3.8.0~ubuntu23.10.1 all [installed]

To be honest, I just moved over to linux full-time, but I also just purchased this mouse that replaced a different Razer Naga. I was not experiencing this on the old mouse, but that device is not supported, I believe. Here is the model number of the old mouse, if that helps: RZ01-0241

lah7, 2024-04-21:
`RZ01-0241` comes up as [Razer Naga Trinity](https://mysupport.razer.com/app/answers/detail/a_id/3751/~/razer-naga-trinity-%7C-rz01-0241-support-%26-faqs). If `lsusb | grep 1532` in the terminal outputs `1532:0067` then that would have been supported.

I'm not sure if I'm misinterpreting your original post, sorry. The horizontal wheel tilt isn't actually supported (#201) _- or at least, nothing special happens -_ except for the following mice:
https://github.com/openrazer/openrazer/blob/ed37ec96e6bf5d9e8dba8c9c18cc39e7b6ccf16d/driver/razermouse_driver.c#L4515-L4524

I had a Mamba TE with a similar left/right tilting mechanism, but its behaviour would quickly scroll up or down outside of the official Razer Synapse driver (it only worked under Windows).

Are you saying that when you scroll down, a third of the time it horizontally scrolled to the right instead? Please could you describe what happens in a program? We need to rule out if it's a hardware or software problem.

highhbrid, 2024-05-02:
Sorry for the late response. Yes, I have found that searching for that number does yield results referencing a Trinity. I have attached a picture to show you that is was indeed a Naga Classic
[image]

To be clear, the mouse that was not having the issue is the one in the picture. My new mouse, the Razer Naga Pro (RZ01-0342), is the one having the issue.

About 30% of the time, when scrolling down with the mouse wheel, it will instead trigger the button which is programmed when tilting the mouse wheel rightward. I am not able to replicate this issue in Windows, even one time. I am not using the wireless dongle, only a wired connection with cable to simply the scenario.

Thank you again for your attention on this. It’s really the only thing keeping me from staying Linux for gaming at the moment. I happen to use the scroll wheel and tilt buttons quite heavily in my raiding.

closed 2025-03-18 by github-actions[bot]