openrazer/openrazer #2240: Razer Cobra - Daemon fails to stay active

Shade325, 2024-04-28:
### Distribution

Linux Mint 21.3 Cinnamon

### Support

- [x] Have you tried rebooting? 

- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Since turning my PC back on the Daemon has failed to stay on when I type in 

`openrazer-daemon -Fv`

i get the following:

> Starting daemon.
> Traceback (most recent call last):
>   File "/usr/lib/python3/dist-packages/daemonize.py", line 248, in start
>     self.action(*privileged_action_result)
>   File "/usr/bin/openrazer-daemon", line 141, in run_daemon
>     daemon = RazerDaemon(verbose=args.verbose,
>   File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 91, in __init__
>     self.read_persistence(persistence_file)
>   File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 306, in read_persistence
>     self._persistence.read(persistence_file)
>   File "/usr/lib/python3.10/configparser.py", line 699, in read
>     self._read(fp, filename)
>   File "/usr/lib/python3.10/configparser.py", line 1022, in _read
>     for lineno, line in enumerate(fp, start=1):
>   File "/usr/lib/python3.10/codecs.py", line 322, in decode
>     (result, consumed) = self._buffer_decode(data, self.errors, final)
> UnicodeDecodeError: 'utf-8' codec can't decode byte 0x9d in position 3: invalid start byte
> 
> Stopping daemon.
> Exception ignored in atexit callback: <bound method Daemonize.exit of <daemonize.Daemonize object at 0x7f2ffb436e90>>
> Traceback (most recent call last):
>   File "/usr/lib/python3/dist-packages/daemonize.py", line 72, in exit
>     sys.exit(0)
> SystemExit: 0

I'm at a complete loss, and would be thankful for any help. Thank you.

z3ntu, 2024-04-28:
`rm ~/.config/openrazer/persistence.conf` should fix this

Can you share what `openrazer-daemon --version` reports?

Shade325, 2024-04-28:
> `rm ~/.config/openrazer/persistence.conf` should fix this
> 
> Can you share what `openrazer-daemon --version` reports?

well what you do you know, it worked thank you:

it said the following: 

`openrazer-daemon 3.8.0`

okay i'll close this. and bookmark this for if it happens again.

z3ntu, 2024-04-28:
At some point we had a bug where this happened to more users but you seem to be the first person in a while to hit this apparently but we should probably still just delete the file when we fail to parse it and not just crash completely - since the file is not too important anyways if we lose the data once in a while.

z3ntu, 2024-04-29:
I see we've already added some "workaround" in https://github.com/openrazer/openrazer/commit/74b5f642a5619fe7977adccb72c30847696c52f3 but for you it was crashing with `UnicodeDecodeError` so this exception wasn't caught.

commit 72db40bf (2024-04-29): daemon: Also handle UnicodeDecodeError for persistence file fail

Fixes #2240
closed 2024-04-29 by z3ntu