openrazer/openrazer #2242: Support for razer blade 14 (2024)

jetoile, 2024-04-28:
### Device Info

* Name: Razer Blade 14 2024
* Product Number: Blade 14 - RZ09-0508
* Model Number: -

### Device Information
```
Bus 001 Device 004: ID 1532:02b6 Razer USA, Ltd Razer Blade
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x02b6 Razer Blade
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Blade
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x006d
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      1 Keyboard
      iInterface              2 Razer Blade
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     327
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval               1
Device Status:     0x0010
  (Bus Powered)
```

```
lrwxrwxrwx 1 root root  10 avril 29 00:06 usb-Razer_Razer_Blade-event-kbd -> ../event18
lrwxrwxrwx 1 root root  10 avril 29 00:06 usb-Razer_Razer_Blade-if01-event-kbd -> ../event20
lrwxrwxrwx 1 root root  10 avril 29 00:06 usb-Razer_Razer_Blade-if02-event-mouse -> ../event15
lrwxrwxrwx 1 root root   9 avril 29 00:06 usb-Razer_Razer_Blade-if02-mouse -> ../mouse3
```

<details>

```
001:004:003:DESCRIPTOR         1714343491.199943
 05 59 09 01 A1 01 85 01 09 02 A1 02 09 03 15 00
 27 FF FF 00 00 75 10 95 01 B1 03 09 04 09 05 09
 06 09 07 09 08 15 00 27 FF FF FF 7F 75 20 95 05
 B1 03 C0 85 02 09 20 A1 02 09 21 15 00 27 FF FF
 00 00 75 10 95 01 B1 02 C0 85 03 09 22 A1 02 09
 21 15 00 27 FF FF 00 00 75 10 95 01 B1 02 09 23
 09 24 09 25 09 27 09 26 15 00 27 FF FF FF 7F 75
 20 95 05 B1 02 09 28 09 29 09 2A 09 2B 09 2C 09
 2D 15 00 26 FF 00 75 08 95 06 B1 02 C0 85 04 09
 50 A1 02 09 03 09 55 15 00 25 08 75 08 95 02 B1
 02 09 21 15 00 27 FF FF 00 00 75 10 95 08 B1 02
 09 51 09 52 09 53 09 54 09 51 09 52 09 53 09 54
 09 51 09 52 09 53 09 54 09 51 09 52 09 53 09 54
 09 51 09 52 09 53 09 54 09 51 09 52 09 53 09 54
 09 51 09 52 09 53 09 54 09 51 09 52 09 53 09 54
 15 00 26 FF 00 75 08 95 20 B1 02 C0 85 05 09 60
 A1 02 09 55 15 00 25 08 75 08 95 01 B1 02 09 61
 09 62 15 00 27 FF FF 00 00 75 10 95 02 B1 02 09
 51 09 52 09 53 09 54 15 00 26 FF 00 75 08 95 04
 B1 02 C0 85 06 09 70 A1 02 09 71 15 00 25 01 75
 08 95 01 B1 02 C0 C0

001:004:002:DESCRIPTOR         1714343491.203107
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 06 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0

001:004:001:DESCRIPTOR         1714343491.205928
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 0E 81 00 C0 05 0C 09 01 A1 01 85
 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75 10 81
 00 75 08 95 0D 81 01 C0 05 01 09 80 A1 01 85 03
 19 81 29 83 15 00 25 01 75 01 95 03 81 02 95 05
 81 01 75 08 95 0E 81 01 C0 05 01 09 00 A1 01 85
 04 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 05 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0

001:004:000:DESCRIPTOR         1714343491.208934
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0
```

</details>

fallenpixel, 2024-04-28:
Was able to compile and test on Archlinux on my Razer Blade 2024 and can confirm this works for the most part. When using a Custom editor the left alt key does not illuminate, but it does work when setting a static color or effect. Also, the options include an option for "Logo Lighting" despite the device having no logo to illuminate. Behavior gets a little wonky when changing the logo lighting options.

z3ntu, 2024-04-29:
@fallenpixel Let's take this to #2241 to keep the conversation in one place.

PR #2241 (2024-07-06): Add support for Razer Blade 14 (2024)
Changes based on #2237 .

Add support for new razer blade 14 (2024)

```bash
lsusb | grep '1532:'
Bus 001 Device 003: ID 1532:02b6 Razer USA, Ltd Razer Blade
```

Everything seems to work fine at least on my laptop (ubuntu 24.04)

Fixes #2242
closed 2024-07-06 by z3ntu