openrazer/openrazer #2244: Support for [device name here]

chris50080, 2024-05-02:
### Device Info

* Name: Razer Naga v2 pro
* Product Number: RZ01-04400100
* Model Number: RZ01-0440

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
Bus 005 Device 002: ID 1532:00a8 Razer USA, Ltd Razer Naga V2 Pro
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x00a8 Razer Naga V2 Pro
  bcdDevice            1.00
  iManufacturer           1 Razer
  iProduct                2 Razer Naga V2 Pro
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     100
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     314
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)

```
Please post the output of
    ls -lah /dev/input/by-id/
```
total 0
drwxr-xr-x 2 root root 240 May  1 19:03 .
drwxr-xr-x 4 root root 600 May  1 19:03 ..
lrwxrwxrwx 1 root root   9 May  1 19:03 usb-Logitech_Gaming_Keyboard_G910_0686386B3231-event-kbd -> ../event8
lrwxrwxrwx 1 root root   9 May  1 19:03 usb-Logitech_Gaming_Keyboard_G910_0686386B3231-if01-event-kbd -> ../event9
lrwxrwxrwx 1 root root  10 May  1 19:03 usb-MSI_MYSTIC_LIGHT_7D7522120106-event-if00 -> ../event10
lrwxrwxrwx 1 root root   9 May  1 19:03 usb-Razer_Razer_Naga_V2_Pro-event-if01 -> ../event6
lrwxrwxrwx 1 root root   9 May  1 19:03 usb-Razer_Razer_Naga_V2_Pro-event-mouse -> ../event3
lrwxrwxrwx 1 root root   9 May  1 19:03 usb-Razer_Razer_Naga_V2_Pro-if01-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 May  1 19:03 usb-Razer_Razer_Naga_V2_Pro-if01-event-mouse -> ../event5
lrwxrwxrwx 1 root root   9 May  1 19:03 usb-Razer_Razer_Naga_V2_Pro-if01-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 May  1 19:03 usb-Razer_Razer_Naga_V2_Pro-if02-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 May  1 19:03 usb-Razer_Razer_Naga_V2_Pro-mouse -> ../mouse0

<details>

```
Please post the output of
    usbhid-dump -m 1532 -ed
```
total 0
drwxr-xr-x 2 root root 240 May  1 19:03 .
drwxr-xr-x 4 root root 600 May  1 19:03 ..
lrwxrwxrwx 1 root root   9 May  1 19:03 usb-Logitech_Gaming_Keyboard_G910_0686386B3231-event-kbd -> ../event8
lrwxrwxrwx 1 root root   9 May  1 19:03 usb-Logitech_Gaming_Keyboard_G910_0686386B3231-if01-event-kbd -> ../event9
lrwxrwxrwx 1 root root  10 May  1 19:03 usb-MSI_MYSTIC_LIGHT_7D7522120106-event-if00 -> ../event10
lrwxrwxrwx 1 root root   9 May  1 19:03 usb-Razer_Razer_Naga_V2_Pro-event-if01 -> ../event6
lrwxrwxrwx 1 root root   9 May  1 19:03 usb-Razer_Razer_Naga_V2_Pro-event-mouse -> ../event3
lrwxrwxrwx 1 root root   9 May  1 19:03 usb-Razer_Razer_Naga_V2_Pro-if01-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 May  1 19:03 usb-Razer_Razer_Naga_V2_Pro-if01-event-mouse -> ../event5
lrwxrwxrwx 1 root root   9 May  1 19:03 usb-Razer_Razer_Naga_V2_Pro-if01-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 May  1 19:03 usb-Razer_Razer_Naga_V2_Pro-if02-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 May  1 19:03 usb-Razer_Razer_Naga_V2_Pro-mouse -> ../mouse0
[chris@arch-pc ~]$ sudo usbhid-dump -m 1532 -ed
005:002:002:DESCRIPTOR         1714608777.440856
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

005:002:001:DESCRIPTOR         1714608777.443845
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 0E 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0 05 01 09
 02 A1 01 85 0A 09 01 A1 00 05 09 19 01 29 06 15
 00 25 01 75 01 95 06 81 02 75 01 95 02 81 03 06
 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05 01
 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09 31
 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 C0 05 0C
 09 01 A1 01 85 02 19 00 2A 3C 02 15 00 26 3C 02
 95 01 75 10 81 00 75 08 95 0D 81 01 C0 05 01 09
 80 A1 01 85 03 19 81 29 83 15 00 25 01 75 01 95
 03 81 02 95 05 81 01 75 08 95 0E 81 01 C0 05 01
 09 00 A1 01 85 04 09 03 15 00 26 FF 00 35 00 46
 FF 00 75 08 95 0F 81 00 C0 05 01 09 00 A1 01 85
 05 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 08 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0 05
 01 09 00 A1 01 85 09 09 03 15 00 26 FF 00 35 00
 46 FF 00 75 08 95 0F 81 00 C0

005:002:000:DESCRIPTOR         1714608777.446849
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 75 01 95 08 81 03 05 0C 0A 38 02 15 81 25 7F 75
 08 95 01 81 06 05 01 09 38 15 81 25 7F 75 08 95
 01 81 06 09 30 09 31 16 00 80 26 FF 7F 75 10 95
 02 81 06 C0 06 00 FF 09 02 15 00 25 01 75 08 95
 5A B1 01 C0
</details>

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

chris50080, 2024-05-02:
Sorry didn't realize this has already been open for 2 years..

closed 2024-05-02 by chris50080

z3ntu, 2024-05-02:
Yep, duplicate of #1952