openrazer/openrazer #2247: Spurious battery level notification

NeilGirdhar, 2024-05-10:
Ubuntu 24.04
openrazer-daemon: 3.7.0+dfsg-1

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Getting battery notifications despite:
Battery level is 41%
razer.conf shows:
```
# Battery notification frequency [s] (0 to disable)
battery_notifier_freq = 600

# Battery notifications appear when device reaches this percentage
battery_notifier_percent = 33
```

z3ntu, 2024-05-10:
The percentage from the config is read at https://github.com/openrazer/openrazer/blob/72db40bf0290c998655b8558f7ed5944729e3780/daemon/openrazer_daemon/hardware/device_base.py#L1066 and then used in https://github.com/openrazer/openrazer/blob/72db40bf0290c998655b8558f7ed5944729e3780/daemon/openrazer_daemon/misc/battery_notifier.py#L110 , so not quite sure where it could be going wrong.

Also are you sure the message is what you wrote and not "Battery is 41%"? That would be the message from the daemon.

NeilGirdhar, 2024-05-10:
@z3ntu Thanks for taking a look.  The thing is that the alert that I get says I have 0%, but the battery level showing in the controller says 94%.

Is there a way for me to install the openrazor Python library so that I can check your code?  Trying to use it in the normal way:
```python
File ~/.pyenv/versions/3.12.3/lib/python3.12/site-packages/openrazer/__init__.py:1
----> 1 raise RuntimeError("Should never be installed")
```

z3ntu, 2024-05-10:
> The thing is that the alert that I get says I have 0%

Ah, the 0% issue, I thought you said you get a popup that the battery is at 41% even though the config says only notifications below 33%.

See https://github.com/openrazer/openrazer/issues/2203#issuecomment-1925411515 and https://github.com/openrazer/openrazer/issues/2122 then.

z3ntu, 2024-05-10:
Also the "openrazer" package on PyPI is just a dummy package, you shouldn't install it from there - that's what that RuntimeError is saying.

NeilGirdhar, 2024-05-10:
> Ah, the 0% issue, I thought you said you get a popup that the battery is at 41% even though the config says only notifications below 33%.

Sorry, I should have been clearer.  Can we just ignore 0% battery level if it's sporadic?  What do you think my PR?

> Also the "openrazer" package on PyPI is just a dummy package, you shouldn't install it from there - that's what that RuntimeError is saying.

Right. Should I just install it from github then?

commit b996ba4c (2024-07-02): daemon: Ignore bogus 0% battery level when notifying

Due to various issues in the stack sometimes we get bogus 0% battery
percentage and showing a notification for that is pretty useless, so
ignore that.

[Luca: refactor a bit]

Fixes #2247
closed 2024-07-02 by z3ntu