openrazer/openrazer #2251: Support for Viper V3 Pro

hadronized, 2024-05-21:
### Device Info

* Name: <See under device>
* Product Number: <See under device>
* Model Number: <See under device>

### Device Information
```
Bus 003 Device 002: ID 1532:00c1 Razer USA, Ltd Razer Viper V3 Pro
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x00c1 Razer Viper V3 Pro
  bcdDevice            1.01
  iManufacturer           1 Razer
  iProduct                2 Razer Viper V3 Pro
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              2
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
          Report Descriptors:
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      0
      iInterface              2
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     260
          Report Descriptors:
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               4
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              2
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
          Report Descriptors:
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               4
```

```
Please post the output of
lrwxrwxrwx 1 root root   9 May 21 17:54 usb-Razer_Razer_Viper_V3_Pro-event-if01 -> ../event7
lrwxrwxrwx 1 root root   9 May 21 17:54 usb-Razer_Razer_Viper_V3_Pro-event-mouse -> ../event4
lrwxrwxrwx 1 root root   9 May 21 17:54 usb-Razer_Razer_Viper_V3_Pro-if01-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 May 21 17:54 usb-Razer_Razer_Viper_V3_Pro-if01-event-mouse -> ../event6
lrwxrwxrwx 1 root root   9 May 21 17:54 usb-Razer_Razer_Viper_V3_Pro-if01-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 May 21 17:54 usb-Razer_Razer_Viper_V3_Pro-if02-event-kbd -> ../event8
lrwxrwxrwx 1 root root   9 May 21 17:54 usb-Razer_Razer_Viper_V3_Pro-mouse -> ../mouse0
```

<details>

```
sudo usbhid-dump -m 1532:00c1 -ed
003:002:002:DESCRIPTOR         1716310546.381398
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

003:002:001:DESCRIPTOR         1716310546.381592
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 0E 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0 05 01 09
 02 A1 01 85 0A 09 01 A1 00 05 09 19 01 29 06 15
 00 25 01 75 01 95 06 81 02 75 01 95 02 81 03 06
 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05 01
 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09 31
 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 C0 05 0C
 09 01 A1 01 85 02 19 00 2A 3C 02 15 00 26 3C 02
 95 01 75 10 81 00 75 08 95 0D 81 01 C0 05 01 09
 80 A1 01 85 03 19 81 29 83 15 00 25 01 75 01 95
 03 81 02 95 05 81 01 75 08 95 0E 81 01 C0 05 01
 09 00 A1 01 85 04 09 03 15 00 26 FF 00 35 00 46
 FF 00 75 08 95 0F 81 00 C0 05 01 09 00 A1 01 85
 05 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0

003:002:000:DESCRIPTOR         1716310546.382092
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 06 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 01 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0
```

</details>

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

hadronized, 2024-05-21:
I can try to work on that BTW, I’m interested.

z3ntu, 2024-06-05:
Thanks for opening!

This document https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md should tell you how it's possible to add support for this device and open a pull request. And in case you can take packet captures on Windows with Razer Synapse (https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) I can give some hints which `case` statements need to be used inside the razermouse kernel driver.

jasonphi, 2024-06-24:
Wireless: 1532:00c1
Wired: 1532:00c0

Here are some packet captures.

[viper-v3-wired.txt](https://github.com/user-attachments/files/15949053/viper-v3-wired.txt)
[viper-v3-wireless.txt](https://github.com/user-attachments/files/15949054/viper-v3-wireless.txt)

I have a branch with the required changes. One thing I'm curious about, though: in `razer_chroma_misc_set_dpi_xy` DPI is currently clamped to 30,000 but the Viper v3 Pro supports 35,000. Should I just raise the clamped value, create a new function, or parameterize the max DPI?

https://github.com/openrazer/openrazer/blob/67815f736750c189e0fb4c9de1c2b52fbbc4c078/driver/razerchromacommon.c#L1216-L1217

closed 2024-07-02 by z3ntu