openrazer/openrazer #2252: Kernel module is not loading

SoulRaven, 2024-05-22:
OS: Pop!_OS 22.04 LTS
kernel: Linux version 6.8.0-76060800daily20240311-generic (jenkins@warp.pop-os.org) (x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #202403110203~1715181801~22.04~aba43ee SMP PREEMPT_DYNAMIC Wed M

i am using: Polychromatic 0.8.5 [Stable/jammy main]
the latest version 0.9.1 is not available yet on Jammy

i have try to reboot and all the standard procedure, is not working,
the kernel module is not loaded and throws this error:

> sudo modprobe -f razerkbd
> modprobe: ERROR: could not insert 'razerkbd': Exec format error

also the dmesg shows

> [ 1101.962662] razerkbd: disagrees about version of symbol hid_unregister_driver
> [ 1101.962667] razerkbd: Unknown symbol hid_unregister_driver (err -22)
> [ 1101.962677] razerkbd: disagrees about version of symbol hid_hw_start
> [ 1101.962678] razerkbd: Unknown symbol hid_hw_start (err -22)
> [ 1101.962691] razerkbd: disagrees about version of symbol __hid_register_driver
> [ 1101.962692] razerkbd: Unknown symbol __hid_register_driver (err -22)
> [ 1101.962704] razerkbd: disagrees about version of symbol hid_hw_stop
> [ 1101.962706] razerkbd: Unknown symbol hid_hw_stop (err -22)
> [ 1101.962733] razerkbd: disagrees about version of symbol hid_report_raw_event
> [ 1101.962734] razerkbd: Unknown symbol hid_report_raw_event (err -22)
> [ 1101.962750] razerkbd: disagrees about version of symbol hid_open_report
> [ 1101.962751] razerkbd: Unknown symbol hid_open_report (err -22)

any idea how to fix this?
from what i have found, the module needs to be compiled using the latest kernel version or something similar.

z3ntu, 2024-05-22:
Looks like you need to get dkms to rebuild the module against the kernel you have installed, can't really judge what went wrong there.

Try this?
```
sudo dkms remove openrazer-driver/3.8.0 --all
sudo dkms install openrazer-driver/3.8.0
```

SoulRaven, 2024-05-22:
yes, fixed.
good to know for the future.

z3ntu, 2024-05-22:
Good to hear!

closed 2024-05-22 by z3ntu