openrazer/openrazer #2253: No work my mouse

AlexMoon556, 2024-05-23:
### Distribution

### Support

- [ ] Have you tried rebooting?
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

TQQEU, 2024-05-23:
Hi @AlexMoon556, 

welcome to Github!
Please provide some helpful information (at least the model-name / -number of your device!) when opening an issue.

As you checked that you read the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page: What were your outputs?

AlexMoon556, 2024-05-23:
Hey my device Bus 001 Device 006: ID 1532:0098 Razer USA, Ltd Razer
DeathAdder Essential

AlexMoon556, 2024-05-23:
> Hi @AlexMoon556,
> 
> welcome to Github! Please provide some helpful information (at least the model-name / -number of your device!) when opening an issue.
> 
> As you checked that you read the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page: What were your outputs?

I tried all the applications but nothing changes, maybe I don’t have the drivers, although everything sees fine and it also doesn’t see my Barracuda X headphones

z3ntu, 2024-06-01:
What's the output of `sudo modprobe razerkbd` ?

closed 2025-03-16 by github-actions[bot]