openrazer/openrazer #2255: No device was detected

f0rks7331n, 2024-05-28:
### Distribution
Pop!_OS 22.04 LTS x86_64

### Razer Version
RazerGenie v1.1.0
OpenRazerDaemon v3.8.0

### Support

- [ Yes] Have you tried rebooting?
- [ Yes] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Hello there, 2 days ago my razer app worked perfect .. i've been cleaning my laptop ( that means i took out my battery ), after that razer app isn't recognize any device.I have 2 device's that appear like they are connected : 
Bus 003 Device 006: ID 1532:006e Razer USA, Ltd DeathAdder Essential
Bus 003 Device 004: ID 1532:024e Razer USA, Ltd Razer BlackWidow V3
[image]

z3ntu, 2024-05-28:
What does `sudo dkms status` say?

f0rks7331n, 2024-05-28:
> What does `sudo dkms status` say?

openrazer-driver/3.8.0, 6.8.0-76060800daily20240311-generic, x86_64: installed

f0rks7331n, 2024-05-28:
> And `sudo modprobe razerkbd`?
modprobe: ERROR: could not insert 'razerkbd': Invalid argument

z3ntu, 2024-05-28:
Check dmesg then. But probably a combination of 
`sudo dkms remove openrazer-driver/3.8.0 --all` and 
`sudo dkms install openrazer-driver/3.8.0` should fix it.

f0rks7331n, 2024-05-28:
> Check dmesg then. But probably a combination of `sudo dkms remove openrazer-driver/3.8.0 --all` and `sudo dkms install openrazer-driver/3.8.0` should fix it.

Ty bro .. it worked! Have a good night :100: