openrazer/openrazer #2258: Support for [Razer Blackshark V2 Pro]

MonadMania, 2024-06-02:
### Device Info

* Name: <Razer BlackShark V2 Pro>
* Product Number: <1532:0528>
* Model Number: <IPIG-KOR-210726GY>

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
Bus 001 Device 004: ID 1532:0528 Razer USA, Ltd Razer BlackShark V2 Pro
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0528 Razer BlackShark V2 Pro
  bcdDevice            1.00
  iManufacturer           3 Razer
  iProduct                4 Razer BlackShark V2 Pro
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0114
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              1 Razer BlackShark V2 Pro
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0047
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             6
        wTerminalType      0x0201 Microphone
        bAssocTerminal          0
        bNrChannels             1
        wChannelConfig     0x0000
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
[... 285 lines cut ...]
 00 FF 09 22 A1 02 85 90 09 33 95 01 91 02 09 34
 91 02 05 01 09 3B 75 10 27 FF FF 00 00 91 02 06
 00 FF 09 0A A1 02 05 0A 19 01 29 0A 95 0A 75 08
 26 FF 00 91 02 C0 C0 06 00 FF 09 22 A1 02 85 C4
 09 33 95 01 81 02 09 34 81 02 05 01 09 3B 75 10
 27 FF FF 00 00 81 02 06 00 FF 09 0A A1 02 05 0A
 19 01 29 3B 95 3B 75 08 26 FF 00 81 02 C0 C0 06
 00 FF 09 22 A1 02 85 C4 09 33 95 01 91 02 09 34https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol
 91 02 05 01 09 3B 75 10 27 FF FF 00 00 91 02 06
 00 FF 09 0A A1 02 05 0A 19 01 29 3B 95 3B 75 08
 26 FF 00 91 02 C0 C0 06 00 FF 09 0C A1 02 85 FF
 05 01 09 3B 95 01 B1 02 09 3A A1 02 05 0A 19 01
 29 3E 95 3E B1 02 C0 C0 06 00 FF 09 0C A1 02 85
 E2 05 01 09 3B 95 01 81 02 09 3A A1 02 05 0A 19
 01 29 3E 95 3E 81 02 C0 C0 C0

</details>

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

makarov-roman, 2024-06-03:
there is https://github.com/openrazer/openrazer/issues/1280
but 2020 and 2023 versions are technically difference headsets, so I'm not sure if it makes sense to have a single issue for both of them

z3ntu, 2024-06-05:
Duplicate of #1280

closed 2024-06-05 by z3ntu