openrazer/openrazer #2263: Supported device no longer detected, invalid argument

soratoyuki, 2024-06-17:
Pop!_OS 22.04 LTS
Openrazer 3.8.0
polychromatic 0.8.5.

Yes I've restarted, yes I've gone through the troubleshooter, and no, I do not have secure boot enabled.  And unfortunately, I'm somewhat of a Linux noob.

I currently have a Naga Pro Wireless mouse, which worked fine with until recently. I opened up Polychromatic for the first time in a few weeks to change my bindings, and am greeted with "This supported device wasn't detected by the OpenRazer deamon v.3.8.0." Running the built in troubleshooter points to "Failed: Kernal Module is loaded." Running $sudo modprobe razermouse returns: 

modprobe: ERROR: could not insert 'razermouse': Invalid argument

I've likewise followed the tutorial webpage and everything seems fine. Running $sudo dkms install openrazer-driver/3.8.0
 returns

Module openrazer-driver/3.8.0 already installed on kernel 6.8.0-76060800daily20240311-generic (x86_64).

as expected, I think.  However, I don't see that Invalid argument response anywhere in troubleshooting guide, so I'm stuck as to how to proceed. Any help would be appreciated.

z3ntu, 2024-06-17:
Please check `dmesg` for any output relating to `razermouse`, that should have an error message.

olifluous, 2024-06-17:
He is not the only one, same here;
```
sudo modprobe razermouse
modprobe: ERROR: could not insert 'razermouse': Invalid argument
```

# outputs:
## general info
```
lsusb | grep 1532
Bus 001 Device 004: ID 1532:009c Razer USA, Ltd Razer DeathAdder V2 X HyperSpeed
```
```
uname -r
6.8.0-76060800daily20240311-generic

lsb_release -a
No LSB modules are available.
Distributor ID:	Pop
Description:	Pop!_OS 22.04 LTS
Release:	22.04
Codename:	jammy
```

## checking dkms
```
sudo dkms status

openrazer-driver/3.8.0, 6.5.0-35-generic, x86_64: installed
openrazer-driver/3.8.0, 6.8.0-76060800daily20240311-generic, x86_64: installed
system76/1.0.14~1707324875~22.04~da04fd6, 6.5.0-35-generic, x86_64: installed
system76/1.0.14~1707324875~22.04~da04fd6, 6.5.6-76060506-generic, x86_64: installed
system76/1.0.14~1707324875~22.04~da04fd6, 6.8.0-76060800daily20240311-generic, x86_64: installed
system76_acpi/1.0.2~1716558197~22.04~a2020c4, 6.5.0-35-generic, x86_64: installed
system76_acpi/1.0.2~1716558197~22.04~a2020c4, 6.8.0-76060800daily20240311-generic, x86_64: installed (original_module exists)
system76-io/1.0.3~1707324885~22.04~3dd4c32, 6.5.0-35-generic, x86_64: installed
system76-io/1.0.3~1707324885~22.04~3dd4c32, 6.5.6-76060506-generic, x86_64: installed
system76-io/1.0.3~1707324885~22.04~3dd4c32, 6.8.0-76060800daily20240311-generic, x86_64: installed
```

```
sudo dkms install openrazer-driver/3.8.0
Module openrazer-driver/3.8.0 already installed on kernel 6.8.0-76060800daily20240311-generic (x86_64).
```
#### driver is installed for the latest kernel

## kernel headers are installed
`linux-headers-generic` and `linux-headers-generic-hwe-22.04` have been installed
and user is member of the plugdev group.

## currently loaded modules
```
lsmod | grep razer
<no output>
```

## Kernel module log
```
sudo dmesg | grep razer

[    6.961832] razermouse: disagrees about version of symbol hid_unregister_driver
[    6.961838] razermouse: Unknown symbol hid_unregister_driver (err -22)

this message repeats all the time and seems to be pointing to an issue, but I just wasn't able to duckduck a solution yet
```

## razer_mount log
```
sudo journalctl -b | grep -i razer_mount
Jun 17 20:54:47 digby razer_mount[1299]: Driver razermouse
[... 53 lines cut ...]
2024-05-22 21:23:55 | razer.device0                  | INFO     | Resuming RazerDeathAdderV2XHyperSpeed
2024-05-22 22:55:30 | razer.device0                  | INFO     | Suspending RazerDeathAdderV2XHyperSpeed
2024-05-22 22:56:40 | razer.device0                  | INFO     | Resuming RazerDeathAdderV2XHyperSpeed
2024-05-23 00:11:55 | razer                          | INFO     | Initialising Daemon (v3.8.0). Pid: 2516
2024-05-23 00:11:55 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2024-05-23 00:11:55 | razer                          | INFO     | Found device.0: 0003:1532:009C.0002
2024-05-23 00:11:55 | razer.device0                  | INFO     | Initialising device.0 RazerDeathAdderV2XHyperSpeed
2024-05-23 00:11:56 | razer.device0.batterynotifier  | WARNING  | Failed to init notification daemon, err: org.freedesktop.DBus.Error.ServiceUnknown: The name is not activatable

this message is spewed all the time
```

## list sysfs permissions
```
ls -al /sys/bus/hid/drivers/razer*/*1532*/
ls: cannot access '/sys/bus/hid/drivers/razer*/*1532*/': No such file or directory
```
```
ls /sys/bus/hid/drivers/
hid-generic  playstation
```

I hope this info can help both @soratoyuki as myself.

if I am hijacking this issue, let me know and I'll create a new one.

z3ntu, 2024-06-17:
@olifluous Definitely hijacking this thread but for you running `sudo dkms remove openrazer-driver/3.8.0 --all` and 
`sudo dkms install openrazer-driver/3.8.0` should fix it.

olifluous, 2024-06-18:
@z3ntu i can confirm this resolves the issue.

@soratoyuki please try this as well.

I wasn't sure this was a hijacking, as @soratoyuki and I are running the same OS, same OS version, same kernel version, same openrazer and same polychromatic version.
AND we both have the same error message. (A lot of overlap;))

That's why I decided to add my error logs to this issue, instead of creating a new one, as chances are high, it ends up being duplicates. 

So, thanks again, for your solution, it worked flawlessly.

soratoyuki, 2024-06-18:
That worked for me as well. Thank you!

closed 2024-06-18 by soratoyuki