openrazer/openrazer #2265: [Feature Request] Add ability to change extra mouse buttons

NGSpace, 2024-06-22:
On my mouse I have 2 extra buttons and I want to put them to good use, either make them run a shell script or make them spam left click.

I can not use a normal input mapper since linux doesn't actually recognize those buttons as seperate buttons. one button does nothing while the other does a right click (making it indistinguishable from a normal right click)

z3ntu, 2024-07-02:
Which mouse do you have?

But for getting anything that the device does not send on its own we need to switch the device to "driver mode" (instead of default "device mode") and then we need to handle a bunch things more ourselves - but it also allows us to handle more things like that.

NGSpace, 2024-07-02:
I have a "mamba elite" mouse

z3ntu, 2024-07-02:
If you want you can already try this:
```
cd $(dirname /sys/bus/hid/drivers/razermouse/0*/device_serial)
echo -n -e "\x03\x00" > device_mode
```
Then the device should be in "driver mode", do any extra button presses appear there already? Also would be interesting if anything else from the mouse stops working as you'd expect (minus the DPI buttons I think, we don't have those handled as far as I know)

NGSpace, 2024-07-03:
Yes it does now appear as "EXTRA" and "SIDE" buttons. Is there a way to make it happen during startup?

z3ntu, 2024-07-03:
Right now we don't have something to set it on startup, you could make a custom login script or something that changes the mode for you.

For the other buttons and functionality on the mouse, do they change functionality or stop working when you set the `device_mode`? Like DPI buttons, pressing the scroll wheel to the left/right, etc?

NGSpace, 2024-07-03:
The DPI buttons do not seem to send anything, my mouse does not have the ability to move the scroll wheel left or right.

All functionality on my mouse except the DPI buttons work.

z3ntu, 2024-07-03:
I actually also have the Mamba Elite (RZ01-0256) and when I press the scroll wheel left or right by default it fast scrolls up and down as long as I hold the button.

But good to know the rest works for you, thanks!

closed 2025-03-16 by github-actions[bot]