openrazer/openrazer #2268: Support for Razer Kiyo X

luigicampanaro, 2024-06-25:
### Device Info

* Name: Razer Kiyo X
* Product Number: RZ19-0417
* Model Number: RZ19-04170100-R3M1

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
$     lsusb -d 1532: -v

Bus 001 Device 020: ID 1532:0e06 Razer USA, Ltd Razer Kiyo X
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0e06 
  bcdDevice            0.46
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x03dd
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass         14 Video
      bFunctionSubClass       3 Video Interface Collection
      bFunctionProtocol       0 
      iFunction               4 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      1 Video Control
      bInterfaceProtocol      0 
      iInterface              4 
      VideoControl Interface Descriptor:
        bLength                13
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdUVC               1.00
        wTotalLength       0x006d
        dwClockFrequency       48.000000MHz
        bInCollection           1
        baInterfaceNr( 0)       1
      VideoControl Interface Descriptor:
        bLength                18
[... 923 lines cut ...]
Please post the output of
    ls -lah /dev/input/by-id/
```
$     ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root  60 Jun 25 19:59 .
drwxr-xr-x 4 root root 620 Jun 25 19:59 ..
lrwxrwxrwx 1 root root  10 Jun 25 19:45 usb-Razer_Inc_Razer_Kiyo_X_01.00.00-event-if00 -> ../event26

```
Please post the output of
    usbhid-dump -m 1532 -ed
```

$     usbhid-dump -m 1532 -ed
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/020: Permission denied
libusb: error [_get_usbfs_fd] libusb requires write access to USB device nodes.
Failed to find and open the devices: Access denied (insufficient permissions)
### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
No Windows Environment available atm

z3ntu, 2024-07-02:
Hi, thanks for opening the issue!

What specific functionality that's not just regular webcam usage would you expect to be able to manage with OpenRazer? At least for the Kiyo Pro (#1665) there doesn't seem to be anything to manage.

luigicampanaro, 2024-07-03:
Hello, thanks for the kind reply!

My main concern was controlling the fps and other settings, but I solved with  `v4l2-ctl`. So it could still be convenient to have a GUI, but settings can be accessed in other ways.

z3ntu, 2024-07-03:
Yeah I'm pretty sure you can use standard tools for everything V4L2-related.

And unless the webcam has some RGB lights I'm not aware of, I guess there's nothing to do :)

closed 2024-07-03 by z3ntu