openrazer/openrazer #2270: Two controllers, two computers, one computer doesn't function

ASCI-Blue, 2024-06-26:
Garuda 6.9.6 zen kernel 

I have rebooted several times
I have read troubleshooting page and tried recommended console commands. 

Issue here>

Preface: Two machines both running Garuda, polychromatic, Razer RGB controllers. 

On MY machine both rgb controllers work just fine, strips are addressable and seen by polychromatic. Daemon running, program runs. Tried with muliple strips from my rig and her rig. Ryzen 7600, MSI B650 Carbon, Corsair Dominator RAM. 

Her system, daemon running, no errors from Polychromatic, yet neither Razer RGB controller works when plugged in as in no lights on. No errors via troubleshooting, user account is added. Tried rebooting several times with no success. Tried power cycles with no success. Hardware: Ryzen 5600x, MSI B550 Carbon, Viper RAM. 

Both systems had Polychromatic plus dependencies installed via Pamac/pacman.  Her system has been a bit flaky from install date, I'm considering reinstalling the entire OS. Any aid appreciated.

lah7, 2024-06-26:
Does the kernel see the USB device? This command lists Razer devices:

    lsusb | grep 1532

Let us know the IDs of the RGB controllers, so we can check driver support.

If you don't see any Razer devices at all (and still not listed with just `lsusb` either), then it sounds like a USB, hardware or kernel issue - assuming the same RGB controller works if plugged into your machine. The kernel log might give you some details:

    sudo dmesg | grep usb

ASCI-Blue, 2024-06-26:
hi there thanks for the reply! 

grep USB: 

[    4.704875] usb 1-2.1: New USB device found, idVendor=1532, idProduct=0f1f, bcdDevice= 2.00
[    4.704879] usb 1-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    4.704881] usb 1-2.1: Product: Razer Chroma Addressable RGB Controller
[    4.704883] usb 1-2.1: Manufacturer: Razer
[    4.752311] hid-generic 0003:1532:0F1F.0004: hidraw3: USB HID v1.11 Device [Razer Razer Chroma Addressable RGB Controller] on usb-0000:02:00.0-2.1/input0
[    4.762321] hid-generic 0003:1532:0F1F.0005: hidraw4: USB HID v1.11 Device [Razer Razer Chroma Addressable RGB Controller] on usb-0000:02:00.0-2.1/input1
[    4.772314] hid-generic 0003:1532:0F1F.0006: hidraw5: USB HID v1.11 Device [Razer Razer Chroma Addressable RGB Controller] on usb-0000:02:00.0-2.1/input2

Beyond that is other hardware, no errors. 

lsusb: 

╰─λ lsusb | grep 1532
Bus 001 Device 004: ID 1532:0f1f Razer USA, Ltd Razer Chroma Addressable RGB Controller

under lsmod the razer driver is loaded.

lah7, 2024-06-27:
That's good that the hardware side appears to be OK.

> Her system, daemon running, no errors from Polychromatic, yet neither Razer RGB controller works when plugged in as in no lights on.

I'm guessing the devices _are_ recognised and listed. Is that right? Are you able to see the effects and brightness slider in Polychromatic, but nothing happens?

If the devices _are_ there, but do nothing, try setting a brightness of around 30% _first_, then set the effect, since some Razer firmware may end up "Off" if the device powered on with 0% brightness. You could disable persistence and "off on screensaver" in `~/.config/openrazer/razer.conf` (or via Polychromatic's OpenRazer settings) so there isn't anything interfering with the selected effect, in case it is just entering an "Off" state.

Hardware wise, if it's anything like the _Chroma HDK_ I have: if there's not enough power/current, the device shuts off and blinks one of the LEDs in red. In my case, one of power connections was optional. If this controller is similiar, maybe try just one (or both) of the two USB power inputs?

If both controllers work in your system, but neither work in the other, and OpenRazer/Polychromatic runs as expected on both, I would suspect the Razer firmware isn't happy with the power input. Could try another port or a powered USB hub?

ASCI-Blue, 2024-06-27:
I'll be doing this in 2 parts. Now it's 0936 Mountain time US, not sure what that is UTC. 

1. Correct on devices recognized and listed. Correct I can see effects and brightness slider in Polychromatic but nothing happens. 

2. I tried setting for 30, then selecting and still nothing happened. I also disabled off on screensaver yesterday or the day before. 

3. This unit has one barrel to molex power plug and one JST to motherboard USB 2 header plug. To clarify it is this one: https://www.razer.com/gaming-pc-accessories/razer-chroma-addressable-rgb-controller

4. I'm going to try to get it running in Windows to see if maybe the issue is at the OS level. I'm not sure what happened but at some point during the OS install on Wife's machine a ton of stuff got broken. I had to keyboard command open CLI to properly install a bunch of KDE stuff and there's some other oddities with her install as well.

lah7, 2024-06-27:
No worries about time. You can hover over my avatar for mine 🕐 🌍 

It does seem pretty strange. Although if the driver is throwing an error, it should appear in `sudo dmesg -w`  (`-w` for watch to keep it open)

In case it's the D-Bus session bus that's not delivering the message, you could try the driver directly:

    cd /sys/bus/hid/drivers/razeraccessory/

There will be three folders per device, one will have all the OpenRazer sysfs files, e.g. `device_serial`, `firmware_version`, etc. They should be owned by `plugdev` group too:

```
$  pwd
/sys/bus/hid/drivers/razeraccessory/0003:1532:0F08.0007

$  ls -lah
drwxr-xr-x 5 root plugdev    0 Jun 27 15:42 .
drwxr-xr-x 5 root root       0 Jun 27 17:35 ..
-rw-rw---- 1 root plugdev 4.0K Jun 27 15:42 device_mode
-r--r----- 1 root plugdev 4.0K Jun 27 15:42 device_serial
-r--r----- 1 root plugdev 4.0K Jun 27 15:42 device_type
```

This would set a green static effect:

    echo -n -e "\x00\xFF\x00" > matrix_effect_static

ASCI-Blue, 2024-06-27:
I likes stories so pardon while I embellish a little. Good call on power btw. 

So after rebooting to Windows and having Synapse tell me plug in DC power, it was, I busted out ye olde multimeter. 750 watt EVGA power supply so not a junker and tested the power rails. It seems the perif port may have the wrong power cable plugged into it as the 12v is not where it should be based on plugging in the Razer's power cable. For some reason the 5v rail meets the 12v power on the controller.

lah7, 2024-06-27:
Bit curious, which shell are you using? Can't see the group in your output:

    .rw-rw---- 4.1k root 27 Jun 10:53  device_mode
    .r--r----- 4.1k root 27 Jun 10:53  device_serial

Whereas under bash:
    
    -rw-rw---- 1 root plugdev 4.0K Jun 27 15:42 device_mode
    -r--r----- 1 root plugdev 4.0K Jun 27 15:42 device_serial
                      ^^^^^^^

Doesn't matter anyway, since the daemon (and Polychromatic troubleshooting) would report an error if the user account isn't in the `plugdev` group. The `gpasswd` command doesn't leave a confirmation message. It's just needed to be able to write to the driver sysfs file without root. The command `groups` can be used to confirm.

lah7, 2024-06-27:
> On her machine it seems until the daemon is started the controller works with both channels. Once the daemon starts the other occupied channel (not sure which one off the top of my head) shuts off leaving one working.

Looking at the integration for this device, I just realised that the frontends (Polychromatic & RazerGenie) haven't implemented all of the functions specific to this device, specifically the channels (size of strip, brightness)

* https://github.com/openrazer/openrazer/pull/1929

I just created an issue now:

* https://github.com/polychromatic/polychromatic/issues/520
* _(Will also need some work in OpenRazer's Python library - functions are missing)_

This _could_ explain why some channels are going into darkness after the daemon starts if we assume it's configured to only have zero LEDs on one strip (channel). I've never used this device before, so just a guess.

Assuming you've got the right voltage now and it (half) illuminates 💡 , Polychromatic has an **Inspect Matrix** button in **Device Info** to test each LED. That could test if the channel settings are important or if they're only necessary for hardware effects. I guess also, what does the brightness slider do if each channel has its own brightness too?

In the meantime, you can use **QDBusViewer** (from package `qt6-tools`) to send some channel commands to the hardware. It's under **Session Bus** → `org.razer` → `device/<your serial>/razer.device.lighting.channel`. Double click to get/set.

[image]

ASCI-Blue, 2024-06-27:
Shell: Shell: fish 3.7.1 from fastfetch. 

The Matrix related feature doesn't work on either machine, in theory it should allow me as a midwit to do cool stuff but it doesn't. All I've been able to do with it is turn things off and when exiting it comes back. 

As for using Qt QDBusViewer, I can see everything, but when I go to change something it doesn't save. 

I wonder now if this is simply the support isn't there so the hardware is somewhat flaky even if OpenRazer claims support. 

Interestingly and as a complete aside when OpenRazer is installed OpenRGB can't detect the Razer RGB Controller.

lah7, 2024-06-28:
It's possible support is flaky; or maybe it only worked for a specific (older?) firmware version. What firmware version is yours?

Ironically, the guy originally contributing the driver code **is** the maintainer of OpenRGB:

* 👉🏼 https://github.com/openrazer/openrazer/issues/1357 👈🏼 
* 👉🏼 https://github.com/openrazer/openrazer/pull/1391 👈🏼 

For just the driver part; others built on top for the missing daemon parts:
  * https://github.com/openrazer/openrazer/pull/1762
  * https://github.com/openrazer/openrazer/pull/1929

I see [a comment](https://github.com/openrazer/openrazer/pull/1929#issuecomment-1307961686) the channel methods are working. It is incomplete since the Python library interface is missing, so Polychromatic wasn't aware of those extra methods.

I honestly can't remember how OpenRGB uses OpenRazer, if it does at all. I know under Windows it either uses Razer Synapse SDK or [openrazer-win32](https://github.com/CalcProgrammer1/openrazer-win32). I haven't had luck in the past with OpenRGB myself under Linux.

> As for using Qt QDBusViewer, I can see everything, but when I go to change something it doesn't save.

On both machines? At the very least, are there any `razeraccessory` errors seen if you leave a terminal open running `sudo dmesg -w`? That would show if the firmware wasn't happy with the command it received.

Not sure if the input is 0 or 1-based. Aka. channel 0 == port 1, or channel 1 == port 1? Or must it be the exact number or less (e.g. if set to `80` for a strip of 20 LEDs, it doesn't work if it's out of range?)

Something out of curiosity, but does this Razer device "remember" its settings across power cycles? Could it be possible to set up lighting in Razer Synapse on Windows, power off, and boot into BIOS setup and still light up the same? (Obviously, booting into Linux with OpenRazer will activate 'driver mode')

closed 2025-03-16 by github-actions[bot]